coolsnowwolf/lede #9286: 询问关于 fstab 做过的修改

LawssssCat, 2022-04-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

询问关于 fstab 做过的修改

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

我希望通过修改 /etc/config/fstab 进行 /overlay 扩容
在lede固件里成功了
但是在其他固件里不生效，像原生openwrt、lienol、immoralwrt
所以我想请问，你在lede里做了哪些修改？
导致 
写了 /etc/config/fstab 的/overlay挂载并重启后
```bash
$ mount 
...
/dev/loop0 on /overlay type f2fs (rw,lazytime,noatime,background_gc=on,nsdiscard,no_heap,use ... 
```
就会变成
```bash
$ mount 
...
/dev/sdb1 on /overlay type ext4 (rw,relatime)
...
```

LawssssCat, 2022-04-21:
请问你做了哪些设置，使得修改了 /etc/config/fstab 后
系统会使用  /etc/config/fstab 里指定的块挂载 /overlay 而不是用 /dev/loop0  来挂载 /overlay ?
[image]

@coolsnowwolf

WYC-2020, 2022-04-24:
自己不会看commit嘛，所有修改都有记录

LawssssCat, 2022-04-26:
> 自己不会看提交嘛，所有修改都有记录

可以告诉我从哪看起吗？

javanli, 2022-06-18:
https://github.com/coolsnowwolf/lede/blob/master/package/system/fstools/patches/0001-fstools-support-extroot-for-non-MTD-rootfs_data.patch

https://github.com/coolsnowwolf/lede/pull/5472

https://github.com/coolsnowwolf/lede/issues/5353

coolsnowwolf, 2026-09-05:
这是关于不同发行版 overlay/fstab 行为的咨询，没有提供 LEDE 与其他系统的完整 UCI、挂载顺序和启动日志；无法确认仓库改动，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf