coolsnowwolf/lede #9290: 5.15内核编译mt7915出错

liwenjie119, 2022-04-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：全新拉库从头编译，包含有mt7915的路由器如cr660x在编译至build_dir/target-mipsel_24kc_musl/lin
mt7621/mt76-2022-04-07-506bb060/mt7915/pci.c时报错

### (2) 路由器型号和固件版本

A：包含有mt7915的路由器如cr660x等，使用5.15内核

### (3) 详细日志

A：
-- Build files have been written to: /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/tools
touch /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/.configured_b1c6ae92f8e974e60b47b3e5ad3573cc
rm -f /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/.built
touch /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/.built_check
make  -C "/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.35" KCFLAGS="-fmacro-prefix-map=/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/liwenjie/lede/staging_dir/host/include -I/home/liwenjie/lede/staging_dir/hostpkg/include -I/home/liwenjie/lede/staging_dir/target-mipsel_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Apr 21 08:16:10 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/liwenjie/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/liwenjie/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/cryptodev-linux.symvers /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/gpio-button-hotplug.symvers /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/mac80211.symvers" KERNELRELEASE=5.15.35 CONFIG_MT76_CONNAC_LIB=m CONFIG_MT7615_COMMON=m CONFIG_MT7915E=m M="/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060" NOSTDINC_FLAGS="-nostdinc  -isystem /home/liwenjie/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/include -I/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060 -I/home/liwenjie/lede/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport/uapi -I/home/liwenjie/lede/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport -I/home/liwenjie/lede/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/uapi -I/home/liwenjie/lede/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211 -include backport/autoconf.h -include backport/backport.h -DCONFIG_MAC80211_MESH" modules
make[4]: Entering directory '/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.35'
  CC [M]  /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7615/main.o
  CC [M]  /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7615/init.o
  CC [M]  /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7615/mcu.o
  CC [M]  /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7615/eeprom.o
  CC [M]  /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7615/mac.o
  CC [M]  /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7615/debugfs.o
  CC [M]  /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7615/trace.o
  LD [M]  /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7615/mt7615-common.o
  CC [M]  /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915/pci.o
/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915/pci.c: In function 'mt7915_pci_probe':
/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915/pci.c:209:43: error: 'PCI_IRQ_ALL_TYPES' undeclared (first use in this function); did you mean 'NUM_RFKILL_TYPES'?
   ret = pci_alloc_irq_vectors(pdev, 1, 1, PCI_IRQ_ALL_TYPES);
                                           ^~~~~~~~~~~~~~~~~
                                           NUM_RFKILL_TYPES
/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915/pci.c:209:43: note: each undeclared identifier is reported only once for each function it appears in
/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915/pci.c:260:3: error: implicit declaration of function 'pci_free_irq_vectors'; did you mean 'pci_alloc_irq_vectors'? [-Werror=implicit-function-declaration]
   pci_free_irq_vectors(pdev);
   ^~~~~~~~~~~~~~~~~~~~
   pci_alloc_irq_vectors
cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:277: /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915/pci.o] Error 1
make[5]: *** [scripts/Makefile.build:540: /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/mt7915] Error 2
make[4]: *** [Makefile:1868: /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060] Error 2
make[4]: Leaving directory '/home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.35'
make[3]: *** [Makefile:469: /home/liwenjie/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-04-07-506bb060/.built] Error 2
make[3]: Leaving directory '/home/liwenjie/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#11.23#2.57#14.26
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/home/liwenjie/lede'
make[1]: *** [package/Makefile:110: /home/liwenjie/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/liwenjie/lede'
make: *** [/home/liwenjie/lede/include/toplevel.mk:230：world] 错误 2

liwenjie119, 2022-04-21:
我测试了openwrt/openwrt库使用5.15内核编译，没有问题，但是本库编译出错，替换mt76到最新版本也不行
类似的错误有：
https://github.com/coolsnowwolf/lede/issues/9217
https://github.com/coolsnowwolf/lede/issues/9211
https://github.com/coolsnowwolf/lede/issues/9231

WYC-2020, 2022-04-22:
不知所谓，请说明复现步骤，怎么选择的平台

britneyjeanspears, 2022-04-22:
> 不知所谓，请说明复现步骤，怎么选择的平台

不说了选cr660x+5.15内核，mt76就编译失败。这说得很清楚了。

liwenjie119, 2022-04-22:
> 不知所谓，请说明复现步骤，怎么选择的平台

选cr660x，只更改选中5.15内核：
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_xiaomi_mi-router-cr660x=y
CONFIG_LINUX_5_15=y
CONFIG_TESTING_KERNEL=y

WYC-2020, 2022-04-23:
> > 不知所谓，请说明复现步骤，怎么选择的平台
> 
> 选cr660x，只更改选中5.15内核： CONFIG_TARGET_ramips=y CONFIG_TARGET_ramips_mt7621=y CONFIG_TARGET_ramips_mt7621_DEVICE_xiaomi_mi-router-cr660x=y CONFIG_LINUX_5_15=y CONFIG_TESTING_KERNEL=y

亲测mt76 没有任何问题，只是eip93这个驱动有点问题而已，我已经修复

WYC-2020, 2022-04-23:
> > 不知所谓，请说明复现步骤，怎么选择的平台
> 
> 不说了选cr660x+5.15内核，mt76就编译失败。这说得很清楚了。

没问题，说的很清楚了

WYC-2020, 2022-04-23:
> > 不知所谓，请说明复现步骤，怎么选择的平台
> 
> 选cr660x，只更改选中5.15内核： CONFIG_TARGET_ramips=y CONFIG_TARGET_ramips_mt7621=y CONFIG_TARGET_ramips_mt7621_DEVICE_xiaomi_mi-router-cr660x=y CONFIG_LINUX_5_15=y CONFIG_TESTING_KERNEL=y

另外自己更新mt到l相同版本，如果不是相同版本，自己的组合问题，自己解决了

liwenjie119, 2022-04-24:
> > > 不知所谓，请说明复现步骤，怎么选择的平台
> > 
> > 
> > 选cr660x，只更改选中5.15内核： CONFIG_TARGET_ramips=y CONFIG_TARGET_ramips_mt7621=y CONFIG_TARGET_ramips_mt7621_DEVICE_xiaomi_mi-router-cr660x=y CONFIG_LINUX_5_15=y CONFIG_TESTING_KERNEL=y
> 
> 另外自己更新mt到l相同版本，如果不是相同版本，自己的组合问题，自己解决了

大佬，我重新装了20.04，全新拉库，编译cr660x，使用5.15内核，的确遇到了这个问题。但是我只在lean的库复现了这个问题，你的库没有问题，官方openwrt库也没有遇到这个问题。目前只有lean的库出现这个问题

ndevel, 2022-04-25:
我给京东云无线宝编译也遇到同样问题，选择测试版内核，即5.15内核，仓库最新代码，一模一样的问题

uniessey, 2022-05-08:
是这样的，我已经编译出了5.15，我自己改了rt_liunx.c，也补齐了5.15的配置文件。编译过程很顺利，没出错。准备今天测试一下。但是我的硬件版本是NEWIFI-D2，万一硬件版本不支持，可能还是无法启动。

coolsnowwolf, 2026-09-05:
报告中的 mt7915 编译错误属于旧版 mt76/5.15 组合；当前树已更新 mt76 源码并包含对应内核 API 适配，现行包不再使用该旧快照，关闭为“已完成”。

closed 2026-09-05 by coolsnowwolf