coolsnowwolf/lede #9291: ASUS RT-ACRH17_batman-adv报错信息

zhb7670, 2022-04-21:
make  -C "/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-5.4.[1](https://github.com/zhb7670/ACRH17/runs/6110960710?check_suite_focus=true#step:12:1)88" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi=target-arm_cortex-a7+neon-vfpv4_musl_eabi" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Apr 20 21:35:59 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/gpio-button-hotplug.symvers /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/shortcut-fe.symvers /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/ath10k-ct.symvers /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/mac80211.symvers /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/ntfs3-oot.symvers /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/antfs.symvers" KERNELRELEASE=5.4.188 M="/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/net/batman-adv" CONFIG_BATMAN_ADV=m CONFIG_BATMAN_ADV_DEBUG=n CONFIG_BATMAN_ADV_BLA=y CONFIG_BATMAN_ADV_DAT=y CONFIG_BATMAN_ADV_MCAST=y CONFIG_BATMAN_ADV_NC=n CONFIG_BATMAN_ADV_BATMAN_V=y CONFIG_BATMAN_ADV_TRACING=n  EXTRA_CFLAGS=" -DCONFIG_BATMAN_ADV=1  -DCONFIG_BATMAN_ADV_BLA=1  -DCONFIG_BATMAN_ADV_DAT=1  -DCONFIG_BATMAN_ADV_MCAST=1  -DCONFIG_BATMAN_ADV_BATMAN_V=1 " NOSTDINC_FLAGS="-nostdinc  -isystem /workdir/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/include -I/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/net/batman-adv -I/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport -I/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/uapi -I/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211 -I/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211/uapi -I/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/include/ -include backport/autoconf.h -include backport/backport.h -include /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/compat-hacks.h -DBATADV_SOURCE_VERSION=\\\"2022.0-openwrt-1\\\"" modules
make[4]: Entering directory '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-5.4.188'
  CC [M]  /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/net/batman-adv/bat_algo.o
In file included from <command-line>:
/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/compat-hacks.h: In function 'br_multicast_has_router_adjacent':
/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/compat-hacks.h:32:53: error: 'struct br_ip' has no member named 'dst'
       ipv6_addr_is_ll_all_routers(&br_ip_entry->addr.dst.ip6))
                                                     ^
make[5]: *** [scripts/Makefile.build:262: /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/net/batman-adv/bat_algo.o] Error 1
make[4]: *** [Makefile:[17](https://github.com/zhb7670/ACRH17/runs/6110960710?check_suite_focus=true#step:12:17)34: /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/net/batman-adv] Error 2
make[3]: *** [Makefile:95: /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/.built] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-5.4.[18](https://github.com/zhb7670/ACRH17/runs/6110960710?check_suite_focus=true#step:12:18)8'
make[3]: Leaving directory '/workdir/openwrt/feeds/routing/batman-adv'
time: package/feeds/routing/batman-adv/compile#1.05#0.31#1.34
    ERROR: package/feeds/routing/batman-adv failed to build.
make[2]: *** [package/Makefile:116: package/feeds/routing/batman-adv/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:[23](https://github.com/zhb7670/ACRH17/runs/6110960710?check_suite_focus=true#step:12:23)0: world] Error 2
Error: Process completed with exit code 2.
0s

WYC-2020, 2022-04-22:
更新代码，全新编译

JoseCoW, 2022-04-22:
要把 feeds 里的 routing 改成
```
src-git routing https://git.openwrt.org/feed/routing.git;openwrt-21.02
```
原因就是上游主线不支持 5.4 了。

WYC-2020, 2022-04-22:
> make -C "/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-5.4.[1](https://github.com/zhb7670/ACRH17/runs/6110960710?check_suite_focus=true#step:12:1)88" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi=target-arm_cortex-a7+neon-vfpv4_musl_eabi" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Apr 20 21:35:59 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V='' cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/gpio-button-hotplug.symvers /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/shortcut-fe.symvers /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/ath10k-ct.symvers /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/mac80211.symvers /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/ntfs3-oot.symvers /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/symvers/antfs.symvers" KERNELRELEASE=5.4.188 M="/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/net/batman-adv" CONFIG_BATMAN_ADV=m CONFIG_BATMAN_ADV_DEBUG=n CONFIG_BATMAN_ADV_BLA=y CONFIG_BATMAN_ADV_DAT=y CONFIG_BATMAN_ADV_MCAST=y CONFIG_BATMAN_ADV_NC=n CONFIG_BATMAN_ADV_BATMAN_V=y CONFIG_BATMAN_ADV_TRACING=n EXTRA_CFLAGS=" -DCONFIG_BATMAN_ADV=1 -DCONFIG_BATMAN_ADV_BLA=1 -DCONFIG_BATMAN_ADV_DAT=1 -DCONFIG_BATMAN_ADV_MCAST=1 -DCONFIG_BATMAN_ADV_BATMAN_V=1 " NOSTDINC_FLAGS="-nostdinc -isystem /workdir/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/include -I/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/net/batman-adv -I/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport -I/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/uapi -I/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211 -I/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211/uapi -I/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/include/ -include backport/autoconf.h -include backport/backport.h -include /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/compat-hacks.h -DBATADV_SOURCE_VERSION=\"2022.0-openwrt-1\"" modules make[4]: Entering directory '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-5.4.188' CC [M] /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/net/batman-adv/bat_algo.o In file included from : /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/compat-hacks.h: In function 'br_multicast_has_router_adjacent': /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/compat-hacks.h:32:53: error: 'struct br_ip' has no member named 'dst' ipv6_addr_is_ll_all_routers(&br_ip_entry->addr.dst.ip6)) ^ make[5]: *** [scripts/Makefile.build:262: /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/net/batman-adv/bat_algo.o] Error 1 make[4]: *** [Makefile:[17](https://github.com/zhb7670/ACRH17/runs/6110960710?check_suite_focus=true#step:12:17)34: /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/net/batman-adv] Error 2 make[3]: *** [Makefile:95: /workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/batman-adv-2022.0/.built] Error 2 make[4]: Leaving directory '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-5.4.[18](https://github.com/zhb7670/ACRH17/runs/6110960710?check_suite_focus=true#step:12:18)8' make[3]: Leaving directory '/workdir/openwrt/feeds/routing/batman-adv' time: package/feeds/routing/batman-adv/compile#1.05#0.31#1.34 ERROR: package/feeds/routing/batman-adv failed to build. make[2]: *** [package/Makefile:116: package/feeds/routing/batman-adv/compile] Error 1 make[2]: Leaving directory '/workdir/openwrt' make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2 make[1]: Leaving directory '/workdir/openwrt' make: *** [/workdir/openwrt/include/toplevel.mk:[23](https://github.com/zhb7670/ACRH17/runs/6110960710?check_suite_focus=true#step:12:23)0: world] Error 2 Error: Process completed with exit code 2. 0s

把这个自己fork,然后撤回这个提交batman-adv: Drop compat hacks for kernels < 5.10

WYC-2020, 2022-04-22:
> make -C "/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/linux-5.4.[1](https://github.com/zhb7670/ACRH17/runs/6110960710?check_suite_focus=true#step:12:1)88" [...]

https://github.com/WYC-2020/lede/commit/ea63f816b9cdcd13d44a9cd6a5c070b760d82972 用这个版本，install的时候不要加入-f 强制覆盖即可，其实我建议使用linux 自带的原生驱动，但是没有测试过，我这边加入了看你敢不敢测试https://github.com/WYC-2020/lede/commit/f6cec788776b8436ae3ab787a81af6b40e7839a7

WYC-2020, 2022-04-22:
@zhb7670  功能都需要测试哈，这个兼容5.4是参考5.4自带的原生驱动修改的，看下第三方的能不能工作，反正我是建议用linux自带的原生这样就不存在兼容问题了

ecsv, 2022-04-22:
The native one is build against the internal cfg80211 of the kernel. But openwrt is shipping cfg80211 in an external mac80211 package which is not always 100% ABI compatible (and often not even API compatible). And you are not the first one who tries to use the batman-adv from the kernel against cfg80211 from the mac80211 package - which resulted in the past (and most likely also in the future - or maybe at the moment in your OpenWrt fork) in hard to debug crashes or (not) funny behavior of the module.

One way to avoid these kind of crashes is not to enable `CONFIG_BATMAN_ADV_BATMAN_V=y`. At least at the moment, batman-adv will then be build without cfg80211 dependency.

WYC-2020, 2022-04-22:
> The native one is build against the internal cfg80211 of the kernel. But openwrt is shipping cfg80211 in an external mac80211 package which is not always 100% ABI compatible (and often not even API compatible). And you are not the first one who tries to use the batman-adv from the kernel against cfg80211 from the mac80211 package - which resulted in the past (and most likely also in the future - or maybe at the moment in your OpenWrt fork) in hard to debug crashes or (not) funny behavior of the module.
> 
> One way to avoid these kind of crashes is not to enable `CONFIG_BATMAN_ADV_BATMAN_V=y`. At least at the moment, batman-adv will then be build without cfg80211 dependency.
> 
> Btw. you should really think about getting more in sync with upstream OpenWrt instead of doing wild cherry picking (if you can call this cherry picking at all) and mixing of various things from upstream OpenWrt.

Thanks for the explanation. I know how to change it

zhb7670, 2022-04-22:
> 更新代码，全新编译

 https://github.com/zhb7670/lede.git  更新过的

WYC-2020, 2022-04-23:
> > 更新代码，全新编译
> 
> https://github.com/zhb7670/lede.git 更新过的

撤销上游的这两个提交即可
[image]

huah0235, 2022-04-27:
同样问题

xiaopang0v0, 2023-02-14:
因为你要编译的是老版本，并且 目标内核 < 5.10 ，只需要在
`feeds.conf.default` 中 
```
src-git routing https://git.openwrt.org/feed/routing.git
```
的结尾加上 `;openwrt-21.02` 即可。(openwrt-21.02 是分支名，请不要自作主张将其改为你要编译的版本)
最终效果为
```
src-git routing https://git.openwrt.org/feed/routing.git;openwrt-21.02
```
然后在 lede 根目录更新 feeds 文件

```
./scripts/feeds update -a
./scripts/feeds install -a
```

coolsnowwolf, 2026-09-05:
该 issue 与 #9320 同为旧 batman-adv 2022.0/5.4 兼容性失败。当前 feeds/routing/batman-adv 已更新到 2023.3，旧 `struct br_ip` 字段错误已不再适用，关闭为 completed。

closed 2026-09-05 by coolsnowwolf