coolsnowwolf/lede #9298: 小米路由4a百兆版MT7628AN大流量断网，wifi信号消失

2391627307, 2022-04-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：路由器在大负载运行一会儿后，wifi信号消失，过几分钟又出现。

### (2) 路由器型号和固件版本

A：小米路由4a百兆版
具体型号：
Xiaomi Mi Router 4A (100M Edition) (CpuMark : 1373.906510 Scores)
SOC:
MediaTek MT7628AN ver:1 eco:2 x 1
固件版本：
OpenWrt R22.3.13 / LuCI Master (git-22.086.14749-160352f)
内核版本：5.4.187
### (3) 详细日志

A：
Fri Apr 22 22:56:25 2022 kern.err kernel: [37230.123564] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Apr 22 22:56:25 2022 kern.info kernel: [37230.130155] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:00000055
Fri Apr 22 22:56:25 2022 kern.info kernel: [37230.136562] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=03990000, max=1024, ctx=799, dtx=799, fdx=763, next=799
Fri Apr 22 22:56:25 2022 kern.info kernel: [37230.147794] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=02f0c000, max=1024, calc=836, drx=837
Fri Apr 22 22:56:35 2022 kern.err kernel: [37240.123651] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Apr 22 22:56:35 2022 kern.info kernel: [37240.130214] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:00000055
Fri Apr 22 22:56:35 2022 kern.info kernel: [37240.136635] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=03990000, max=1024, ctx=174, dtx=174, fdx=145, next=174
Fri Apr 22 22:56:35 2022 kern.info kernel: [37240.147864] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=01914000, max=1024, calc=114, drx=115
Fri Apr 22 22:56:45 2022 kern.err kernel: [37250.123427] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Apr 22 22:56:45 2022 kern.info kernel: [37250.129983] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:00000055
Fri Apr 22 22:56:45 2022 kern.info kernel: [37250.136440] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=03990000, max=1024, ctx=168, dtx=168, fdx=32, next=168
Fri Apr 22 22:56:45 2022 kern.info kernel: [37250.147577] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=02f0c000, max=1024, calc=123, drx=124
Fri Apr 22 22:56:55 2022 kern.err kernel: [37260.123434] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Apr 22 22:56:55 2022 kern.info kernel: [37260.129992] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:00000055
Fri Apr 22 22:56:55 2022 kern.info kernel: [37260.136390] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=03990000, max=1024, ctx=38, dtx=38, fdx=16, next=38
Fri Apr 22 22:56:55 2022 kern.info kernel: [37260.147269] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=01914000, max=1024, calc=61, drx=62
Fri Apr 22 22:57:15 2022 kern.err kernel: [37280.123348] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Apr 22 22:57:15 2022 kern.info kernel: [37280.129920] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:00000055
Fri Apr 22 22:57:15 2022 kern.info kernel: [37280.136319] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=03990000, max=1024, ctx=684, dtx=684, fdx=632, next=684
Fri Apr 22 22:57:15 2022 kern.info kernel: [37280.147544] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=02f0c000, max=1024, calc=575, drx=576
Fri Apr 22 22:57:25 2022 kern.err kernel: [37290.123356] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Apr 22 22:57:25 2022 kern.info kernel: [37290.129931] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:00000055
Fri Apr 22 22:57:25 2022 kern.info kernel: [37290.136334] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=03990000, max=1024, ctx=147, dtx=147, fdx=33, next=147
Fri Apr 22 22:57:25 2022 kern.info kernel: [37290.147479] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=01914000, max=1024, calc=122, drx=123
Fri Apr 22 22:57:35 2022 kern.err kernel: [37300.123339] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Apr 22 22:57:35 2022 kern.info kernel: [37300.129895] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:00000055
Fri Apr 22 22:57:35 2022 kern.info kernel: [37300.136317] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=03990000, max=1024, ctx=45, dtx=45, fdx=16, next=45
Fri Apr 22 22:57:35 2022 kern.info kernel: [37300.147191] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=02f0c000, max=1024, calc=64, drx=65
Fri Apr 22 22:57:45 2022 kern.err kernel: [37310.123273] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Apr 22 22:57:45 2022 kern.info kernel: [37310.129844] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:00000055
Fri Apr 22 22:57:45 2022 kern.info kernel: [37310.136240] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=03990000, max=1024, ctx=132, dtx=132, fdx=106, next=132
Fri Apr 22 22:57:45 2022 kern.info kernel: [37310.147475] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=01914000, max=1024, calc=175, drx=176
Fri Apr 22 22:57:55 2022 kern.err kernel: [37320.123257] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Apr 22 22:57:55 2022 kern.info kernel: [37320.129841] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:00000055
Fri Apr 22 22:57:55 2022 kern.info kernel: [37320.136238] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=03990000, max=1024, ctx=214, dtx=214, fdx=32, next=214
Fri Apr 22 22:57:55 2022 kern.info kernel: [37320.147383] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=02f0c000, max=1024, calc=200, drx=201
Fri Apr 22 22:58:05 2022 kern.err kernel: [37330.123678] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Apr 22 22:58:05 2022 kern.info kernel: [37330.130294] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:00000055
Fri Apr 22 22:58:05 2022 kern.info kernel: [37330.136722] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=03990000, max=1024, ctx=39, dtx=39, fdx=16, next=39
Fri Apr 22 22:58:05 2022 kern.info kernel: [37330.147592] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=01914000, max=1024, calc=141, drx=142
Fri Apr 22 22:58:15 2022 kern.err kernel: [37340.124331] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Apr 22 22:58:15 2022 kern.info kernel: [37340.130886] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:00000055
Fri Apr 22 22:58:15 2022 kern.info kernel: [37340.137311] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=03990000, max=1024, ctx=19, dtx=19, fdx=16, next=19
Fri Apr 22 22:58:15 2022 kern.info kernel: [37340.148206] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=02f0c000, max=1024, calc=72, drx=73
[... 164 lines cut ...]
Fri Apr 22 22:59:46 2022 daemon.notice hostapd: wlan1: interface state HT_SCAN->ENABLED
Fri Apr 22 22:59:46 2022 daemon.notice hostapd: wlan1: AP-ENABLED
Fri Apr 22 22:59:46 2022 daemon.notice netifd: Network device 'wlan0' link is up
Fri Apr 22 22:59:46 2022 daemon.notice netifd: Network device 'wlan1' link is up
Fri Apr 22 22:59:50 2022 daemon.info hostapd: wlan1: STA dc:a6:32:aa:3d:a7 IEEE 802.11: authenticated
Fri Apr 22 22:59:50 2022 daemon.info hostapd: wlan1: STA dc:a6:32:aa:3d:a7 IEEE 802.11: associated (aid 1)
Fri Apr 22 22:59:50 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED dc:a6:32:aa:3d:a7
Fri Apr 22 22:59:50 2022 daemon.info hostapd: wlan1: STA dc:a6:32:aa:3d:a7 RADIUS: starting accounting session 0DF61E82EBE5EEDD
Fri Apr 22 22:59:50 2022 daemon.info hostapd: wlan1: STA dc:a6:32:aa:3d:a7 WPA: pairwise key handshake completed (RSN)
Fri Apr 22 22:59:50 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED dc:a6:32:aa:3d:a7
Fri Apr 22 23:00:30 2022 kern.info kernel: [37474.338223] rt3050-esw 10110000.esw: port 2 link down
Fri Apr 22 23:00:33 2022 kern.info kernel: [37477.473580] rt3050-esw 10110000.esw: port 2 link up
Fri Apr 22 23:00:46 2022 daemon.info hostapd: wlan1: STA 4c:77:66:26:4f:b8 IEEE 802.11: authenticated
Fri Apr 22 23:00:46 2022 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 4c:77:66:26:4f:b8 1
Fri Apr 22 23:00:46 2022 daemon.info hostapd: wlan1: STA 4c:77:66:26:4f:b8 IEEE 802.11: associated (aid 2)
Fri Apr 22 23:00:46 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 4c:77:66:26:4f:b8
Fri Apr 22 23:00:46 2022 daemon.info hostapd: wlan1: STA 4c:77:66:26:4f:b8 RADIUS: starting accounting session 22C4109BD34FB914
Fri Apr 22 23:00:46 2022 daemon.info hostapd: wlan1: STA 4c:77:66:26:4f:b8 WPA: pairwise key handshake completed (RSN)
Fri Apr 22 23:00:46 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 4c:77:66:26:4f:b8
Fri Apr 22 23:01:30 2022 daemon.info hostapd: wlan1: STA 3c:f8:62:31:79:2d IEEE 802.11: authenticated
Fri Apr 22 23:01:30 2022 daemon.info hostapd: wlan1: STA 3c:f8:62:31:79:2d IEEE 802.11: associated (aid 3)
Fri Apr 22 23:01:30 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 3c:f8:62:31:79:2d
Fri Apr 22 23:01:30 2022 daemon.info hostapd: wlan1: STA 3c:f8:62:31:79:2d RADIUS: starting accounting session 0A821F1A3C88881F
Fri Apr 22 23:01:30 2022 daemon.info hostapd: wlan1: STA 3c:f8:62:31:79:2d WPA: pairwise key handshake completed (RSN)
Fri Apr 22 23:01:30 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 3c:f8:62:31:79:2d

LINGJP, 2022-04-22:
7620 7628这种u就别跑5.4内核了，回4内核稳定

2391627307, 2022-04-23:
> 

我试一下，谢谢

2391627307, 2022-04-24:
> 7620 7628 这种你就别跑5.4内核了，回4内核稳定了

> 7620 7628这种u就别跑5.4内核了，回4内核稳定
怎么切换回4内核呀，该仓库没有找到跨版本升级的提交

LINGJP, 2022-05-16:
7621驱动相关修复的commit能提供一下看看吗？

2391627307, 2022-05-17:
> 7621驱动相关修复的commit能提供一下看看吗？

使用[Lienol](https://github.com/Lienol)这个仓库，测试没有问题，稳定许多。硬件小米4
a百兆版。

chobitsos, 2022-05-17:
> HG553

我也找不到那个commit，但是我测试编译22.5.5的7621固件，K2P和小米R4A千兆版测试了上下同时跑满没有出现这个BUG，以前我编译的K2P上下一跑就断流，22.5.5，我又测试了7620,还是同样又这个BUG有提过相关的ISSUES,你可以翻一下

2391627307, 2022-05-17:
> > HG553
> 
> 我也找不到那个commit，但是我测试编译22.5.5的7621固件，K2P和小米R4A千兆版测试了上下同时跑满没有出现这个BUG，以前我编译的K2P上下一跑就断流，22.5.5我又测试了7620,还是同样又这个BUG有提过相关的ISSUES,你可以翻一下

好的，我找找看。

chobitsos, 2022-05-17:
> > > HG553
> > 
> > 
> > 我也找不到那个commit，但是我测试编译22.5.5的7621固件，K2P和小米R4A千兆版测试了上下同时跑满没有出现这个BUG，以前我编译的K2P上下一跑就断流，22.5.5我又测试了7620,还是同样又这个BUG有提过相关的ISSUES,你可以翻一下
> 
> 好的，我找找看。

我翻出来了，看上面的回复

2391627307, 2022-05-17:
> > > > HG553
> > > 
> > > 
> > > 我也找不到那个commit，但是我测试编译22.5.5的7621固件，K2P和小米R4A千兆版测试了上下同时跑满没有出现这个BUG，以前我编译的K2P上下一跑就断流，22.5.5我又测试了7620,还是同样又这个BUG有提过相关的ISSUES,你可以翻一下
> > 
> > 
> > 好的，我找找看。
> 
> 我翻出来了，看上面的回复

我看了你提交的关于驱动的修改，那个修改你是对照其他稳定驱动找到的bug吗？

chobitsos, 2022-05-17:
> > > > > HG553
> 
> > > > 
> 
> > > > 
> 
> > > > 我也找不到那个commit，但是我测试编译22.5.5的7621固件，K2P和小米R4A千兆版测试了上下同时跑满没有出现这个BUG，以前我编译的K2P上下一跑就断流，22.5.5我又测试了7620,还是同样又这个BUG有提过相关的ISSUES,你可以翻一下
> 
> > > 
> 
> > > 
> 
> > > 好的，我找找看。
> 
> > 
> 
> > 我翻出来了，看上面的回复
> 
> 
> 
> 我看了你提交的关于驱动的修改，那个修改你是对照其他稳定驱动找到的bug吗？

打了这个补丁也没用，不能彻底解决这个问题，只能减少断流几率，实际我测试的话一样断流，减少纪律而已，意义不大，所以根本上还是没有解决这个bug

2391627307, 2022-05-17:
事实证明解决这个问题最简单的方式就是切换到lienol的仓库

chobitsos, 2022-05-19:
> 

你用的是https://github.com/Lienol/lean-lede这个仓库吗，我去编译一个试试

chobitsos, 2022-05-22:
已换SDK
测试30分钟双向跑满没有断流，适配lean的packges仓库
[image]

[image]

[image]

2391627307, 2022-05-23:
> > 
> 
> 你用的是https://github.com/Lienol/lean-lede 这个仓库吗，我去编译一个试试

是的，就是这个仓库，短时间内能够稳定运行，但是长期运行（一周以上）还是会出现断流WIFI消失的问题，看起来这个问题是很多仓库共有的。

coolsnowwolf, 2026-09-05:
MT7628AN 旧版 `ralink_soc_eth` transmit timeout 日志来自 5.4.187，缺少当前内核复现和硬件/流量对照；无法安全移植修复，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf