coolsnowwolf/lede #9300: 重新编译了，mi-router-4a-gigabit还是不启动，

yuzhe1129, 2022-04-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

更新了源码编译了，还是一样不启动，是不是不适合5.4内核？

### (2) 路由器型号和固件版本
小米4A千兆版

### (3) 详细日志
OpenWrt kernel loader for MIPS based SoC
Copyright (C) 2011 Gabor Juhos <juhosg@openwrt.org>
Decompressing kernel... done!
Starting kernel at 80001000...

[    0.000000] Linux version 5.4.190 (bbb@ubuntu) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4479-2a91c7245)) #0 SMP Fri Apr 22 14:50:56 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 4A Gigabit Edition
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00042e00
[    0.000000] Readback ErrCtl register=00042e00
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 120324K/131072K available (6144K kernel code, 254K rwdata, 1288K rodata, 1260K init, 238K bss, 10748K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007818] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.073993] pid_max: default: 32768 minimum: 301
[    0.078802] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.086002] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096837] rcu: Hierarchical SRCU implementation.
[    0.102249] smp: Bringing up secondary CPUs ...
[    3.533929] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    3.533940] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    3.533953] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    3.534068] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.167093] Synchronize counters for CPU 1: done.
[    3.245060] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    3.245068] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[... 380 lines cut ...]
[    1.790181]  (driver?)
[    1.796701] 1f06            1024 mtdblock6
[    1.796704]  (driver?)
[    1.803205] 1f07           14848 mtdblock7
[    1.803207]  (driver?)
[    1.809722] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    1.817988] Rebooting in 1 seconds..

Boot and Recovery Environment for Embedded Devices
Copyright (C) 2021 HackPascal <hackpascal@gmail.com>
Build date 2021-12-15 [git-f9b74d0]
Version 1.1 (r1337)

DRAM: 128MB
Platform: MediaTek MT7621A ver 1, eco 3
Board: PandoraBox PBR-M1 / Newifi Y2S
Clocks: CPU: 880MHz, DDR: 1040MHz, Bus: 220MHz, Ref: 40MHz
Flash: GigaDevice GD25Q128 (16MB) on mt7621-spi.0
rt2880-eth: MAC address from EEPROM is invalid, using default settings.
rt2880-eth: Using MAC address 00:0c:43:00:00:01
eth0: MediaTek MT7530 Gigabit switch

Network started on eth0, inet addr 192.168.1.1, netmask 255.255.255.0

Press any key to interrupt autoboot ... 4

zys91, 2022-04-23:
参考这个帖修改内存布局可以实现 https://www.right.com.cn/FORUM/thread-4052254-1-1.html
希望大雕有空可以将R4A-Gi和R3G-V2的内存布局文件独立出来，修复一下这个问题

yuzhe1129, 2022-05-25:
> 应该是Breed和flash芯片品牌的修改，修改了下面的环境变量参数，可以根据布局文件链接，直接编译使用 [openwrt/openwrt#9442（comment）的问题固件](https://github.com/openwrt/openwrt/issues/9442#issuecomment-1073452013)
试过了，修改环境变量也是不行

yuzhe1129, 2022-05-25:
> 应该是Breed和flash芯片品牌兼容的问题，参照下面的链接修改breed的环境变量参数，即可不用修改内存布局文件，直接使用编译出来的固件了 [openwrt/openwrt#9442 (comment)](https://github.com/openwrt/openwrt/issues/9442#issuecomment-1073452013)

我换了华邦闪存还是不行啊，也修改了环境变量

zys91, 2022-05-26:
> > 应该是Breed和flash芯片品牌兼容的问题，参照下面的链接修改breed的环境变量参数，即可不用修改内存布局文件，直接使用编译出来的固件了 [openwrt/openwrt#9442 (comment)](https://github.com/openwrt/openwrt/issues/9442#issuecomment-1073452013)
> 
> 我换了华邦闪存还是不行啊，也修改了环境变量

你去我仓库看一下吧，修改breed参数和不修改直刷的两种方式都详细说明了，具体可以参考scripts/diy-part1.sh和src修改dtsi文件，如果还不行那我也没办法了，因为我这边两种方式都试过是可以的
仓库地址：https://github.com/zys91/Actions-OpenWrt

coolsnowwolf, 2026-09-05:
小米 4A 千兆启动日志不完整，缺少分区表、bootloader 版本和完整串口输出；当前目标已有对应设备定义，无法据此确认镜像源码缺陷，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf