coolsnowwolf/lede #9301: HC5962 连接有线时，路由器就概率亮红灯，上不了网，过一会就恢复了

modelsun, 2022-04-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： 连接有线时，路由器就概率亮红灯，上不了网，过一会就恢复了

### (2) 路由器型号和固件版本

A：HC5962  7621A

版本  5.10.110内核   OpenWrt R22.4.1 

### (3) 详细日志

A：

modelsun, 2022-04-23:
Sat Apr 23 08:05:46 2022 kern.notice kernel: [    3.667067] ubi0: attaching mtd7
Sat Apr 23 08:05:46 2022 kern.info kernel: [    3.680583] mt7530 mdio-bus:1f: Link is Up - 1Gbps/Full - flow control off
Sat Apr 23 08:05:46 2022 kern.notice kernel: [    4.933561] ubi0: scanning is finished
Sat Apr 23 08:05:46 2022 kern.notice kernel: [    4.955018] ubi0: attached mtd7 (name "ubi", size 121 MiB)
Sat Apr 23 08:05:46 2022 kern.notice kernel: [    4.960550] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
Sat Apr 23 08:05:46 2022 kern.notice kernel: [    4.967392] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
Sat Apr 23 08:05:46 2022 kern.notice kernel: [    4.974173] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
Sat Apr 23 08:05:46 2022 kern.notice kernel: [    4.981119] ubi0: good PEBs: 974, bad PEBs: 0, corrupted PEBs: 0
Sat Apr 23 08:05:46 2022 kern.notice kernel: [    4.987098] ubi0: user volume: 2, internal volumes: 1, max. volumes count: 128
Sat Apr 23 08:05:46 2022 kern.notice kernel: [    4.994303] ubi0: max/mean erase counter: 19/15, WL threshold: 4096, image sequence number: 1646656169
Sat Apr 23 08:05:46 2022 kern.notice kernel: [    5.003582] ubi0: available PEBs: 0, total reserved PEBs: 974, PEBs reserved for bad PEB handling: 20
Sat Apr 23 08:05:46 2022 kern.notice kernel: [    5.012814] ubi0: background thread "ubi_bgt0d" started, PID 412
Sat Apr 23 08:05:46 2022 kern.info kernel: [    5.015236] block ubiblock0_0: created from ubi0:0(rootfs)
Sat Apr 23 08:05:46 2022 kern.notice kernel: [    5.024330] ubiblock: device ubiblock0_0 (rootfs) set to be root filesystem
Sat Apr 23 08:05:46 2022 kern.info kernel: [    5.041062] VFS: Mounted root (squashfs filesystem) readonly on device 254:0.
Sat Apr 23 08:05:46 2022 kern.info kernel: [    5.052398] Freeing unused kernel memory: 1260K
Sat Apr 23 08:05:46 2022 kern.warn kernel: [    5.056935] This architecture does not have kernel memory protection.
Sat Apr 23 08:05:46 2022 kern.info kernel: [    5.063388] Run /sbin/init as init process
Sat Apr 23 08:05:46 2022 kern.debug kernel: [    5.067463]   with arguments:
Sat Apr 23 08:05:46 2022 kern.debug kernel: [    5.067469]     /sbin/init
Sat Apr 23 08:05:46 2022 kern.debug kernel: [    5.067473]   with environment:
Sat Apr 23 08:05:46 2022 kern.debug kernel: [    5.067478]     HOME=/
Sat Apr 23 08:05:46 2022 kern.debug kernel: [    5.067483]     TERM=linux
Sat Apr 23 08:05:46 2022 user.info kernel: [    6.123225] init: Console is alive
Sat Apr 23 08:05:46 2022 user.info kernel: [    6.127024] init: - watchdog -
Sat Apr 23 08:05:46 2022 user.info kernel: [   10.327186] kmodloader: loading kernel modules from /etc/modules-boot.d/*
Sat Apr 23 08:05:46 2022 kern.notice kernel: [   10.806447] Key type encrypted registered
Sat Apr 23 08:05:46 2022 kern.info kernel: [   10.828780] usbcore: registered new interface driver usbfs
Sat Apr 23 08:05:46 2022 kern.info kernel: [   10.834529] usbcore: registered new interface driver hub
Sat Apr 23 08:05:46 2022 kern.info kernel: [   10.840027] usbcore: registered new device driver usb
Sat Apr 23 08:05:46 2022 kern.info kernel: [   10.856517] ANTFS Module: Version 07.19-b41ba529
Sat Apr 23 08:05:46 2022 kern.info kernel: [   10.865982] raid6: skip pq benchmark and using algorithm int32x8
Sat Apr 23 08:05:46 2022 kern.info kernel: [   10.872020] raid6: using intx1 recovery algorithm
Sat Apr 23 08:05:46 2022 kern.info kernel: [   10.877977] xor: measuring software checksum speed
Sat Apr 23 08:05:46 2022 kern.info kernel: [   10.895649]    8regs           :   767 MB/sec
Sat Apr 23 08:05:46 2022 kern.info kernel: [   10.914636]    8regs_prefetch  :   673 MB/sec
Sat Apr 23 08:05:46 2022 kern.info kernel: [   10.931837]    32regs          :   767 MB/sec
Sat Apr 23 08:05:46 2022 kern.info kernel: [   10.950856]    32regs_prefetch :   671 MB/sec
Sat Apr 23 08:05:46 2022 kern.info kernel: [   10.955192] xor: using function: 32regs (767 MB/sec)
Sat Apr 23 08:05:46 2022 kern.info kernel: [   11.001624] Btrfs loaded, crc32c=crc32c-generic
Sat Apr 23 08:05:46 2022 kern.info kernel: [   11.060530] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
Sat Apr 23 08:05:46 2022 kern.notice kernel: [   11.075048] SCSI subsystem initialized
Sat Apr 23 08:05:46 2022 kern.info kernel: [   11.083900] ehci-fsl: Freescale EHCI Host controller driver
Sat Apr 23 08:05:46 2022 kern.info kernel: [   11.091385] ehci-platform: EHCI generic platform driver
Sat Apr 23 08:05:46 2022 kern.info kernel: [   11.100440] ehci-pci: EHCI PCI platform driver
Sat Apr 23 08:05:46 2022 kern.info kernel: [   11.107803] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
Sat Apr 23 08:05:46 2022 kern.info kernel: [   11.115632] ohci-platform: OHCI generic platform driver
Sat Apr 23 08:05:46 2022 kern.info kernel: [   11.123560] uhci_hcd: USB Universal Host Controller Interface driver
Sat Apr 23 08:05:46 2022 kern.info kernel: [   11.131932] ohci-pci: OHCI PCI platform driver
Sat Apr 23 08:05:46 2022 kern.warn kernel: [   11.145430] xhci-mtk 1e1c0000.xhci: supply vbus not found, using dummy regulator
Sat Apr 23 08:05:46 2022 kern.warn kernel: [   11.153105] xhci-mtk 1e1c0000.xhci: supply vusb33 not found, using dummy regulator
Sat Apr 23 08:05:46 2022 kern.info kernel: [   11.160932] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
Sat Apr 23 08:05:46 2022 kern.info kernel: [   11.166262] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 1
[... cut ...]
Sat Apr 23 08:10:28 2022 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 8c:0d:76:70:ba:2e
Sat Apr 23 08:10:28 2022 daemon.info hostapd: wlan0: STA 8c:0d:76:70:ba:2e RADIUS: starting accounting session 3EB4B2B42318A563
Sat Apr 23 08:10:28 2022 daemon.info hostapd: wlan0: STA 8c:0d:76:70:ba:2e WPA: pairwise key handshake completed (RSN)
Sat Apr 23 08:10:28 2022 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 8c:0d:76:70:ba:2e
Sat Apr 23 08:10:34 2022 daemon.notice hostapd: wlan0: BSS-TM-RESP 8c:0d:76:70:ba:2e status_code=1 bss_termination_delay=0

xiangfeigao, 2022-04-27:
我都没编成功过

closed 2022-05-25 by modelsun