coolsnowwolf/lede #9303: 关于编译过程中cmake的Error1和Error2报错

klefjkl, 2022-04-24:
1.问题
最近编译一直卡在cmake相关步骤，最后error2，编译失败
Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：基于甲骨文云主机编译x86固件，之前没有任何问题，环境也没有发生变动，最近编译老失败，一直卡在cmake相关步骤，最后error2，编译失败。把lede目录全删除后，重新拉取，单线程编译也不行。

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

install -m0755 /home/ubuntu/lede/build_dir/host/ninja-1.10.2/ninja /home/ubuntu/lede/staging_dir/host/bin/
mkdir -p /home/ubuntu/lede/staging_dir/host/stamp
touch /home/ubuntu/lede/build_dir/host/ninja-1.10.2/.built
touch /home/ubuntu/lede/staging_dir/host/stamp/.ninja_installed
make[3]: Leaving directory '/home/ubuntu/lede/tools/ninja'
time: tools/ninja/compile#175.38#11.59#138.01
make[3]: Entering directory '/home/ubuntu/lede/tools/cmake'
. /home/ubuntu/lede/include/shell.sh; gzip -dc /home/ubuntu/lede/dl/cmake-3.22.2.tar.gz | tar -C /home/ubuntu/lede/build_dir/host/cmake-3.22.2/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/ubuntu/lede/build_dir/host/cmake-3.22.2

Applying ./patches/100-no-testing.patch using plaintext: 
patching file Modules/CTest.cmake
patching file Modules/Dart.cmake
patching file Tests/Contracts/VTK/Dashboard.cmake.in

Applying ./patches/120-curl-fix-libressl-linking.patch using plaintext: 
patching file Utilities/cmcurl/CMakeLists.txt

Applying ./patches/130-bootstrap_parallel_make_flag.patch using plaintext: 
patching file bootstrap
touch /home/ubuntu/lede/build_dir/host/cmake-3.22.2/.preparede06325a70c725905df6c02ddf9442e7c_6664517399ebbbc92a37c5bb081b5c53
(cd /home/ubuntu/lede/build_dir/host/cmake-3.22.2/; if [ -x configure ]; then cp -fpR /home/ubuntu/lede/scripts/config.{guess,sub} /home/ubuntu/lede/build_dir/host/cmake-3.22.2// && CC="gcc" CFLAGS="-O2 -I/home/ubuntu/lede/staging_dir/host/include " CXX="g++" CPPFLAGS="-I/home/ubuntu/lede/staging_dir/host/include " LDFLAGS="-L/home/ubuntu/lede/staging_dir/host/lib " CONFIG_SHELL="/usr/bin/env bash" CC="gcc" CXX="g++" MAKEFLAGS="" CXXFLAGS="-O2 -I/home/ubuntu/lede/staging_dir/host/include " MAKE="/home/ubuntu/lede/staging_dir/host/bin/ninja"  bash ./configure  --prefix="/home/ubuntu/lede/staging_dir/host" --generator=Ninja ; fi )
---------------------------------------------
CMake 3.22.2, Copyright 2000-2021 Kitware, Inc. and Contributors
C compiler on this system is: gcc -O2 -I/home/ubuntu/lede/staging_dir/host/include   
C++ compiler on this system is: g++ -O2 -I/home/ubuntu/lede/staging_dir/host/include    
Ninja processor on this system is: /home/ubuntu/lede/staging_dir/host/bin/ninja
g++ has setenv
g++ has unsetenv
g++ does not have environ in stdlib.h
g++ has stl wstring
g++ has <ext/stdio_filebuf.h>
---------------------------------------------
[1/268] g++ -O2 -I/home/ubuntu/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Bootstrap.cmk   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/LexerParser   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities/std   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities  -c /home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/cmGetTestPropertyCommand.cxx -o cmGetTestPropertyCommand.o
[2/268] g++ -O2 -I/home/ubuntu/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Bootstrap.cmk   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/LexerParser   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities/std   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities  -c /home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/cmSystemTools.cxx -o cmSystemTools.o
[3/268] g++ -O2 -I/home/ubuntu/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Bootstrap.cmk   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/LexerParser   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities/std   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities  -c /home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/cmExportTryCompileFileGenerator.cxx -o cmExportTryCompileFileGenerator.o
[4/268] g++ -O2 -I/home/ubuntu/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Bootstrap.cmk   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/LexerParser   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities/std   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities  -c /home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/cmContinueCommand.cxx -o cmContinueCommand.o
[5/268] g++ -O2 -I/home/ubuntu/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Bootstrap.cmk   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/LexerParser   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities/std   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities  -c /home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/cmBuildCommand.cxx -o cmBuildCommand.o
[6/268] g++ -O2 -I/home/ubuntu/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Bootstrap.cmk   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/LexerParser   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities/std   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities  -c /home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/cmBinUtilsMacOSMachOGetRuntimeDependenciesTool.cxx -o cmBinUtilsMacOSMachOGetRuntimeDependenciesTool.o
[7/268] g++ -O2 -I/home/ubuntu/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Bootstrap.cmk   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/LexerParser   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities/std   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities  -c /home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/cmInstallFilesGenerator.cxx -o cmInstallFilesGenerator.o
[8/268] g++ -O2 -I/home/ubuntu/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Bootstrap.cmk   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/LexerParser   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities/std   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities  -c /home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/cmGeneratorExpression.cxx -o cmGeneratorExpression.o
[11/268] g++ -O2 -I/home/ubuntu/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Bootstrap.cmk   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/LexerParser   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities/std   -I/home/ubuntu/lede/build_dir/host/cmake-3.22.2/Utilities  -c /home/ubuntu/lede/build_dir/host/cmake-3.22.2/Source/cmParseArgumentsCommand.cxx -o cmParseArgumentsCommand.o
ninja: build stopped: subcommand failed.
---------------------------------------------
Error when bootstrapping CMake:
Problem while running /home/ubuntu/lede/staging_dir/host/bin/ninja
---------------------------------------------
Log of errors: /home/ubuntu/lede/build_dir/host/cmake-3.22.2/Bootstrap.cmk/cmake_bootstrap.log
---------------------------------------------
make[3]: *** [Makefile:53: /home/ubuntu/lede/build_dir/host/cmake-3.22.2/.configured] Error 9
make[3]: Leaving directory '/home/ubuntu/lede/tools/cmake'
time: tools/cmake/compile#99.04#386.86#7205.67
    ERROR: tools/cmake failed to build.
[... 1064 lines cut ...]
    abi::__cxa_demangle(function, buffer, &bufferLen, &status);
  return status;
}
#endif

#ifdef TEST_KWSYS_STL_HAS_WSTRING
#  include <string>
void f(std::wstring*)
{
}
int main()
{
  return 0;
}
#endif

#ifdef TEST_KWSYS_CXX_HAS_EXT_STDIO_FILEBUF_H
#  include <ext/stdio_filebuf.h>
int main()
{
  return 0;
}
#endif
------------------------------------------
Test succeeded

WYC-2020, 2022-04-24:
这从来没遇到过，建议重新部署一下环境再试

klefjkl, 2022-04-24:
> 这从来没遇到过，建议重新部署一下环境再试

重新删除了2次都不行，打算重装下系统再试试，感谢解答！

closed 2022-04-25 by AmadeusGhost