coolsnowwolf/lede #9308: Openwrt使用p3terx的Actions Openwrt编译失败

KevinxYep, 2022-04-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我是用p3terx的actions进行编译的，地址是https://github.com/P3TERX/Actions-OpenWrt。只在luci中加了一个qos，别的都没动过。编译了两三次都失败了，都是在固件编译这里失败了。

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：Run cd openwrt
2 thread compile
 make[1] world
 make[2] tools/compile
 make[2] package/cleanup
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/tar compile
 make[3] -C tools/patch compile
 make[3] -C tools/m4 compile
 make[3] -C tools/autoconf-archive compile
 make[3] -C tools/pkgconf compile
 make[3] -C tools/ninja compile
 make[3] -C tools/cpio compile
 make[3] -C tools/lzma compile
 make[3] -C tools/meson compile
 make[3] -C tools/mtools compile
 make[3] -C tools/patch-image compile
 make[3] -C tools/sstrip compile
 make[3] -C tools/ucl compile
 make[3] -C tools/xxd compile
 make[3] -C tools/zip compile
 make[3] -C tools/expat compile
 make[3] -C tools/autoconf compile
 make[3] -C tools/libressl compile
 make[3] -C tools/upx compile
 make[3] -C tools/automake compile
 make[3] -C tools/missing-macros compile
 make[3] -C tools/dosfstools compile
 make[3] -C tools/libtool compile
 make[3] -C tools/flex compile
 make[3] -C tools/e2fsprogs compile
 make[3] -C tools/fakeroot compile
 make[3] -C tools/gengetopt compile
 make[3] -C tools/mkimage compile
 make[3] -C tools/mklibs compile
 make[3] -C tools/mm-macros compile
 make[3] -C tools/patchelf compile
 make[3] -C tools/gmp compile
 make[3] -C tools/bison compile
 make[3] -C tools/cmake compile
 make[3] -C tools/mpfr compile
 make[3] -C tools/bc compile
 make[3] -C tools/findutils compile
 make[3] -C tools/mpc compile
 make[3] -C tools/padjffs2 compile
 make[3] -C tools/quilt compile
 make[3] -C tools/zstd compile
 make[3] -C tools/zlib compile
 make[3] -C tools/firmware-utils compile
 make[3] -C tools/make-ext4fs compile
 make[3] -C tools/mtd-utils compile
 make[3] -C tools/squashfskit4 compile
 make[2] toolchain/compile
 make[3] -C toolchain/gdb compile
 make[3] -C toolchain/binutils compile
make: *** wait: No child processes.  Stop.
make: *** Waiting for unfinished jobs....
make: *** wait: No child processes.  Stop.
Error: Process completed with exit code 1[43](https://github.com/KevinxYep/OP/runs/6146150144?check_suite_focus=true#step:11:43).

WYC-2020, 2022-04-25:
你这个让人猜吗

KevinxYep, 2022-04-25:
是真的小白，不太懂咋办了，对不起

KevinxYep, 2022-04-25:
我只把compile the firmware的部分log截取出来了，上面下载、安装环境啥的都是没问题。
我以前就用actions来编译固件的，去年都没问题的，最近网络不稳定了，我才想着再来编译一个的，结果一直出错。

KevinxYep, 2022-04-25:
2022-04-24T11:22:58.4400800Z ##[group]Run cd openwrt
2022-04-24T11:22:58.4401123Z [36;1mcd openwrt[0m
2022-04-24T11:22:58.4401409Z [36;1mecho -e "$(nproc) thread compile"[0m
2022-04-24T11:22:58.4401735Z [36;1mmake -j$(nproc) || make -j1 || make -j1 V=s[0m
2022-04-24T11:22:58.4402054Z [36;1mecho "::set-output name=status::success"[0m
2022-04-24T11:22:58.4402466Z [36;1mgrep '^CONFIG_TARGET.*DEVICE.*=y' .config | sed -r 's/.*DEVICE_(.*)=y/\1/' > DEVICE_NAME[0m
2022-04-24T11:22:58.4402899Z [36;1m[ -s DEVICE_NAME ] && echo "DEVICE_NAME=_$(cat DEVICE_NAME)" >> $GITHUB_ENV[0m
2022-04-24T11:22:58.4403279Z [36;1mecho "FILE_DATE=_$(date +"%Y%m%d%H%M")" >> $GITHUB_ENV[0m
2022-04-24T11:22:58.4467155Z shell: /usr/bin/bash -e {0}
2022-04-24T11:22:58.4467385Z env:
2022-04-24T11:22:58.4467677Z   REPO_URL: https://github.com/coolsnowwolf/lede
2022-04-24T11:22:58.4467966Z   REPO_BRANCH: master
2022-04-24T11:22:58.4468205Z   FEEDS_CONF: feeds.conf.default
2022-04-24T11:22:58.4468457Z   CONFIG_FILE: .config
2022-04-24T11:22:58.4468762Z   DIY_P1_SH: diy-part1.sh
2022-04-24T11:22:58.4469000Z   DIY_P2_SH: diy-part2.sh
2022-04-24T11:22:58.4469241Z   UPLOAD_BIN_DIR: false
2022-04-24T11:22:58.4469486Z   UPLOAD_FIRMWARE: true
2022-04-24T11:22:58.4469719Z   UPLOAD_COWTRANSFER: false
2022-04-24T11:22:58.4469975Z   UPLOAD_WETRANSFER: false
2022-04-24T11:22:58.4470224Z   UPLOAD_RELEASE: false
2022-04-24T11:22:58.4470769Z   TZ: Asia/Shanghai
2022-04-24T11:22:58.4471009Z ##[endgroup]
2022-04-24T11:22:58.4590828Z 2 thread compile
2022-04-24T11:23:07.7889944Z  make[1] world
2022-04-24T11:23:10.2561422Z  make[2] tools/compile
2022-04-24T11:23:10.3011297Z  make[2] package/cleanup
2022-04-24T11:23:12.6385382Z  make[3] -C tools/flock compile
2022-04-24T11:23:12.7113542Z  make[3] -C tools/xz compile
2022-04-24T11:23:32.9133068Z  make[3] -C tools/sed compile
2022-04-24T11:24:03.2500624Z  make[3] -C tools/tar compile
2022-04-24T11:24:03.2512665Z  make[3] -C tools/patch compile
2022-04-24T11:24:49.0522784Z  make[3] -C tools/m4 compile
2022-04-24T11:24:49.0532242Z  make[3] -C tools/autoconf-archive compile
2022-04-24T11:24:50.9693987Z  make[3] -C tools/pkgconf compile
2022-04-24T11:25:01.4157662Z  make[3] -C tools/ninja compile
2022-04-24T11:25:43.1672189Z  make[3] -C tools/cpio compile
2022-04-24T11:25:47.5599631Z  make[3] -C tools/lzma compile
2022-04-24T11:25:54.5613945Z  make[3] -C tools/meson compile
2022-04-24T11:25:55.0730242Z  make[3] -C tools/mtools compile
2022-04-24T11:26:13.0548501Z  make[3] -C tools/patch-image compile
2022-04-24T11:26:13.2964307Z  make[3] -C tools/sstrip compile
2022-04-24T11:26:15.2101895Z  make[3] -C tools/ucl compile
2022-04-24T11:26:25.9341920Z  make[3] -C tools/xxd compile
2022-04-24T11:26:26.2724997Z  make[3] -C tools/zip compile
2022-04-24T11:26:30.0111003Z  make[3] -C tools/expat compile
2022-04-24T11:26:46.2006801Z  make[3] -C tools/autoconf compile
2022-04-24T11:26:48.6874284Z  make[3] -C tools/libressl compile
2022-04-24T11:26:49.7855002Z  make[3] -C tools/upx compile
2022-04-24T11:27:19.2943339Z  make[3] -C tools/automake compile
2022-04-24T11:27:25.1427306Z  make[3] -C tools/missing-macros compile
2022-04-24T11:27:25.2313287Z  make[3] -C tools/dosfstools compile
2022-04-24T11:27:33.5405861Z  make[3] -C tools/libtool compile
2022-04-24T11:29:13.0742258Z  make[3] -C tools/flex compile
2022-04-24T11:29:15.2441176Z  make[3] -C tools/e2fsprogs compile
2022-04-24T11:29:47.2823601Z  make[3] -C tools/fakeroot compile
2022-04-24T11:30:00.8038376Z  make[3] -C tools/gengetopt compile
2022-04-24T11:30:07.6514858Z  make[3] -C tools/mkimage compile
2022-04-24T11:30:26.1892988Z  make[3] -C tools/mklibs compile
2022-04-24T11:30:36.3201117Z  make[3] -C tools/mm-macros compile
2022-04-24T11:30:40.8586287Z  make[3] -C tools/patchelf compile
2022-04-24T11:30:53.5360865Z  make[3] -C tools/gmp compile
2022-04-24T11:31:42.5089154Z  make[3] -C tools/bison compile
2022-04-24T11:32:13.1686409Z  make[3] -C tools/cmake compile
2022-04-24T11:32:40.5761644Z  make[3] -C tools/mpfr compile
2022-04-24T11:36:13.7176477Z  make[3] -C tools/bc compile
2022-04-24T11:36:24.0232639Z  make[3] -C tools/findutils compile
2022-04-24T11:37:23.7578959Z  make[3] -C tools/mpc compile
2022-04-24T11:37:38.7594696Z  make[3] -C tools/padjffs2 compile
2022-04-24T11:37:38.9123093Z  make[3] -C tools/quilt compile
2022-04-24T11:45:34.2451282Z  make[3] -C tools/zstd compile
2022-04-24T11:45:34.2452069Z  make[3] -C tools/zlib compile
2022-04-24T11:45:40.9791170Z  make[3] -C tools/firmware-utils compile
2022-04-24T11:45:56.4142882Z  make[3] -C tools/make-ext4fs compile
2022-04-24T11:45:58.9341204Z  make[3] -C tools/mtd-utils compile
2022-04-24T11:46:22.6689579Z  make[3] -C tools/squashfskit4 compile
2022-04-24T11:46:29.4474326Z  make[2] toolchain/compile
2022-04-24T11:46:31.8777565Z  make[3] -C toolchain/gdb compile
2022-04-24T11:46:31.8820318Z  make[3] -C toolchain/binutils compile
2022-04-24T11:48:31.6671525Z make: *** wait: No child processes.  Stop.
2022-04-24T11:48:31.6715074Z make: *** Waiting for unfinished jobs....
2022-04-24T11:48:31.6715740Z make: *** wait: No child processes.  Stop.
2022-04-24T11:48:31.6764682Z ##[error]The runner has received a shutdown signal. This can happen when the runner service is stopped, or a manually started runner is canceled.
2022-04-24T11:48:31.6764746Z ##[error]Process completed with exit code 143.
2022-04-24T11:48:31.6976301Z Cleaning up orphan processes

mitsukileung, 2022-04-25:
是不是超过6个小时了？

EnnawYang, 2022-04-25:
盲猜用了ssh

closed 2022-04-25 by AmadeusGhost