coolsnowwolf/lede #9309: 借地方求助个问题，非bug

GreatMichaelLee, 2022-04-25:
feeds.conf.default里
我如何加入一个仓库的多个分支做源，但是放到同一个目录下？
比如src-git passwall https://github.com/xiaorouji/openwrt-passwall.git 这个只拉了package的，luci的branch我手工clone到/feeds/passwall下
那luci的branch如何也同步能通过./scripts/feed update -a来一起更新？  src-git能传-b参数吗

WYC-2020, 2022-04-25:
想什么呢，去学学git再说

GreatMichaelLee, 2022-04-25:
> 想什么呢，去学学git再说

想说的是src-git 如何指定我想要的branch作为当前跟踪分支

WYC-2020, 2022-04-25:
> > 想什么呢，去学学git再说
> 
> 想说的是src-git 如何指定我想要的branch作为当前跟踪分支，是要在脚本里弄吗

这玩意不是以前有人问过嘛，稍微搜索一下就可以的 ，真懒 https://github.com/coolsnowwolf/lede/issues/9095

GreatMichaelLee, 2022-04-25:
> > > 想什么呢，去学学git再说
> > 
> > 
> > 想说的是src-git 如何指定我想要的branch作为当前跟踪分支，是要在脚本里弄吗
> 
> 这玩意不是以前有人问过嘛，稍微搜索一下就可以的 ，真懒 #9095

哈哈，懒是人的天性，彼此彼此:P

WYC-2020, 2022-04-25:
迟早这个提交bug的模块被关闭，让你们提不了

GreatMichaelLee, 2022-04-25:
> 迟早这个提交bug的模块被关闭，让你们提不了

src-git name url;branches  貌似这个方法报语法错误...

Beginner-Go, 2022-04-25:
name不一样就行吧

GreatMichaelLee, 2022-04-25:
> name不一样就行吧

没懂，我知道name要不一样，我的需求是，比如Passwall这个仓库，他有package和luci两个branch，我想做成
src-git package <Repo URL> --> 分支package 
src-git luci <Repo URL> --> 分支luci
但是第一次克隆的时候这只是拉了两个不同的目录，分支都是仓库上 package分支的啊，我如何在这个配置文件里能分别制定不同的branch, 上面那个ticket提到是`src-git name url;branches  ` 这个语法不对啊，./scripts/feed 脚本提示语法错误

wykdg, 2022-04-25:
但凡你看一眼这个仓库的feesds.config.default，就是这样取分支的
src-git packages https://github.com/coolsnowwolf/packages;openwrt-19.07
src-git luci https://github.com/coolsnowwolf/luci
src-git routing https://git.openwrt.org/feed/routing.git;openwrt-19.07
#src-git telephony https://git.openwrt.org/feed/telephony.git;openwrt-19.07
src-git helloworld https://github.com/fw876/helloworld

GreatMichaelLee, 2022-04-25:
> 但凡你看一眼这个仓库的feesds.config.default，就是这样取分支的 src-git packages https://github.com/coolsnowwolf/packages;openwrt-19.07 src-git luci https://github.com/coolsnowwolf/luci src-git routing [...]

多谢，终于弄明白了问题在哪里了，不是分号后加branch有语法问题，而是我name里面包含了破折号，而name的解析居然不支持破折号，只能用下划线，哭了。。。

Boos4721, 2022-04-25:
> > 但凡你看一眼这个仓库的feesds.config.default，就是这样取分支的 src-git packages [coolsnowwolf/packages;openwrt-19.07](https://github.com/coolsnowwolf/packages;openwrt-19.07) src-git luci [coolsnowwolf/luci](https://github.com/coolsnowwolf/luci) src-git routing [git.openwrt.org/feed/routing.git;openwrt-19.07](https://git.openwrt.org/feed/routing.git;openwrt-19.07) #src-git telephony [git.openwrt.org/feed/telephony.git;openwrt-19.07](https://git.openwrt.org/feed/telephony.git;openwrt-19.07) src-git helloworld [fw876/helloworld](https://github.com/fw876/helloworld)
> 
> 多谢，终于弄明白了问题在哪里了，不是分号后加branch有语法问题，而是我name里面包含了破折号，而name的解析居然不支持破折号，只能用下划线，哭了。。。

lmao

closed 2022-04-25 by AmadeusGhost