coolsnowwolf/lede #9313: cortex-a7架构于liboil处失败

ltrsunny, 2022-04-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： gstreamer相关的包在makefile中会调用liboil，如果在config中没有选中liboil的话，编译开始前手动删除dl目录中的liboil包可顺利通过；但部分选择的包依赖于liboil时，似乎不能通过这种手段跳过。

### (2) 路由器型号和固件版本

A：MT7629和MT7623测试均失败。

### (3) 详细日志

A：
```
Making all in arm
make[7]: Entering directory '/media/dev/fedora_localhost-live1/home/dev/lede/build_dir/target-arm_cortex-a7_glibc_eabi/liboil-0.3.17/liboil/arm'
/bin/bash ../../libtool    --mode=compile arm-openwrt-linux-gnueabi-gcc -DHAVE_CONFIG_H -I. -I../..   -I/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/toolchain-arm_cortex-a7_gcc-11.2.0_glibc_eabi/usr/include -I/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/toolchain-arm_cortex-a7_gcc-11.2.0_glibc_eabi/include -I/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/target-arm_cortex-a7_glibc_eabi/usr/lib/libiconv-stub/include -I/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/target-arm_cortex-a7_glibc_eabi/usr/lib/libintl-stub/include   -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/media/dev/fedora_localhost-live1/home/dev/lede/build_dir/target-arm_cortex-a7_glibc_eabi/liboil-0.3.17=liboil-0.3.17 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/target-arm_cortex-a7_glibc_eabi/usr/lib/libiconv-stub/include -I/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/target-arm_cortex-a7_glibc_eabi/usr/lib/libintl-stub/include -DPIC -fpic    -MT math_vfp_asm.lo -MD -MP -MF .deps/math_vfp_asm.Tpo -c -o math_vfp_asm.lo math_vfp_asm.S
OpenWrt-libtool: compile:  arm-openwrt-linux-gnueabi-gcc -DHAVE_CONFIG_H -I. -I../.. -I/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/toolchain-arm_cortex-a7_gcc-11.2.0_glibc_eabi/usr/include -I/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/toolchain-arm_cortex-a7_gcc-11.2.0_glibc_eabi/include -I/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/target-arm_cortex-a7_glibc_eabi/usr/lib/libiconv-stub/include -I/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/target-arm_cortex-a7_glibc_eabi/usr/lib/libintl-stub/include -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/media/dev/fedora_localhost-live1/home/dev/lede/build_dir/target-arm_cortex-a7_glibc_eabi/liboil-0.3.17=liboil-0.3.17 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/target-arm_cortex-a7_glibc_eabi/usr/lib/libiconv-stub/include -I/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/target-arm_cortex-a7_glibc_eabi/usr/lib/libintl-stub/include -DPIC -fpic -MT math_vfp_asm.lo -MD -MP -MF .deps/math_vfp_asm.Tpo -c math_vfp_asm.S  -fPIC -DPIC -o .libs/math_vfp_asm.o
math_vfp_asm.S: Assembler messages:
math_vfp_asm.S:107: Error: selected processor does not support `fldmias r1!,{s0}' in ARM mode
math_vfp_asm.S:107: Error: selected processor does not support `fldmias r2!,{s1}' in ARM mode
math_vfp_asm.S:107: Error: selected processor does not support `fadds s2,s0,s1' in ARM mode
math_vfp_asm.S:107: Error: selected processor does not support `fstmias r0!,{s2}' in ARM mode
math_vfp_asm.S:107: Error: selected processor does not support `fmrx lr,fpscr' in ARM mode
math_vfp_asm.S:107: Error: selected processor does not support `fmxr fpscr,fp' in ARM mode
math_vfp_asm.S:107: Error: selected processor does not support `fldmias r1!,{s8,s9,s10,s11,s12,s13,s14,s15}' in ARM mode
math_vfp_asm.S:107: Error: selected processor does not support `fldmias r2!,{s16,s17,s18,s19,s20,s21,s22,s23}' in ARM mode
math_vfp_asm.S:107: Error: selected processor does not support `fadds s24,s8,s16' in ARM mode
math_vfp_asm.S:107: Error: selected processor does not support `fstmias r0!,{s24,s25,s26,s27,s28,s29,s30,s31}' in ARM mode
math_vfp_asm.S:107: Error: selected processor does not support `fmxr fpscr,lr' in ARM mode
math_vfp_asm.S:108: Error: selected processor does not support `fldmiad r1!,{d0}' in ARM mode
math_vfp_asm.S:108: Error: selected processor does not support `fldmiad r2!,{d1}' in ARM mode
math_vfp_asm.S:108: Error: selected processor does not support `faddd d2,d0,d1' in ARM mode
math_vfp_asm.S:108: Error: selected processor does not support `fstmiad r0!,{d2}' in ARM mode
math_vfp_asm.S:108: Error: selected processor does not support `fmrx lr,fpscr' in ARM mode
math_vfp_asm.S:108: Error: selected processor does not support `fmxr fpscr,fp' in ARM mode
math_vfp_asm.S:108: Error: selected processor does not support `fldmiad r1!,{d4,d5,d6,d7}' in ARM mode
math_vfp_asm.S:108: Error: selected processor does not support `fldmiad r2!,{d8,d9,d10,d11}' in ARM mode
math_vfp_asm.S:108: Error: selected processor does not support `faddd d12,d4,d8' in ARM mode
math_vfp_asm.S:108: Error: selected processor does not support `fstmiad r0!,{d12,d13,d14,d15}' in ARM mode
math_vfp_asm.S:108: Error: selected processor does not support `fmxr fpscr,lr' in ARM mode
math_vfp_asm.S:110: Error: selected processor does not support `fldmias r1!,{s0}' in ARM mode
math_vfp_asm.S:110: Error: selected processor does not support `fldmias r2!,{s1}' in ARM mode
math_vfp_asm.S:110: Error: selected processor does not support `fdivs s2,s0,s1' in ARM mode
math_vfp_asm.S:110: Error: selected processor does not support `fstmias r0!,{s2}' in ARM mode
math_vfp_asm.S:110: Error: selected processor does not support `fmrx lr,fpscr' in ARM mode
math_vfp_asm.S:110: Error: selected processor does not support `fmxr fpscr,fp' in ARM mode
math_vfp_asm.S:110: Error: selected processor does not support `fldmias r1!,{s8,s9,s10,s11,s12,s13,s14,s15}' in ARM mode
math_vfp_asm.S:110: Error: selected processor does not support `fldmias r2!,{s16,s17,s18,s19,s20,s21,s22,s23}' in ARM mode
math_vfp_asm.S:110: Error: selected processor does not support `fdivs s24,s8,s16' in ARM mode
math_vfp_asm.S:110: Error: selected processor does not support `fstmias r0!,{s24,s25,s26,s27,s28,s29,s30,s31}' in ARM mode
math_vfp_asm.S:110: Error: selected processor does not support `fmxr fpscr,lr' in ARM mode
math_vfp_asm.S:111: Error: selected processor does not support `fldmiad r1!,{d0}' in ARM mode
math_vfp_asm.S:111: Error: selected processor does not support `fldmiad r2!,{d1}' in ARM mode
math_vfp_asm.S:111: Error: selected processor does not support `fdivd d2,d0,d1' in ARM mode
math_vfp_asm.S:111: Error: selected processor does not support `fstmiad r0!,{d2}' in ARM mode
math_vfp_asm.S:111: Error: selected processor does not support `fmrx lr,fpscr' in ARM mode
math_vfp_asm.S:111: Error: selected processor does not support `fmxr fpscr,fp' in ARM mode
math_vfp_asm.S:111: Error: selected processor does not support `fldmiad r1!,{d4,d5,d6,d7}' in ARM mode
math_vfp_asm.S:111: Error: selected processor does not support `fldmiad r2!,{d8,d9,d10,d11}' in ARM mode
math_vfp_asm.S:111: Error: selected processor does not support `fdivd d12,d4,d8' in ARM mode
math_vfp_asm.S:111: Error: selected processor does not support `fstmiad r0!,{d12,d13,d14,d15}' in ARM mode
math_vfp_asm.S:111: Error: selected processor does not support `fmxr fpscr,lr' in ARM mode
[... 113 lines cut ...]
math_vfp_asm.S:273: Error: selected processor does not support `fstmiad r0!,{d2}' in ARM mode
math_vfp_asm.S:273: Error: selected processor does not support `fmrx lr,fpscr' in ARM mode
math_vfp_asm.S:273: Error: selected processor does not support `fmxr fpscr,fp' in ARM mode
math_vfp_asm.S:273: Error: selected processor does not support `fldmiad r1!,{d4,d5,d6,d7}' in ARM mode
math_vfp_asm.S:273: Error: selected processor does not support `fnegd d12,d4' in ARM mode
math_vfp_asm.S:273: Error: selected processor does not support `fstmiad r0!,{d12,d13,d14,d15}' in ARM mode
math_vfp_asm.S:273: Error: selected processor does not support `fmxr fpscr,lr' in ARM mode
make[7]: *** [Makefile:415: math_vfp_asm.lo] Error 1
make[7]: Leaving directory '/media/dev/fedora_localhost-live1/home/dev/lede/build_dir/target-arm_cortex-a7_glibc_eabi/liboil-0.3.17/liboil/arm'
make[6]: *** [Makefile:1113: all-recursive] Error 1
make[6]: Leaving directory '/media/dev/fedora_localhost-live1/home/dev/lede/build_dir/target-arm_cortex-a7_glibc_eabi/liboil-0.3.17/liboil'
[... cut ...]
make[2]: *** [package/Makefile:116: package/feeds/packages/liboil/compile] Error 1
make[2]: Leaving directory '/media/dev/fedora_localhost-live1/home/dev/lede'
make[1]: *** [package/Makefile:110: /media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/target-arm_cortex-a7_glibc_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/dev/fedora_localhost-live1/home/dev/lede'
make: *** [/media/dev/fedora_localhost-live1/home/dev/lede/include/toplevel.mk:230: world] Error 2
```

crmmc, 2022-04-25:
我同样遇到了这个问题，想着pull新源码在新虚拟机里编译一套新固件给RPI 4B，但是出现了这个奇怪的错误
google了有人说是因为ubuntu的binutils组建更新导致对arm asm的一些规则的改变，导致不能识别以前的汇编代码格式，我测试了更新binutils到最新（源码编译）也无法解决这个问题，现在唯一的方法就是用旧的Ubuntu环境（带之前的binutils 版本 < 2.23的和稍老GCC8的）来编译，否则必定出错。
我暂时只发现在arm32架构出现这个问题，X86没问题，出现这个问题的包是wolfssl
我有一套之前编译MT300NV2的ub18的虚拟机环境，现在恢复快照后成功编译最新LEDE源码了。
只能由开发者改变这些源码的结构才可能修复，而这样写的源码又一般都是重要的库，感觉这将导致arm32很长一段时间内编译异常。

WYC-2020, 2022-04-26:
你们说的gstreamer我到没看到，我只看到gstreamer1，并且makefile 也不会去依赖那个，不知说的什么鬼

ltrsunny, 2022-04-26:
> 我同样遇到了这个问题，想着pull新源码在新虚拟机里编译一套新固件给RPI 4B，但是出现了这个奇怪的错误 google了有人说是因为ubuntu的binutils组建更新导致对arm asm的一些规则的改变，导致不能识别以前的汇编代码格式，我测试了更新binutils到最新（源码编译）也无法解决这个问题，现在唯一的方法就是用旧的Ubuntu环境（带之前的binutils 版本 < [...]

我用fedora 35跟 Ubuntu 20 & 22分别测试过

ltrsunny, 2022-04-26:
> > 
> > 我用fedora 35跟 Ubuntu 20 & 22分别测试过
> 
> 请问你没有遇到问题吗？ 我编译openwrt for RPI4B 32bit的时候默认选中包wpad-wolfssl会导致此问题，因为交叉编译链的ld版本升级到了2.34，更换会旧版的交叉编译ld可以解决编译总在此失败的问题

分别测试全部失败。😂

WYC-2020, 2022-04-26:
亲测没任何问题，自己排查了，我用的我自己仓库测试的
[image]

ltrsunny, 2022-04-26:
> 你们说的gstreamer我到没看到，我只看到gstreamer1，
PACKAGE_gst1-libav
PACKAGE_gstreamer1-libs
PACKAGE_gstreamer1-plugins-bad
PACKAGE_gstreamer1-plugins-base
PACKAGE_gstreamer1-plugins-good
PACKAGE_gstreamer1-plugins-ugly
PACKAGE_gstreamer1-utils
PACKAGE_gmediarender

WYC-2020, 2022-04-26:
[image]

> > 你们说的gstreamer我到没看到，我只看到gstreamer1，
> 
> PACKAGE_gst1-libav PACKAGE_gstreamer1-libs PACKAGE_gstreamer1-plugins-bad PACKAGE_gstreamer1-plugins-base PACKAGE_gstreamer1-plugins-good PACKAGE_gstreamer1-plugins-ugly PACKAGE_gstreamer1-utils PACKAGE_gmediarender

WYC-2020, 2022-04-26:
直接发.config 给我。或者告诉我怎么型号的选择步骤，我是吧gstreamer1 全部都选择了

> > 你们说的gstreamer我到没看到，我只看到gstreamer1，
> 
> PACKAGE_gst1-libav PACKAGE_gstreamer1-libs PACKAGE_gstreamer1-plugins-bad PACKAGE_gstreamer1-plugins-base PACKAGE_gstreamer1-plugins-good PACKAGE_gstreamer1-plugins-ugly PACKAGE_gstreamer1-utils PACKAGE_gmediarender

ltrsunny, 2022-04-26:
> 直接发.config 给我。或者告诉我怎么型号的选择步骤，我是吧gstreamer1 全部都选择了
[config.txt](https://github.com/coolsnowwolf/lede/files/8559264/config.txt)

WYC-2020, 2022-04-26:
> 直接发.config 给我。或者告诉我怎么型号的选择步骤，我是吧gstreamer1 全部都选择了
> 你们说的gstreamer我到没看到，我只看到gstreamer1，
>
>
> PACKAGE_gst1-libav PACKAGE_gstreamer1-libs PACKAGE_gstreamer1-plugins-bad PACKAGE_gstreamer1-plugins-base [...]

你确定你这个是对的config ，我看都没有选择gstreamer1嘛，liboil也没选择编译

ltrsunny, 2022-04-26:
> 你确定你这个是对的config ，我看都没有选择gstreamer1嘛，liboil也没选择编译

[image]
[image]
都有啊。

crmmc, 2022-04-26:
> 亲测没任何问题，自己排查了，我用的我自己仓库测试的 [image]

请说明一下你的编译环境和系统版本啥的
顺便看看你的armv7交叉编译工具链里ld的版本吧
例如：/home/ub18/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld -v
我曾在尝试切换binutils到2.32

WYC-2020, 2022-04-26:
找到原因为啥不用musl非要去用glibc 用这个玩意就会失败，改用musl 就不会出问题

ltrsunny, 2022-04-26:
> 找到原因为啥不用musl非要去用glibc 用这个玩意就会失败，改用musl 就不会出问题

我一开始使用的是musl，一样失败。
改用glibc是为一些其他后续使用软件的兼容性考虑。

WYC-2020, 2022-04-26:
> > 找到原因为啥不用musl非要去用glibc 用这个玩意就会失败，改用musl 就不会出问题
> 
> 我一开始使用的是musl，一样失败。 改用glibc是为一些其他后续使用软件的兼容性考虑。

musl 我试过没任何问题，这个你自己找原因，glibc 我这边马上修好，等我验证一下

WYC-2020, 2022-04-26:
> > 找到原因为啥不用musl非要去用glibc 用这个玩意就会失败，改用musl 就不会出问题
> 
> 我一开始使用的是musl，一样失败。 改用glibc是为一些其他后续使用软件的兼容性考虑。

看仓库 packages dev 分支，自己去测试，功能正常的话记得说一声我好合并到我的master分支

ltrsunny, 2022-04-26:
> > > 找到原因为啥不用musl非要去用glibc 用这个玩意就会失败，改用musl 就不会出问题
> > 
> > 
> > 我一开始使用的是musl，一样失败。 改用glibc是为一些其他后续使用软件的兼容性考虑。
> 
> 看仓库 packages dev 分支，自己去测试，功能正常的话记得说一声我好合并到我的master分支

好的，谢谢，我测一下。

ltrsunny, 2022-04-26:
> > > > 找到原因为啥不用musl非要去用glibc 用这个玩意就会失败，改用musl 就不会出问题
> > > 
> > > 
> > > 我一开始使用的是musl，一样失败。 改用glibc是为一些其他后续使用软件的兼容性考虑。
> > 
> > 
> > 看仓库 packages dev 分支，自己去测试，功能正常的话记得说一声我好合并到我的master分支
> 
> 好的，谢谢，我测一下。

是这个吗？https://github.com/WYC-2020/packages/tree/dev

crmmc, 2022-04-26:
能研究一下官方源码与lede的差别吗？我拉取了
https://github.com/Lienol/openwrt
的源码，用了同一个的配置文件，添加了额外feed源
echo 'src-git small https://github.com/kenzok8/small' >>feeds.conf.default
echo 'src-git kenzo https://github.com/kenzok8/openwrt-packages' >>feeds.conf.default
也就是和这个一模一样：（我的.config也放在了上面）
https://github.com/crmmc/AutoOpwnert
我在本地用这一套配置成功编译了，wolfssl和binutils都和lede最新版本拉下来的版本一样，但是很奇怪没有像编译lede时那样报错，我现在正在测试github action编译，如果这也成功，那就可能是lede源码有问题了。

ltrsunny, 2022-04-26:
> 用了同一个的配置文件

是指当前lede的feed源吗？

crmmc, 2022-04-26:
> 
源码来源：https://github.com/Lienol/openwrt
我使用的feed源是：https://github.com/Lienol/openwrt自带 + https://github.com/kenzok8/small + https://github.com/kenzok8/openwrt-packages
我使用的配置文件：https://github.com/crmmc/AutoOpwnert的.config
现在情况：github action自动编译完成，无错误
感觉是lede源码可能存在错误

WYC-2020, 2022-04-26:
> wpad-wolfssl

对比那是不可能的，懒得搞，关闭他的asm不就可以了，能提升多少效率嘛

WYC-2020, 2022-04-26:
[image]

WYC-2020, 2022-04-26:
> 源码来源：https://github.com/Lienol/openwrt 我使用的feed源是：[https://github.com/Lienol/openwrt自带](https://github.com/Lienol/openwrt%E8%87%AA%E5%B8%A6) + https://github.com/kenzok8/small + [...]

看仓库，lienol  本身也没开这个asm

ltrsunny, 2022-04-27:
> > > 找到原因为啥不用musl非要去用glibc 用这个玩意就会失败，改用musl 就不会出问题
> > 
> > 
> > 我一开始使用的是musl，一样失败。 改用glibc是为一些其他后续使用软件的兼容性考虑。
> 
> 看仓库 packages dev 分支，自己去测试，功能正常的话记得说一声我好合并到我的master分支

出错点变了
```
17 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/dev/fedora_localhost-live1/home/dev/ledrm_cortex-a7_glibc_eabi/usr/lib/libintl-stub/include -DPIC -fpic    -MT uberopt-uberopt.o -MD -MP -MF .deps/uberopt-uberopt.Tpo -c -o uberopt-uberopt.o `test -f 'uberopt.c' || echo './'`uberopt.c
In file included from /media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/toolchain-arm_cortex-a7_gcc-11.2.0_glibc_eabi/include/bits/libc-header-start.h:33,
                 from /media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/toolchain-arm_cortex-a7_gcc-11.2.0_glibc_eabi/include/stdint.h:26,
                 from /media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/toolchain-arm_cortex-a7_gcc-11.2.0_glibc_eabi/lib/gcc/arm-openwrt-linux-gnueabi/11.2.0/include/stdint.h:9,
                 from ../../liboil/liboil-stdint.h:7,
                 from ../../liboil/liboilutils.h:31,
                 from ../../liboil/liboil.h:31,
                 from uberopt.c:32:
/media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/toolchain-arm_cortex-a7_gcc-11.2.0_glibc_eabi/include/features.h:194:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
  194 | # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
      |   ^~~~~~~
uberopt.c:34:10: fatal error: glib.h: No such file or directory
   34 | #include <glib.h>
      |          ^~~~~~~~
compilation terminated.
make[7]: *** [Makefile:410: uberopt-uberopt.o] Error 1
make[7]: Leaving directory '/media/dev/fedora_localhost-live1/home/dev/lede/build_dir/target-arm_cortex-a7_glibc_eabi/liboil-0.3.17/examples/uberopt'
make[6]: *** [Makefile:820: all-recursive] Error 1
make[6]: Leaving directory '/media/dev/fedora_localhost-live1/home/dev/lede/build_dir/target-arm_cortex-a7_glibc_eabi/liboil-0.3.17/examples'
make[5]: *** [Makefile:492: all-recursive] Error 1
make[5]: Leaving directory '/media/dev/fedora_localhost-live1/home/dev/lede/build_dir/target-arm_cortex-a7_glibc_eabi/liboil-0.3.17'
make[4]: *** [Makefile:399: all] Error 2
make[4]: Leaving directory '/media/dev/fedora_localhost-live1/home/dev/lede/build_dir/target-arm_cortex-a7_glibc_eabi/liboil-0.3.17'
make[3]: *** [Makefile:73: /media/dev/fedora_localhost-live1/home/dev/lede/build_dir/target-arm_cortex-a7_glibc_eabi/liboil-0.3.17/.built] Error 2
make[3]: Leaving directory '/media/dev/fedora_localhost-live1/home/dev/lede/feeds/packages/libs/liboil'
time: package/feeds/packages/liboil/compile#0.34#0.09#0.40
    ERROR: package/feeds/packages/liboil failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/liboil/compile] Error 1
make[2]: Leaving directory '/media/dev/fedora_localhost-live1/home/dev/lede'
make[1]: *** [package/Makefile:110: /media/dev/fedora_localhost-live1/home/dev/lede/staging_dir/target-arm_cortex-a7_glibc_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/dev/fedora_localhost-live1/home/dev/lede'
make: *** [/media/dev/fedora_localhost-live1/home/dev/lede/include/toplevel.mk:230??world] ???? 2
```

WYC-2020, 2022-04-27:
> 出错点变了
>
> ```
> 17 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro [...]

我没遇到过，这个自己排查

coolsnowwolf, 2026-09-05:
liboil 失败日志来自旧 ARM/cortex-a7 工具链和汇编路径，缺少完整首个错误；当前上游包未提供针对该旧组合的修复，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf