coolsnowwolf/lede #9315: containerd-1.5.11 hash value error

GreatMichaelLee, 2022-04-26:
Makefile里Hash值不对，导致编译出错

[image]

[image]

breakings, 2022-04-26:
这值这两天有变动,自动手动改下也不难吧

WYC-2020, 2022-04-26:
这种鬼玩意，搞毛了直接跳过hash

GreatMichaelLee, 2022-04-26:
> 这种鬼玩意，搞毛了直接跳过hash

赞

Wilson-coder, 2022-04-27:
> 这值这两天有变动,自动手动改下也不难吧

Report Bug，是开源社区赖以生存的基础。如果每个人遇到了问题都是自己默默修改代码自己用，或者单纯依赖开发者自己发现Bug，那就不用搞开源社区了。

GreatMichaelLee, 2022-04-27:
> > 这值这两天有变动,自动手动改下也不难吧
> 
> Report Bug，是开源社区赖以生存的基础。如果每个人遇到了问题都是自己默默修改代码自己用，或者单纯依赖开发者自己发现Bug，那就不用搞开源社区了。

坐等一堆大佬来喷你，哈哈。估计他们又会说，你贡献了多少代码，你行你上之类的，所以我作为伸手党，大佬损我时我都只点赞的，哈哈哈。

hong0980, 2022-04-28:
sed -i "s/\(PKG_HASH\|PKG_MD5SUM\|PKG_MIRROR_HASH\).*/\1:=skip/" feeds/packages/utils/containerd/Makefile

closed 2023-04-25 by aiamadeus