coolsnowwolf/lede #9317: 4a-gigabit不启动，反复重启。

yuzhe1129, 2022-04-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
4a-gigabit  小米4A千兆版， 4.26更新源码了还是不启动，可以维护修复下吗
A：

### (3) 详细日志

Boot and Recovery Environment for Embedded Devices
Copyright (C) 2021 HackPascal <hackpascal@gmail.com>
Build date 2021-12-15 [git-f9b74d0]
Version 1.1 (r1337)

DRAM: 128MB
Platform: MediaTek MT7621A ver 1, eco 3
Board: PandoraBox PBR-M1 / Newifi Y2S
Clocks: CPU: 880MHz, DDR: 1040MHz, Bus: 220MHz, Ref: 40MHz
Flash: GigaDevice GD25Q128 (16MB) on mt7621-spi.0
rt2880-eth: MAC address from EEPROM is invalid, using default settings.
rt2880-eth: Using MAC address 00:0c:43:00:00:01
eth0: MediaTek MT7530 Gigabit switch

Network started on eth0, inet addr 192.168.1.1, netmask 255.255.255.0

Press any key to interrupt autoboot ... 0

Trying to boot firmware from 0x00050000 in flash bank 0 ...
Reading data into memory ...
U-Boot firmware image header detected.
    Image Name:   MIPS OpenWrt Linux-5.4.190
    Data Size:    2538146 Bytes
    Load Address: 80001000
    Entry Point:  80001000
Flushing cache ... done.

Starting kernel at 0x80001000...

OpenWrt kernel loader for MIPS based SoC
Copyright (C) 2011 Gabor Juhos <juhosg@openwrt.org>
Decompressing kernel... done!
Starting kernel at 80001000...

[    0.000000] Linux version 5.4.190 (bbb@ubuntu) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4479-2a91c7245)) #0 SMP Mon Apr 25 11:18:22 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 4A Gigabit Edition
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[... 176 lines cut ...]
[    1.786974]  (driver?)
[    1.793493] 1f06            1024 mtdblock6
[    1.793496]  (driver?)
[    1.799996] 1f07           14848 mtdblock7
[    1.799999]  (driver?)
[    1.806516] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    1.814780] Rebooting in 1 seconds..

Boot and Recovery Environment for Embedded Devices
Copyright (C) 2021 HackPascal <hackpascal@gmail.com>
Build date 2021-12-15 [git-f9b74d0]
Version 1.1 (r1337)

DRAM: 128MB
Platform: MediaTek MT7621A ver 1, eco 3
Board: PandoraBox PBR-M1 / Newifi Y2S
Clocks: CPU: 880MHz, DDR: 1040MHz, Bus: 220MHz, Ref: 40MHz
Flash: GigaDevice GD25Q128 (16MB) on mt7621-spi.0
rt2880-eth: MAC address from EEPROM is invalid, using default settings.
rt2880-eth: Using MAC address 00:0c:43:00:00:01
eth0: MediaTek MT7530 Gigabit switch

Network started on eth0, inet addr 192.168.1.1, netmask 255.255.255.0

Press any key to interrupt autoboot ... 2▒▒▒▒▒▒▒▒▒p▒▒

WYC-2020, 2022-04-26:
你是提起好玩吗？开了一个贴还开，有人不是给你回复了咋搞，自己搞嘛

Plutonium141, 2022-05-16:
https://github.com/Plutonium141/XiaoMi-R4A-Gigabit-Actions-OpenWrt
也许可以解决你的问题

chobitsos, 2022-05-16:
> 你是提起好玩吗？开了一个贴还开，有人不是给你回复了咋搞，自己搞嘛

> https://github.com/Plutonium141/XiaoMi-R4A-Gigabit-Actions-OpenWrt 也许可以解决你的问题

> > 你提起了很好玩吗？发现一个贴子还开着，有人不是给你回复了咋搞，自己搞嘛
> 
> 21号lean更新了，我这再编译还是不成功，这不是反馈BUG吗？如果更新了源码还是不启动不能反馈，那lean还更新源码做什么？干脆删仓库跑路算了

他这个问题目前无解，应该是新的一批R4A，用的闪存openwrt不识别，但是毛子固件没问题
新批次的R4A全部都是用 Eon EN25QX128 16MB的闪存，所以openwrt还没适配dts，所以无限重启，只能等官方适配了自己改下DTS就行了

coolsnowwolf, 2026-09-05:
日志显示 PandoraBox bootloader 将设备识别为 PBR-M1/Newifi Y2S，缺少完整启动结尾、分区表和刷写方式；无法确认当前小米 4A 镜像缺陷，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf