coolsnowwolf/lede #9320: batman-adv导致编译失败

anjue39, 2022-04-27:
SHELL= flock /workdir/openwrt/tmp/.batman-adv-2022.0.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "batman-adv-2022.0.tar.gz" "49338705bc207709ac84d766688e702571009c827c0a320788ea51fb887714aa" "" "https://downloads.open-mesh.org/batman/releases/batman-adv-2022.0"    '
rm -f /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/batman-adv-2022.0/.built
touch /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/batman-adv-2022.0/.built_check
make  -C "/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.4.190" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi=target-arm_cortex-a9_musl_eabi" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Apr 25 11:18:22 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/symvers/gpio-button-hotplug.symvers /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/symvers/shortcut-fe.symvers /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/symvers/mac80211.symvers" KERNELRELEASE=5.4.190 M="/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/batman-adv-2022.0/net/batman-adv" CONFIG_BATMAN_ADV=m CONFIG_BATMAN_ADV_DEBUG=n CONFIG_BATMAN_ADV_BLA=y CONFIG_BATMAN_ADV_DAT=y CONFIG_BATMAN_ADV_MCAST=y CONFIG_BATMAN_ADV_NC=n CONFIG_BATMAN_ADV_BATMAN_V=y CONFIG_BATMAN_ADV_TRACING=n  EXTRA_CFLAGS=" -DCONFIG_BATMAN_ADV=1  -DCONFIG_BATMAN_ADV_BLA=1  -DCONFIG_BATMAN_ADV_DAT=1  -DCONFIG_BATMAN_ADV_MCAST=1  -DCONFIG_BATMAN_ADV_BATMAN_V=1 " NOSTDINC_FLAGS="-nostdinc  -isystem /workdir/openwrt/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/include -I/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/batman-adv-2022.0/net/batman-adv -I/workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/usr/include/mac80211-backport -I/workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/usr/include/mac80211-backport/uapi -I/workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/usr/include/mac80211 -I/workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/usr/include/mac80211/uapi -I/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/batman-adv-2022.0/include/ -include backport/autoconf.h -include backport/backport.h -include /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/batman-adv-2022.0/compat-hacks.h -DBATADV_SOURCE_VERSION=\\\"2022.0-openwrt-1\\\"" modules
make[4]: Entering directory '/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.4.190'
  CC [M]  /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/batman-adv-2022.0/net/batman-adv/bat_algo.o
In file included from <command-line>:
/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/batman-adv-2022.0/compat-hacks.h: In function 'br_multicast_has_router_adjacent':
/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/batman-adv-2022.0/compat-hacks.h:32:53: error: 'struct br_ip' has no member named 'dst'
       ipv6_addr_is_ll_all_routers(&br_ip_entry->addr.dst.ip6))
                                                     ^
make[5]: *** [scripts/Makefile.build:262: /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/batman-adv-2022.0/net/batman-adv/bat_algo.o] Error 1
make[4]: *** [Makefile:1734: /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/batman-adv-2022.0/net/batman-adv] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.4.190'
make[3]: *** [Makefile:95: /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/batman-adv-2022.0/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/routing/batman-adv'
time: package/feeds/routing/batman-adv/compile#1.10#0.36#1.44
    ERROR: package/feeds/routing/batman-adv failed to build.
make[2]: *** [package/Makefile:116: package/feeds/routing/batman-adv/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-04-27:
相同问题，自己看另外一个帖子

anjue39, 2022-04-27:
> 相同问题，自己看另外一个帖子

哪个帖子

anjue39, 2022-04-27:
自己fork,然后撤回这个提交batman-adv: Drop compat hacks for kernels < 5.10
是这样？

还是如下图撤销上游两个提交？

发自我的iPhone

WYC-2020, 2022-04-27:
> 自己fork,然后撤回这个提交batman-adv: Drop compat hacks for kernels < 5.10 是这样？ 还是如下图撤销上游两个提交？ 发自我的iPhone
> […](#)

[image]
不会就用我仓库 lede/package/kernel 这下面有我撤销好的，你自己放到你的仓库，在install的时候不要-f 就行了

mesetop599, 2022-04-27:
我用wyc-2020大神的也出错，另外看了下大神的actions最近2天也报错了

WYC-2020, 2022-04-27:
> 我用wyc-2020大神的也出错，另外看了下大神的actions最近2天也报错了

我那个报错是一个垃圾库的hash和这个没关系

WYC-2020, 2022-04-27:
> 我用wyc-2020大神的也出错，另外看了下大神的actions最近2天也报错了

另外用我库不要随便去改源，如果要新增就增加在后面，并且install不要用-f，绝对能编译过得，不要问为什么因为我亲测的

anjue39, 2022-04-28:
> > 我用wyc-2020大神的也出错，另外看了下大神的actions最近2天也报错了
> 
> 另外用我库不要随便去改源，如果要新增就增加在后面，并且install不要用-f，绝对能编译过得，不要问为什么因为我亲测的

我用WYC-2020的修改好的库，测试没问题

WYC-2020, 2022-04-28:
> > > 我用wyc-2020大神的也出错，另外看了下大神的actions最近2天也报错了
> > 
> > 
> > 另外用我库不要随便去改源，如果要新增就增加在后面，并且install不要用-f，绝对能编译过得，不要问为什么因为我亲测的
> 
> 我用WYC-2020的修改好的库，测试没问题

那是，最近我更新了一波，有几个地方有被更新出问题，直接fuck了几个hash，加入了几个kconfig就可以了

anjue39, 2022-04-28:
👍，你lede是同步更新最新的吗

发自我的iPhone

beenbin, 2022-04-28:
> > 自己fork,然后撤回这个提交batman-adv: Drop compat hacks for kernels < 5.10 是这样？ 还是如下图撤销上游两个提交？ 发自我的iPhone
> > […](#)
> 
> [image] 不会就用我仓库 lede/package/kernel 这下面有我撤销好的，你自己放到你的仓库，在install的时候不要-f 就行了

我用了WYC-2020大佬的kernel，终于编译成功了！！ 请问以后你会了lean大的lede同步更新么？

WYC-2020, 2022-04-28:
> > > 自己fork,然后撤回这个提交batman-adv: Drop compat hacks for kernels < 5.10 是这样？ 还是如下图撤销上游两个提交？ 发自我的iPhone
> > > […](#)
> > 
> > 
> > [image] 不会就用我仓库 lede/package/kernel 这下面有我撤销好的，你自己放到你的仓库，在install的时候不要-f 就行了
> 
> 我用了WYC-2020大佬的kernel，终于编译成功了！！ 请问以后你会了lean大的lede同步更新么？

你看我那次不是同步了的，只是如果我觉得不合理我可能会改，不过大部分相同，功能我也会按自己的需求增加。所以不想这样的话，你们可以自己fork大佬的，然后把我那个还原的放到相同目录即可

anjue39, 2022-04-28:
其实我一直在lede大佬仓库没找到那两个文件commit

发自我的iPhone

beenbin, 2022-04-29:
> 其实我一直在lede大佬仓库没找到那两个文件commit 发自我的iPhone
> […](#)

我也是，找不到哪两个需要回溯的commit，所以我就按WYC-2020大佬说的，把他的kernel文件fork下来覆盖了lean大佬的，然后编译成功了。不过听说是kernel内的linux修改导致的，所以应该是fork linux里的文件就可以了

WYC-2020, 2022-04-29:
> 其实我一直在lede大佬仓库没找到那两个文件commit 发自我的iPhone
> […](#)

真是服你们，这个在上游啊，自然找不到commit了，在l大这里

anjue39, 2022-04-29:
你试试只动linux部分。看行不行
我试了全新pull重新编译，不行

发自我的iPhone

anjue39, 2022-04-29:
就是在上游lede那里没找到，我在那个修改者那里找到了

发自我的iPhone

nywangyu, 2022-04-29:
用[WYC-2020](https://github.com/WYC-2020)大佬的编译成功了。

anjue39, 2022-04-29:
> > 相同问题，自己看另外一个帖子
> 
> 哪个帖子 #9291
你说过有两种改法，一种是 直接在代码里加一个宏，如果是小于5.10的就用原来的逻辑，如果大于等于5.10的就用atomic_read，还有一种是直接吧5.4的也修复成用atomic类型，然后驱动里面就不用加宏了直接全用atomic_read

你现在库里是采用的第一种还是第二种？

binge8, 2022-04-30:
请教一下，actions云编译如何取消-f

binge8, 2022-05-01:
费这个吊劲，换个思路嘛，看看什么插件依赖这玩意，别选不就完事了。我编译的armv8架构的，看了看是luci-app-easymesh的依赖，这玩意也没什么吊用，取消不就ok了，batman-adv也不要选，亲测成功

WYC-2020, 2022-05-01:
> 费这个吊劲，换个思路嘛，看看什么插件依赖这玩意，别选不就完事了。我编译的armv8架构的，其他架构思路也一样，看了看是luci-app-easymesh的依赖，这玩意也没什么吊用，取消不就ok了，batman-adv也不要选，亲测成功

你这说了当没说，别人就是要用那个玩意，所以才回说编译问题嘛

anjue39, 2022-05-01:
> > 费这个吊劲，换个思路嘛，看看什么插件依赖这玩意，别选不就完事了。我编译的armv8架构的，其他架构思路也一样，看了看是luci-app-easymesh的依赖，这玩意也没什么吊用，取消不就ok了，batman-adv也不要选，亲测成功
> 
> 你这说了当没说，别人就是要用那个玩意，所以才回说编译问题嘛

嗯，默认是勾选的，原则上保持lede的上游源库，加上自己的自定义
  要么直接用WYC-2020的库做源码库
  要么把大神kernel单独弄个库+lede的源码库，编译前kernel做个rm和replace，
  要么直接把5.4的也修复成用atomic类型
-----------
等上游慢慢合并好吧，这是临时的三种解决方法

anjue39, 2022-05-01:
> > 费这个吊劲，换个思路嘛，看看什么插件依赖这玩意，别选不就完事了。我编译的armv8架构的，其他架构思路也一样，看了看是luci-app-easymesh的依赖，这玩意也没什么吊用，取消不就ok了，batman-adv也不要选，亲测成功
> 
> 你这说了当没说，别人就是要用那个玩意，所以才回说编译问题嘛

嗯，默认是勾选的，原则上保持lede的上游源库，加上自己的自定义
要么直接用WYC-2020的库做源码库
要么把大神kernel单独弄个库+lede的源码库，编译前kernel做个rm和replace，
要么直接把5.4的也修复成用atomic类型
------
等上游慢慢合并解决好吧，以上三种临时解决方式
顺便问@WYC-2020，我后台make menuconfig怎么找不到use testing kernel，如何选择kernel？

anjue39, 2022-05-02:
同样的方法我试过了，编译不成功

发自我的iPhone

WYC-2020, 2022-05-02:
> 同样的方法我试过了，编译不成功 发自我的iPhone
> […](#)

自己问题，那么多人成功不可能偏偏你失败

anjue39, 2022-05-02:
看错action了，我也成功了。

发自我的iPhone

  

  
自己问题，那么多人成功不可能偏偏你失败

anjue39, 2022-10-11:
好像默认就不带-f，在workflows里对应yml文件代码

发自我的iPhone

anjue39, 2022-10-11:
我看到了，恢复到上一次改动，还是不行

发自我的iPhone

anjue39, 2022-10-11:
是把你lede/package/kernel下所有的覆盖到lede对应的目录还是只把lede/package/kernel/batman-adv/这个文件夹覆盖复制到kernel下？
原lede目录下好像没有batman-adv

发自我的iPhone

  

  
上游在这里(https://git.openwrt.org/feed/routing.git)，但是你没法直接修改的，官方应该不会加回来的，所以你只能自己fork一份修改，然后在openwrt引用你修改的，要么就像我那样，单独吧那个库提取出来，放到自己的仓库，最终编译的时候用你自己那份就行了，不想在维护 routing仓库，所以就没frok

xiaopang0v0, 2023-02-14:
因为编译的 目标内核 < 5.10 ，只需要在
`feeds.conf.default` 中 
```
src-git routing https://git.openwrt.org/feed/routing.git
```
的结尾加上 `;openwrt-21.02` 即可。(openwrt-21.02 是分支名，请不要自作主张将其改为你要编译的版本)
最终效果为
```
src-git routing https://git.openwrt.org/feed/routing.git;openwrt-21.02
```
然后在 lede 根目录更新 feeds 文件

```
./scripts/feeds update -a
./scripts/feeds install -a
```

coolsnowwolf, 2026-09-05:
当前 batman-adv 已更新至 2023.3，旧 issue 使用 2022.0 在 Linux 5.4 上访问 `struct br_ip.addr.dst` 的兼容性错误；当前包与补丁栈已更新，原始编译问题不再对应当前代码，关闭为 completed。

closed 2026-09-05 by coolsnowwolf