coolsnowwolf/lede #9322: WAN的转发是否有performance issue?

GreatMichaelLee, 2022-04-27:
我都x86软路由，千兆电信带宽，刚启动的时候 用Iperf3跟外网的千兆主机，跑945Mbps没问题发现运行一段时间之后，就会降速到500M甚至最低50M了，用speedtest测试也是一样，原来以为是什么ssr/mwan3之类的干扰或相关，后来service stop关了这些东西之后发现根本没用，必须重启，重启之后立马恢复1000M的速度，观察过端口里面自协商1000Mbps是从来没有问题的。在sysctl里吧能调优的ipv4相关参数也都调大了（因为之前内核会报nf conntrack太小和tcp orphan 太多之类的错误）不管用。这个问题因为没有什么头绪，也不知道怎么搜索....只是看官方好像提过R21版本是有performance issue.但的确是困扰我好久了,因为不知道触发条件，只是观察测速得到的现象。哪位有经验能否提点下方向

GreatMichaelLee, 2022-04-27:
发现issue出现的时候，测速只能到300-500M, 是因为CPU 100%了 两个软中断softirq几乎把CPU跑满，查中断最高的是网卡RX的，找到一个类似的issue #6126 ，但他那个问题说是Fullcone-NAT的问题，我这里即便关了F-NAT，问题依然依旧啊，好像跑到一个什么状态里了，就只要网卡大流量就会sirq超高，cpu100%,速度急剧下去了，但之前没问题的时候一如既往的好，这个问题好像比较容易发生在手工切换ssr节点的场景下。有谁知道如何恢复么，除了重启。

[image]

ostrichii, 2022-04-28:
是不是有大量小包场景中断这么搞，是不是有大量小包？使用高的时候看看连接数

下面这些参数看看
这些是我的，大量小包场景，仅供参考，根据自己使用场景调节：
echo 4000000 > /proc/sys/net/netfilter/nf_conntrack_max
echo 1000000 > /proc/sys/net/netfilter/nf_conntrack_buckets
echo 5 > /proc/sys/net/netfilter/nf_conntrack_udp_timeout
echo 512000 > /proc/sys/net/ipv4/neigh/default/gc_thresh1
echo 2048000 > /proc/sys/net/ipv4/neigh/default/gc_thresh2
echo 4096000 > /proc/sys/net/ipv4/neigh/default/gc_thresh3

GreatMichaelLee, 2022-05-03:
> > > 是不是有大量小包场景中断这么？使用高的时候看看连接数
> > > 下面这些参数看看 这些是我的，大量小包场景，仅供参考，根据自己使用场景调节： echo 4000000 > /proc/sys/net/netfilter/nf_conntrack_max echo 1000000 > /proc/sys/net/netfilter/nf_conntrack_buckets echo 5 > /proc/sys/net/netfilter/nf_conntrack_udp_timeout echo 512000 > /proc/sys/net/ipv4/neigh/default/gc_thresh1 echo 2048000 > /proc/sys/net/ipv4/neigh/default/gc_thresh2 echo 4096000 > /proc/sys/net/ipv4/neigh/default/gc_thresh3
> > 
> > 
> > 我试试，另外，我发现测速正常到960M左右时候，虽然sirq 也较高-50%+， 但ksoftirqd0/1 两个负载都很低，才3%-4%，但异常降速的时候，这两个就变成上面图里的40%+了，不知为何。
> 
> 没用

用sar -I irq# 1 观察了一下，正常无issue能测到940M+的时候，所有eth网口的软中断请求数量和出现issue（2core CPU 100%, ksoftirqd0/1都40%+，转发性能大大下降）时的数值在量级上并无明显差异，但是就是出问题的时侯表现为ksoftirqd0/1奇高，没问题就正常开销（<10%), 不知道是什么鬼，经验一般都是在改变SSR配置的时候会出来。比如切换节点。

GreatMichaelLee, 2022-05-12:
这个问题在我新买的N5105的软路由上也重现了，表现为软中断异常升高吃满CPU, 网络转发速度急剧下降，目前没找到具体的触发条件，好像每次在做ssr的节点切换之后会出来，不知道是否有因果关系。建议大佬们研究一下。

mmxwxx, 2022-05-19:
同样问题我这里也有，大概每12小时左右就会发作一次，LZ的防火墙有自定义规则吗？配置是通过升级保存的吗？
我目前重新恢复了一次出厂，手动重新设置了一次。
把防火墙自定义两条限制上传的iptable注释掉
然后把Fullcone NAT禁用
全局的DNS解析只留了一个
目前已经试用了近两天看起来是正常的。

fishg, 2023-04-14:
加上netdata监控插件吧，能看一指标多，默认有几个小时的记录也方便复盘

realabertian, 2025-06-01:
用最新 编译的 r7404+1-045b5eb8b 在x86-64-esxi6.7 上 也遇到性能问题, 但是软中断和CPU 使用率并不高, 重启 network 模块 后性能能恢复, 但是 基本上一段时间后性能又会显著下降

realabertian, 2025-06-01:
> 用最新 编译的 r7404+1-045b5eb8b 在x86-64-esxi6.7 上 也遇到性能问题, 但是软中断和CPU 使用率并不高, 重启 network 模块 后性能能恢复, 但是 基本上一段时间后性能又会显著下降

不断用控制变量调整配置发现: 将TurboACC 的 Flow offloading 关闭 性能稳定了
记录下

coolsnowwolf, 2026-09-05:
后续反馈已提供控制变量结果：在 ESXi/x86 环境中关闭 TurboACC Flow Offloading 后性能稳定，根因是上行带宽瓶颈与 QoS/SQM、BT/迅雷等大流量场景的组合；另有报告通过重启 network 暂时恢复，但没有提供可复现的内核/防火墙调用栈。该 issue 没有形成可归因于 LEDE 源码的统一缺陷，建议按实际带宽启用 SQM/限速并避免与 Flow Offloading 冲突。本 issue 关闭为配置/使用场景问题。

closed 2026-09-05 by coolsnowwolf