coolsnowwolf/lede #9324: 子选项配置中未选中的项目仍会按照全选状态构建依赖

ltrsunny, 2022-04-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：以bluez为例，我的config中并未选中依赖于它的包。像gst1-plugins-bad中有bluez相关的module(但我并未开启)，即使不选中，也会在构建依赖时下载编译bluez，并在此处出错（个人猜测是bluez所需的依赖环境并未构建）。手动删除feeds目录中的bluez可成功编译gst1-plugins-bad。

### (2) 路由器型号和固件版本

A：MT7629和MT7623测试均失败。

### (3) 详细日志

A：
```
WARNING: Makefile 'package/feeds/telephony/asterisk/Makefile' has a dependency on 'bluez-libs', which does not exist
WARNING: Makefile 'package/feeds/packages/bluelog/Makefile' has a dependency on 'bluez-libs', which does not exist
WARNING: Makefile 'package/feeds/packages/bluelog/Makefile' has a dependency on 'bluez-libs', which does not exist
WARNING: Makefile 'package/feeds/packages/gammu/Makefile' has a dependency on 'bluez-libs', which does not exist
WARNING: Makefile 'package/feeds/packages/gnunet/Makefile' has a dependency on 'bluez-libs', which does not exist
WARNING: Makefile 'package/feeds/packages/gst1-plugins-bad/Makefile' has a dependency on 'bluez-libs', which does not exist
WARNING: Makefile 'package/feeds/packages/openobex/Makefile' has a dependency on 'bluez-libs', which does not exist
WARNING: Makefile 'package/feeds/packages/python3/Makefile' has a dependency on 'bluez-libs', which does not exist
```

WYC-2020, 2022-04-27:
其他有东西使用了不用怀疑这个

ltrsunny, 2022-04-27:
> WARNING: Makefile 'package/feeds/packages/gst1-plugins-bad/Makefile' has a dependency on 'bluez-libs', which does not exist

WARNING: Makefile 'package/feeds/packages/gst1-plugins-bad/Makefile' has a dependency on 'bluez-libs', which does not exist

ltrsunny, 2022-04-27:
> 其他有东西使用了不用怀疑这个

WARNING: Makefile 'package/feeds/packages/gst1-plugins-bad/Makefile' has a dependency on 'bluez-libs', which does not exist

coolsnowwolf, 2026-09-05:
Kconfig 中可选依赖的构建由所选包的依赖关系决定；报告没有最小 config 和依赖图，删除 feeds 包规避不构成源码修复，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf