coolsnowwolf/lede #9326: 配置不能保存

tqangxl, 2022-04-28:
## 日志信息(必填！)
 删除相关防火墙规则完成。
 清空并关闭相关程序和缓存完成。
 TCP节点：[my-NodeA]216.240.128.200:11740，监听端口：1041
 过滤服务配置：准备接管域名解析...
   + [0]Pdnsd (127.0.0.1:7913)...
   | - [0]上游DNS：8.8.4.4:53
   - 域名解析：pdnsd + 使用(TCP节点)解析域名...
   * 请确认上游 DNS 支持 TCP 查询，如非直连地址，确保 TCP 代理打开，并且已经正确转发！
   | - (chinadns-ng) 最高支持4级域名过滤...
   | - [0](chinadns-ng) 代理域名表合并到防火墙域名表
   | - [0](chinadns-ng) 域名白名单合并到中国域名表
   + 过滤服务：ChinaDNS-NG(:7914)：国内DNS：223.5.5.5,8.8.8.8，可信DNS：127.0.0.1#7913
   - 节点列表中的域名(vpsiplist)：223.5.5.5,8.8.8.8
   - 域名白名单(whitelist)：223.5.5.5,8.8.8.8
   - 节点订阅域名(blacklist)：127.0.0.1#7913
   - 代理域名表(blacklist)：127.0.0.1#7913
   - 防火墙域名表(gfwlist)：默认
   - 中国域名表(chnroute)：默认
   - 以上所列以外及默认：127.0.0.1#7914
   - PassWall必须依赖于Dnsmasq，如果你自行配置了错误的DNS流程，将会导致域名(直连/代理域名)分流失效！！！
 开始加载防火墙规则...
 加入负载均衡的节点到ipset[vpsiplist]直连完成
 加入所有节点到ipset[vpsiplist]直连完成
 加载路由器自身 TCP 代理...
   - [0]将上游 DNS 服务器 8.8.4.4:53 加入到路由器自身代理的 TCP 转发链
 加载路由器自身 UDP 代理...
 TCP默认代理：使用TCP节点[my-NodeA] [中国列表以外](REDIRECT:1041)代理所有端口
 UDP默认代理：使用UDP节点[my-NodeA] [中国列表以外](TPROXY:1041)代理所有端口
 防火墙规则加载完成！
 重启 dnsmasq 服务
 配置定时任务：自动更新规则。
 运行完成！

## 截图

[image]

## 系统相关信息(必填)

Chrome 版本 101.0.4951.41（正式版本） （64 位）未启用安全dns

## 其他相关信息

kern.info : Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
kern.debugl: On node 0 totalpages: 65536
kern.debugl:   Normal zone: 576 pages used for memmap
kern.debugl:   Normal zone: 0 pages reserved
kern.debugl:   Normal zone: 65536 pages, LIFO batch:15
kern.info : MMU: Allocated 1088 bytes of context maps for 255 contexts
kern.debugl: pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
kern.debugl: pcpu-alloc: [0] 0
kern.info : Built 1 zonelists, mobility grouping on.  Total pages: 64960
kern.noticel] Kernel command line: rootfstype=squashfs noinitrd
kern.info : Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
kern.info : Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
kern.info : mem auto-init: stack:off, heap alloc:off, heap free:off
kern.info : Memory: 251732K/262144K available (5876K kernel code, 220K rwdata, 1272K rodata, 172K init, 229K bss, 10412K reserved, 0K cma-reserved)
kern.info : Kernel virtual memory layout:
kern.info :   * 0xfffdf000..0xfffff000  : fixmap
kern.info :   * 0xd1000000..0xfffdf000  : vmalloc & ioremap
kern.info : SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
kern.info : NR_IRQS: 512, nr_irqs: 512, preallocated irqs: 16
kern.warn : UIC0 (32 IRQ sources) at DCR 0xc0
kern.warn : UIC1 (32 IRQ sources) at DCR 0xd0
kern.warn : UIC2 (32 IRQ sources) at DCR 0xe0
kern.warn : UIC3 (32 IRQ sources) at DCR 0xf0
kern.noticel] random: get_random_u32 called from start_kernel+0x2e4/0x3f4 with crng_init=0
kern.debugl: time_init: decrementer frequency = 1000.000010 MHz
kern.debugl: time_init: processor frequency   = 1000.000010 MHz
[... 954 lines cut ...]
ath: Country alpha2 being used: US
ath: Regpair used: 0x3a
ath: regdomain 0x8348 dynamically updated by user
br-lan: port 2(wlan1) entered blocking state
br-lan: port 2(wlan1) entered disabled state
device wlan1 entered promiscuous mode
IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
br-lan: port 2(wlan1) entered blocking state
br-lan: port 2(wlan1) entered forwarding state
br-lan: port 3(wlan0) entered blocking state
br-lan: port 3(wlan0) entered disabled state
device wlan0 entered promiscuous mode
br-lan: port 3(wlan0) entered blocking state
br-lan: port 3(wlan0) entered forwarding state
IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
xhci_hcd 0000:45:00.0: firmware failed to load
xhci_hcd 0000:45:00.0: Direct firmware load for K2026090.mem failed with error -2
xhci_hcd 0000:45:00.0: Falling back to sysfs fallback for: K2026090.mem
xhci_hcd 0000:45:00.0: firmware failed to load
xhci_hcd 0000:45:00.0: Unloading driver
appfilter version:5.0.1
init oaf sysctl...ok
recv user msg
oaf msg handle, action = 0
module init.........

Purple-wind-bell, 2022-04-29:
空间不足的典型结果，别问怎么知道的。。。

vvv-jinhuichen, 2022-04-29:
我遇到同样的问题，对比其他的openwrt文件系统发现
Filesystem                Size      Used Available Use% Mounted on
tmpfs                    28.1M     12.8M     15.3M  46% /
tmpfs                    28.1M    232.0K     27.9M   1% /tmp
tmpfs                   512.0K         0    512.0K   0% /dev
问题机器的Filesystem   都是tmpfs  不知道是什么问题导致的。苦恼中

vvv-jinhuichen, 2022-04-30:
我的问题找到了可以提供点参考
openwrt编译后会生成许多安装包
squashfs-sysupgrade.bin 这是正常的包后缀名，是可以保存配置的
initramfs-kernel.bin 这样的后缀名刷进去文件系统都是tmpfs，是无法保存配置的

我是粗心大意，编译的文件大小超越了设备模型只生成了initramfs-kernel.bin 刷进去各种要命问题
后来在仔细排查后发现少了包仔细检查log发现设备模型存储太小了无法生成squashfs-sysupgrade.bin 包。吧8M存储模型修改到16M就出来了刷进去就有正常的文件系统了

gusheng2000, 2022-05-31:
我默认编译 两个文件都出来了 结果刷入还是这样一重启就系统重置编译出来固件才8M
好迷惑> 我的问题找到了可以提供点参考
> 
> openwrt编译后会生成许多安装包
> 
> squashfs-sysupgrade.bin 这是正常的包后缀名，是可以保存配置的
> 
> initramfs-kernel.bin 这样的后缀名刷进去文件系统都是tmpfs，是无法保存配置的
> 
> 
> 
> 我是粗心大意，编译的文件大小超越了设备模型只生成了initramfs-kernel.bin 刷进去各种要命问题
> 
> 后来在仔细排查后发现少了包仔细检查log发现设备模型存储太小了无法生成squashfs-sysupgrade.bin 包。吧8M存储模型修改到16M就出来了刷进去就有正常的文件系统了
> 
>

tqangxl, 2022-06-10:
> 我的问题找到了可以提供点参考 openwrt编译后会生成许多安装包 squashfs-sysupgrade.bin 这是正常的包后缀名，是可以保存配置的 initramfs-kernel.bin 这样的后缀名刷进去文件系统都是tmpfs，是无法保存配置的
> 
> 我是粗心大意，编译的文件大小超越了设备模型只生成了initramfs-kernel.bin 刷进去各种要命问题 后来在仔细排查后发现少了包仔细检查log发现设备模型存储太小了无法生成squashfs-sysupgrade.bin 包。吧8M存储模型修改到16M就出来了刷进去就有正常的文件系统了

你都改了那些文件的那些内容，能分享下吗

tqangxl, 2022-06-10:
@vvv-jinhuichen 你都改了那些文件的那些内容 

能分享下你的经验吗？谢谢

tqangxl, 2022-06-10:
@WYC-2020 的确是空间用光的原因，能否分享个改成128M的教程呀，谢谢

XFFF2022, 2022-07-16:
大佬你好，为什么squashfs-sysupgrade.bin 在bread里刷入后无法启动，无限重启

coolsnowwolf, 2026-09-05:
报告是旧版第三方代理脚本/配置在 WNDR4700 上的保存问题，日志没有 UCI 写入失败信息，也没有最小配置；无法归因当前 LEDE，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf