coolsnowwolf/lede #9327: mac编译失败

jzdayz, 2022-04-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无法编译通过

### (2) 路由器型号和固件版本

A： 小米R4a 千兆

### (3) 详细日志

A：
```
(cd /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0; CFLAGS="-pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0=mbedtls-2.28.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections " CXXFLAGS="-pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0=mbedtls-2.28.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections " LDFLAGS="-L/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=mipsel -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="" -DCMAKE_C_COMPILER="/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="" -DCMAKE_CXX_COMPILER="/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="" -DCMAKE_ASM_COMPILER="/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/Volumes/OpenWrt/lede/staging_dir/target-mipsel_24kc_musl/usr;/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/Volumes/OpenWrt/lede/staging_dir/target-mipsel_24kc_musl -DCMAKE_PREFIX_PATH=/Volumes/OpenWrt/lede/staging_dir/target-mipsel_24kc_musl -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EXPORT_PACKAGE_REGISTRY=FALSE -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=TRUE -DCMAKE_GENERATOR="Ninja" -DUSE_SHARED_MBEDTLS_LIBRARY:Bool=ON -DENABLE_TESTING:Bool=OFF -DENABLE_PROGRAMS:Bool=ON /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0 )
-- The C compiler identification is GNU 8.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python3: /Volumes/OpenWrt/lede/staging_dir/host/bin/python3 (found version "3.8.9") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_ASM_COMPILER
    CMAKE_ASM_COMPILER_LAUNCHER
    CMAKE_CXX_COMPILER
    CMAKE_CXX_FLAGS_RELEASE
    CMAKE_EXPORT_NO_PACKAGE_REGISTRY
    CMAKE_EXPORT_PACKAGE_REGISTRY
    CMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY
    CMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY
    CMAKE_MODULE_LINKER_FLAGS
    DL_LIBRARY

-- Build files have been written to: /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0
awk 'BEGIN { rc = 1 } /#define MBEDTLS_DEBUG_C/ { $0 = "// #define MBEDTLS_DEBUG_C"; rc = 0 } /#define MBEDTLS_ARMV8CE_AES_C/ { $0 = "// #define MBEDTLS_ARMV8CE_AES_C"; rc = 0 } /#define MBEDTLS_HAVE_SSE2/ { $0 = "// #define MBEDTLS_HAVE_SSE2"; rc = 0 } { print } END { exit(rc) }' /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/include/mbedtls/config.h >/Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/include/mbedtls/config.h.new && mv /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/include/mbedtls/config.h.new /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/include/mbedtls/config.h
awk 'BEGIN { rc = 1 } /#define MBEDTLS_HKDF_C/ { $0 = "// #define MBEDTLS_HKDF_C"; rc = 0 } { print } END { exit(rc) }' /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/include/mbedtls/config.h >/Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/include/mbedtls/config.h.new && mv /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/include/mbedtls/config.h.new /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/include/mbedtls/config.h
sed -i '/fuzz/d' /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/programs/CMakeLists.txt
sed -i '/test/d' /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/programs/CMakeLists.txt
touch /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/.configured_a1321993df4f599e2c5aebaa4d379c95
rm -f /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/.built
touch /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0/.built_check
MAKEFLAGS="" /Volumes/OpenWrt/lede/staging_dir/host/bin/ninja  -j1 -C /Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0 
ninja: Entering directory `/Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0'
[178/362] Linking C executable programs/fuzz/fuzz_client
FAILED: programs/fuzz/fuzz_client 
: && /Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-gcc -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/Volumes/OpenWrt/lede/build_dir/target-mipsel_24kc_musl/mbedtls-2.28.0=mbedtls-2.28.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -Wall -Wextra -Wwrite-strings -Wformat=2 -Wno-format-nonliteral -Wvla -Wlogical-op -Wshadow -Wformat-signedness -Wformat-overflow=2 -Wformat-truncation -Werror -O2 -L/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/Volumes/OpenWrt/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro   -rdynamic CMakeFiles/mbedtls_test.dir/tests/src/asn1_helpers.c.o CMakeFiles/mbedtls_test.dir/tests/src/drivers/hash.c.o CMakeFiles/mbedtls_test.dir/tests/src/drivers/platform_builtin_keys.c.o CMakeFiles/mbedtls_test.dir/tests/src/drivers/test_driver_aead.c.o CMakeFiles/mbedtls_test.dir/tests/src/drivers/test_driver_cipher.c.o CMakeFiles/mbedtls_test.dir/tests/src/drivers/test_driver [... cut ...]
make[3]: Leaving directory '/Volumes/OpenWrt/lede/package/libs/mbedtls'
time: package/libs/mbedtls/compile#14.75#3.12#18.44
    ERROR: package/libs/mbedtls failed to build.
make[2]: *** [package/Makefile:116: package/libs/mbedtls/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [package/Makefile:110: /Volumes/OpenWrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:230: world] Error 2
```

jzdayz, 2022-04-29:
这个是单线程日志。

WYC-2020, 2022-04-29:
没条件没法修

jzdayz, 2022-04-29:
> 没条件没法修

好吧，那我开虚拟机用Ubuntu编译，谢谢解答

closed 2022-04-29 by jzdayz

retamia, 2022-11-08:
把package/libs/mbedtls/Makefile文件里的Build/Configure的两个sed
```
	sed -i '/fuzz/d' $(PKG_BUILD_DIR)/programs/CMakeLists.txt
	sed -i '/test/d' $(PKG_BUILD_DIR)/programs/CMakeLists.txt
```
放到$(Build/Configure/Default) 之前。 先去掉fuzz，然后在cmake