coolsnowwolf/lede #9331: 求助！ax3600配置，编译出来的固件确实qnap_301w的！

JohyC, 2022-04-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
[image]
我编译的配置文件；但是编译后出来的固件确实 qnap_301w! 
make menuconfig 里面找不到ax3600的编译选项了。
### (1) 具体问题

A：无法编译 ax3600
[image]

make menuconfig  里面只有这四个选项。。。

### (2) 路由器型号和固件版本

A：ax3600 IPQ807A1

### (3) 详细日志

A：

EnnawYang, 2022-04-30:
此机型早已被扬

JohyC, 2022-04-30:
> 此机型早已被扬

啊！额。。。这怎么一说呀？

minlang112, 2022-04-30:
此机型已转入qsdk那个库了吧

JohyC, 2022-04-30:
> 此机型已转入qsdk那个库了吧

有地址吗？我好久没编译ax3600固件了。我说怎么见鬼找不到东西了。

he110777, 2022-04-30:
唉，我也想编译ax3600的固件，谁知删库了

JohyC, 2022-04-30:
> 唉，我也想编译ax3600的固件，谁知删库了

我把我fork的仓库回退到 drop 之前了。
https://github.com/JohyC/lede-Source-code
你可以用这个暂时编译用用。

XiaoliChan, 2022-04-30:
I think you can use this source code: [robimarko-5.15](https://github.com/robimarko/openwrt/tree/ipq807x-5.15)

I used it for a month, it was freaking stable. :)

1582130940, 2022-04-30:
revert edbd8d2e9839357f3a4f0a06174d243f362b1544
Please fix the conflict yourself

closed 2022-05-01 by JohyC