coolsnowwolf/lede #9332: 疑是 Turbo ACC 网络加速 引起磁盘错误

tianiue, 2022-04-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
出现错误日志时Turbo ACC 网络加速不正常，重启恢复，用两天左右再次出现
A：

### (2) 路由器型号和固件版本
X86
A：

### (3) 详细日志
[320103.089774] blk_update_request: operation not supported error, dev loop0, sector 488368 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[320164.444457] print_req_error: 10 callbacks suppressed
[320164.444461] blk_update_request: operation not supported error, dev loop0, sector 489672 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[321809.714808] blk_update_request: operation not supported error, dev loop0, sector 489624 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[321809.717981] blk_update_request: operation not supported error, dev loop0, sector 489680 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[323341.833615] blk_update_request: operation not supported error, dev loop0, sector 489688 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[324078.067200] blk_update_request: operation not supported error, dev loop0, sector 288176 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[324083.701139] blk_update_request: operation not supported error, dev loop0, sector 47736 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[324083.704338] blk_update_request: operation not supported error, dev loop0, sector 288264 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[324083.707745] blk_update_request: operation not supported error, dev loop0, sector 489512 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[324083.711237] blk_update_request: operation not supported error, dev loop0, sector 489720 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[324386.797556] blk_update_request: operation not supported error, dev loop0, sector 489704 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[325268.182310] blk_update_request: operation not supported error, dev loop0, sector 489736 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[326352.314273] blk_update_request: operation not supported error, dev loop0, sector 41168 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[326352.317455] blk_update_request: operation not supported error, dev loop0, sector 489752 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[327201.187717] blk_update_request: operation not supported error, dev loop0, sector 489768 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[327201.191076] blk_update_request: operation not supported error, dev loop0, sector 489784 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[327265.953824] blk_update_request: operation not supported error, dev loop0, sector 489800 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[328172.426552] blk_update_request: operation not supported error, dev loop0, sector 489792 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[328172.429809] blk_update_request: operation not supported error, dev loop0, sector 489808 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[330002.779042] blk_update_request: operation not supported error, dev loop0, sector 489816 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[330066.774221] blk_update_request: operation not supported error, dev loop0, sector 489840 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[331233.591154] blk_update_request: operation not supported error, dev loop0, sector 288272 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
[331293.494236] blk_update_request: operation not supported error, dev loop0, sector 47768 op 0x3:(DISCARD) flags 0x0 phys_seg 1 prio class 0
A：
[image]

shenyz, 2022-04-30:
acc 表示不背黑锅，看下挂载点/dev/loop0 是不是100%了

tianiue, 2022-04-30:
[image]
 一直没有找到什么原因引起

hyy-666, 2022-05-01:
有过一次，我这里不是acc的锅，是我的U盘供电不足

tianiue, 2022-05-03:
> 有过一次，我这里不是acc的锅，是我的U盘供电不足

我是ESXi虚拟机环境，在插件更新，开机几天会出现这个情况

Sirius9525, 2022-05-30:
我也遇到同样的问题了 esxi虚拟环境下。有解决办法了不。。

tianiue, 2022-05-31:
> 我也遇到同样的问题了 esxi虚拟环境下。有解决办法了不。。

[passwall](https://github.com/xiaorouji/openwrt-passwall)更新组件和规则时也会出现，不确定问题

yyysuo, 2022-08-31:
我通过删除esxi快照，整合磁盘，重启openwrt解决了这个问题。
猜测可能的原因：
1：esxi快照过多或者快照太老，时间跨度大，触发了esxi关于磁盘方面的bug。
2：op虚拟机磁盘参数设置有与op不兼容的地方。

希望楼主尝试并反馈，谢谢。

Karmylr, 2022-10-05:
我也遇到了

yyysuo, 2023-04-15:
> > 有过一次，我这里不是acc的锅，是我的U盘供电不足
> 
> 我是ESXi虚拟机环境，在插件更新，开机几天会出现这个情况

esxi+scsi+smartdns+squashfs，就是这个效果。
https://github.com/pymumu/smartdns/issues/1373#issuecomment-1508163009

yyysuo, 2023-04-15:
> 我也遇到同样的问题了 esxi虚拟环境下。有解决办法了不。。

esxi+scsi+smartdns+squashfs，就是这个效果。
https://github.com/pymumu/smartdns/issues/1373#issuecomment-1508163009

yyysuo, 2023-04-15:
> 我也遇到了

esxi+scsi+smartdns+squashfs，就是这个效果。
https://github.com/pymumu/smartdns/issues/1373#issuecomment-1508163009

Kisoul, 2024-01-07:
exsi出现这个问题还可以尝试把虚拟机硬盘的驱动改为IDE，我是这么解决的

coolsnowwolf, 2026-09-05:
日志中的 `blk_update_request: operation not supported` 明确来自虚拟块设备 `loop0` 对 DISCARD 请求不支持，属于宿主机/磁盘映射对 loop 设备的能力限制，并非 Turbo ACC 网络加速模块的网络或磁盘 I/O 代码错误。issue 未提供磁盘拓扑、loop 挂载来源或 Turbo ACC 触发调用栈，无法形成可修复的 LEDE 补丁；建议在宿主机关闭对该 loop 设备的 discard，或改用支持 TRIM 的块设备。该历史问题关闭。

closed 2026-09-05 by coolsnowwolf