coolsnowwolf/lede #9333: 编译错误，不知哪里的问题

HiJwm, 2022-04-30:
反馈bug/问题模板，提建议请删除

 (1) 具体问题

A：编译不成功

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：time: package/feeds/packages/runc/compile#0.21#0.10#0.28
make[3]: Entering directory '/workdir/openwrt/feeds/packages/utils/containerd'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.containerd-1.5.11.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "containerd-1.5.11.tar.gz" "02b79d5e2b07b5e64cd28f1fe84395ee11eef95fc49fd923a9ab93022b148be6" "" "https://codeload.github.com/containerd/containerd/tar.gz/v1.5.11?"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://codeload.github.com/containerd/containerd/tar.gz/v1.5.11?/containerd-1.5.11.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  1 7318k    1 85249    0     0   335k      0  0:00:21 --:--:--  0:00:21  334k
100 7318k  100 7318k    0     0  13.8M      0 --:--:-- --:--:-- --:--:-- 13.8M
Hash of the downloaded file does not match (file: 6a289406c1c0583763e5a9754e31a1eced55cd5f162a7bc2a3a315d5eb05c7a1, requested: 02b79d5e2b07b5e64cd28f1fe84395ee11eef95fc49fd923a9ab93022b148be6) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/containerd-1.5.11.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/containerd-1.5.11.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/containerd-1.5.11.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.

EnnawYang, 2022-04-30:
github的问题，可以自己改下hash或者直接skip

HiJwm, 2022-04-30:
containerd-1.5.11 hash value error

Hokia, 2022-04-30:
我也遇到了同样的问题

Hokia, 2022-04-30:
> github的问题，可以自己改下hash或者直接skip

你好，如果改hash值要在哪里改？

Hokia, 2022-04-30:
> github的问题，可以自己改下hash或者直接skip

你好，如果改hash值要在哪里改？流程是自动的，出错自动退出了，如果跳过该怎么操作。

WYC-2020, 2022-05-01:
fuck掉这玩意的hash

binge8, 2022-05-01:
已修复，可以关闭了

songlijun, 2022-05-14:
[image]
如果你有http proxy梯子，可以加在这个地方

closed 2022-05-23 by AmadeusGhost