coolsnowwolf/lede #9334: 编译过程中多次在这里卡住，下载的文件hash不匹配，其他源又不可用，手动试过这几个源确实是404

Hokia, 2022-04-30:
make[3]: Entering directory '/home/r/lede/feeds/packages/utils/containerd'
mkdir -p /home/r/lede/dl
SHELL= flock /home/r/lede/tmp/.containerd-1.5.11.tar.gz.flock -c '      /home/r/lede/scripts/download.pl "/home/r/lede/dl" "containerd-1.5.11.tar.gz" "02b79d5e2b07b5e64cd28f1fe84395ee11eef95fc49fd923a9ab93022b148be6" "" "https://codeload.github.com/containerd/containerd/tar.gz/v1.5.11?"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://codeload.github.com/containerd/containerd/tar.gz/v1.5.11?/containerd-1.5.11.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 7318k  100 7318k    0     0  2784k      0  0:00:02  0:00:02 --:--:-- 2783k
Hash of the downloaded file does not match (file: 6a289406c1c0583763e5a9754e31a1eced55cd5f162a7bc2a3a315d5eb05c7a1, requested: 02b79d5e2b07b5e64cd28f1fe84395ee11eef95fc49fd923a9ab93022b148be6) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/containerd-1.5.11.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/containerd-1.5.11.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/containerd-1.5.11.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:62: /home/r/lede/dl/containerd-1.5.11.tar.gz] Error 2
make[3]: Leaving directory '/home/r/lede/feeds/packages/utils/containerd'
time: package/feeds/packages/containerd/compile#0.45#0.19#5.64
    ERROR: package/feeds/packages/containerd failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/containerd/compile] Error 1
make[2]: Leaving directory '/home/r/lede'
make[1]: *** [package/Makefile:110: /home/r/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/r/lede'
make: *** [/home/r/lede/include/toplevel.mk:230: world] Error 2

Jason6111, 2022-05-01:
sed -i 's/PKG_HASH.*/PKG_HASH:=skip/' feeds/packages/utils/containerd/Makefile

WYC-2020, 2022-05-01:
fuck掉这个玩意的hash

binge8, 2022-05-01:
已修复，可以关闭了

closed 2022-05-05 by AmadeusGhost