coolsnowwolf/lede #9338: ipv4和ipv6冲突并且ivp4间歇断开

wuzhuang123, 2022-05-02:
极路由HC5962.
开启ipv4和ipv6后重启后都不能获取地址，关闭ipv6后ipv4可以获取，此时再开启ipv6就可以获取到ipv6地址。但在10分钟之内就会丢失ip几秒钟后再次获取，导致游戏断开，只保留ipv4后间歇断网无改善。附日志。

同时启用ipv4和ipv6后重启日志:
Mon May  2 13:05:04 2022 kern.info kernel: [   77.869514] mt7530 mdio-bus:1f wan: Link is Down
Mon May  2 13:05:04 2022 daemon.notice netifd: Network device 'wan' link is down
Mon May  2 13:05:04 2022 daemon.notice netifd: Interface 'wan' has link connectivity loss
Mon May  2 13:05:04 2022 daemon.notice netifd: Interface 'WAN6' has link connectivity loss
Mon May  2 13:05:04 2022 daemon.notice netifd: wan (6989): udhcpc: received SIGTERM
Mon May  2 13:05:05 2022 daemon.notice netifd: wan (6989): udhcpc: entering released state
Mon May  2 13:05:05 2022 daemon.notice netifd: ZEROTIER (6807): udhcpc: broadcasting discover
Mon May  2 13:05:05 2022 daemon.notice netifd: wan (6989): Command failed: Permission denied
Mon May  2 13:05:05 2022 daemon.notice netifd: Interface 'wan' is now down
Mon May  2 13:05:05 2022 daemon.err nginx[6768]: 2022/05/02 13:05:02 [emerg] 6768#0: still could not bind()
Mon May  2 13:05:05 2022 daemon.notice netifd: Interface 'WAN6' is now down
Mon May  2 13:05:06 2022 daemon.notice netifd: Network device 'ztppiulgoh' link is down
Mon May  2 13:05:06 2022 daemon.notice netifd: Interface 'ZEROTIER' has link connectivity loss
Mon May  2 13:05:06 2022 daemon.notice netifd: ZEROTIER (6807): udhcpc: read error: Network is down, reopening socket
Mon May  2 13:05:06 2022 daemon.notice netifd: ZEROTIER (6807): udhcpc: bind: No such device
Mon May  2 13:05:06 2022 daemon.notice netifd: Interface 'ZEROTIER' is now down
Mon May  2 13:05:06 2022 daemon.notice netifd: Interface 'ZEROTIER' is disabled

只保留ipv4后间歇断网的日志:
Mon May  2 13:10:01 2022 daemon.notice netifd: ZEROTIER (15182): udhcpc: started, v1.35.0
Mon May  2 13:10:01 2022 daemon.notice netifd: ZEROTIER (15182): udhcpc: broadcasting discover
Mon May  2 13:10:02 2022 kern.info kernel: [  338.104025] mt7530 mdio-bus:1f wan: Link is Up - 100Mbps/Full - flow control rx/tx
Mon May  2 13:10:02 2022 kern.info kernel: [  338.111845] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
Mon May  2 13:10:02 2022 daemon.notice netifd: Network device 'wan' link is up
Mon May  2 13:10:02 2022 daemon.notice netifd: Interface 'wan' has link connectivity
Mon May  2 13:10:02 2022 daemon.notice netifd: Interface 'wan' is setting up now
Mon May  2 13:10:02 2022 daemon.notice netifd: wan (15335): udhcpc: started, v1.35.0
Mon May  2 13:10:02 2022 daemon.notice netifd: wan (15335): udhcpc: broadcasting discover
Mon May  2 13:10:04 2022 daemon.notice netifd: ZEROTIER (15182): udhcpc: broadcasting discover
Mon May  2 13:10:04 2022 daemon.notice netifd: wan (15335): udhcpc: broadcasting select for 192.168.1.43, server 192.168.1.1
Mon May  2 13:10:04 2022 daemon.notice netifd: wan (15335): udhcpc: lease of 192.168.1.43 obtained from 192.168.1.1, lease time 259200
Mon May  2 13:10:04 2022 daemon.notice netifd: Interface 'wan' is now up
Mon May  2 13:10:04 2022 user.notice firewall: Reloading firewall due to ifup of wan (wan)

aNoo6, 2022-05-02:
C526a同遇
开机无法获取ipv4，重开电脑网卡驱动可以获取
也是大概10分钟掉一次网

coolsnowwolf, 2026-09-05:
极路由 HC5962 的日志明确包含 mt7530 WAN 链路掉线、ZeroTier 和 nginx 绑定错误，IPv4/IPv6 不是唯一变量；缺少当前硬件/链路复现，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf