coolsnowwolf/lede #9339: wsl编译grub2报错

sirstrone, 2022-05-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
Q：是否搜索了issue (使用 "x" 选择)
*  存在类似的issue，但使用对应issue的方法解决均无效

## 2. 详细叙述
wsl编译grub2报错 相同config在github云编译上正常
已尝试过以下解决方案 
make distclean
rm -r .\tmp && rm -r build_dir
apt install / update / upgrade 所需依赖

### (2) 路由器型号和固件版本

A：编译版本 x86_64
      wsl release version : Ubuntu 20.04.3 LTS

### (3) 详细日志
`/home/leo/lede/staging_dir/host/bin/grub-mkimage -d /home/leo/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-core -p /boot/grub -O "x86_64"-efi -c /home/leo/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-early.cfg -o /home/leo/lede/staging_dir/target-x86_64_musl/image/grub2/bootx64.efi at_keyboard boot chain configfile fat linux ls part_gpt reboot serial efi_gop efi_uga
/home/leo/lede/staging_dir/host/bin/grub-mkimage -d /home/leo/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-core -p /boot/grub -O "x86_64"-efi -c ./files/grub-early.cfg -o /home/leo/lede/staging_dir/target-x86_64_musl/image/grub2/iso-bootx64.efi at_keyboard boot chain configfile fat iso9660 linux ls part_msdos part_gpt reboot serial test efi_gop efi_uga
touch /home/leo/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi.installed
mkdir -p /home/leo/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/leo/lede/tmp/.root-copy.flock -c 'cp -fpR /home/leo/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi/. /home/leo/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/leo/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.grub2-efi_installed
echo "grub2-efi" >> /home/leo/lede/staging_dir/target-x86_64_musl/pkginfo/grub2.efi.install
make[3]: Leaving directory '/home/leo/lede/package/boot/grub2'
time: package/boot/grub2/efi/compile#68.46#7.59#87.45
make[2]: Leaving directory '/home/leo/lede'
make[1]: *** [package/Makefile:110: /home/leo/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/leo/lede'
make: *** [/home/leo/lede/include/toplevel.mk:230: world] Error 2`

EnnawYang, 2022-05-02:
不包含有效错误日志，也没看出来是grub2报错

sirstrone, 2022-05-02:
> 不包含有效错误日志，也没看出来是grub2报错

上下文就这坨玩意了 
有方法抓取gcc全部的编译日志吗？

EnnawYang, 2022-05-02:
看起来像是多线程编译的，单线程再跑一次试试，日志尽量再往上截点

sirstrone, 2022-05-02:
> 看起来像是多线程编译的，单线程再跑一次试试，日志尽量再往上截点

单线程编译成功 我在跑一遍多线程编译试一下  Thanks~

closed 2022-05-02 by sirstrone