coolsnowwolf/lede #9340: 特性请求：88X2bu驱动支持

xspeed1989, 2022-05-02:
88x2bu的USB WIFI 驱动（https://github.com/RinCat/RTL88x2BU-Linux-Driver）是否可以集成进固件里面

WYC-2020, 2022-05-02:
必须可以

xspeed1989, 2022-05-02:
> 必须可以

哈哈

EnnawYang, 2022-05-02:
[隔壁](https://github.com/immortalwrt/immortalwrt/tree/openwrt-18.06-k5.4/package/kernel/rtl88x2bu)有  但我没设备也没用过

xspeed1989, 2022-05-02:
> [隔壁](https://github.com/immortalwrt/immortalwrt/tree/openwrt-18.06-k5.4/package/kernel/rtl88x2bu)有 但我没设备也没用过，也不知是否支持5.15

我试试看！

xspeed1989, 2022-05-02:
> [隔壁](https://github.com/immortalwrt/immortalwrt/tree/openwrt-18.06-k5.4/package/kernel/rtl88x2bu)有 但我没设备也没用过，也不知是否支持5.15

可以编译过，但是没法用，会导致系统起不起来

Boos4721, 2022-05-02:
**UNSUPPORT MAC80211** 5.10.3x and higher

EnnawYang, 2022-05-02:
> **UNSUPPORT MAC80211** 5.10.3x and higher

emmm，master分支似乎支持

Boos4721, 2022-05-02:
> > **UNSUPPORT MAC80211** 5.10.3x and higher
> 
> emmm，master分支似乎支持

humm, ive tride but only client mode.

Boos4721, 2022-05-02:
> > > > **UNSUPPORT MAC80211** 5.10.3x and higher
> > > 
> > > 
> > > emmm，master分支似乎支持
> > 
> > 
> > humm, ive tride but only client mode.
> 
> 我没看到哪里有5.10内核的分支呀

https://github.com/morrownr/88x2bu-20210702 try this

xspeed1989, 2022-05-02:
> > > > > **UNSUPPORT MAC80211** 5.10.3x and higher
> > > > 
> > > > 
> > > > emmm，master分支似乎支持
> > > 
> > > 
> > > humm, ive tride but only client mode.
> > 
> > 
> > 我没看到哪里有5.10内核的分支呀
> 
> https://github.com/morrownr/88x2bu-20210702 try this

I want try it, But i can't find Makefile for the repo.

Boos4721, 2022-05-02:
> > > > > > **UNSUPPORT MAC80211** 5.10.3x and higher
> > > > > 
> > > > > 
> > > > > emmm，master分支似乎支持
> > > > 
> > > > 
> > > > humm, ive tride but only client mode.
> > > 
> > > 
> > > 我没看到哪里有5.10内核的分支呀
> > 
> > 
> > [morrownr/88x2bu-20210702](https://github.com/morrownr/88x2bu-20210702) try this
> 
> I want try it, But i can't find Makefile for the repo.

https://github.com/immortalwrt/immortalwrt/tree/master/package/kernel/rtl88x2bu

xspeed1989, 2022-05-02:
> https://github.com/immortalwrt/immortalwrt/tree/master/package/kernel/rtl88x2bu

OK, I will try it and post result here.

xspeed1989, 2022-05-02:
```Makefile
#
# Copyright (C) 2021 ImmortalWrt
# <https://immortalwrt.org>
#
# This is free software, licensed under the GNU General Public License v3.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=rtl88x2bu
PKG_RELEASE:=1

PKG_SOURCE_URL:=https://github.com/morrownr/88x2bu-20210702.git
PKG_SOURCE_PROTO:=git
PKG_SOURCE_DATE:=2021-07-02
PKG_SOURCE_VERSION:=b434bafd4c8a23d1a32cc4ca9ad3ed0cd5d07375
PKG_MIRROR_HASH:=skip

PKG_LICENSE:=GPL-2.0
PKG_LICENSE_FILES:=LICENSE

PKG_BUILD_PARALLEL:=1

STAMP_CONFIGURED_DEPENDS := $(STAGING_DIR)/usr/include/mac80211-backport/backport/autoconf.h

include $(INCLUDE_DIR)/kernel.mk
include $(INCLUDE_DIR)/package.mk

define KernelPackage/rtl88x2bu
  SUBMENU:=Wireless Drivers
  TITLE:=Realtek 8812BU/8822BU support
  DEPENDS:=+kmod-cfg80211 +kmod-usb-core +@DRIVER_11N_SUPPORT +@DRIVER_11AC_SUPPORT
  FILES:=$(PKG_BUILD_DIR)/rtl88x2bu.ko
  AUTOLOAD:=$(call AutoProbe,rtl88x2bu)
  PROVIDES:=kmod-rtl88x2bu
endef

NOSTDINC_FLAGS = \
	-I$(PKG_BUILD_DIR) \
	-I$(PKG_BUILD_DIR)/include \
	-I$(STAGING_DIR)/usr/include/mac80211-backport \
	-I$(STAGING_DIR)/usr/include/mac80211-backport/uapi \
	-I$(STAGING_DIR)/usr/include/mac80211 \
	-I$(STAGING_DIR)/usr/include/mac80211/uapi \
	-include backport/autoconf.h \
	-include backport/backport.h

EXTRA_CFLAGS:= \
	-DRTW_SINGLE_WIPHY \
	-DRTW_USE_CFG80211_STA_EVENT \
	-DCONFIG_IOCTL_CFG80211 \
	-DCONFIG_CONCURRENT_MODE \
	-DBUILD_OPENWRT

EXTRA_KCONFIG:= \
	CONFIG_RTL8822BU=m \
	USER_MODULE_NAME=rtl88x2bu

MAKE_OPTS:= \
	$(KERNEL_MAKE_FLAGS) \
	M="$(PKG_BUILD_DIR)" \
	NOSTDINC_FLAGS="$(NOSTDINC_FLAGS)" \
	USER_EXTRA_CFLAGS="$(EXTRA_CFLAGS)" \
	$(EXTRA_KCONFIG)

define Build/Compile
	+$(MAKE) $(PKG_JOBS) -C "$(LINUX_DIR)" \
		$(MAKE_OPTS) \
		modules
endef

$(eval $(call KernelPackage,rtl88x2bu))
```

xspeed1989, 2022-05-02:
@Boos4721 build error
```bash
Checking out files from the git repository...
Cloning into 'rtl88x2bu-2021-07-02-b434bafd'...
remote: Enumerating objects: 1103, done.
remote: Counting objects: 100% (122/122), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 1103 (delta 81), reused 76 (delta 76), pack-reused 981
Receiving objects: 100% (1103/1103), 4.29 MiB | 4.38 MiB/s, done.
Resolving deltas: 100% (496/496), done.
Note: switching to 'b434bafd4c8a23d1a32cc4ca9ad3ed0cd5d07375'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at b434baf updated version
Packing checkout...
touch /home/john/Desktop/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2021-07-02-b434bafd/.prepared_4c736cedb3d68f4302df4f58b896339c_6664517399ebbbc92a37c5bb081b5c53_check
. /home/john/Desktop/lede/include/shell.sh; xzcat /home/john/Desktop/lede/dl/rtl88x2bu-2021-07-02-b434bafd.tar.xz | tar -C /home/john/Desktop/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2021-07-02-b434bafd/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/john/Desktop/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2021-07-02-b434bafd

Applying ./patches/001-use-kernel-byteorder.patch using plaintext: 
patching file include/drv_types.h

Applying ./patches/010-remove-repeat-flies.patch using plaintext: 
patching file include/drv_conf.h
patching file include/linux/old_unused_rtl_wireless.h (renamed from include/linux/wireless.h)
patching file include/rtl_autoconf.h (renamed from include/autoconf.h)

Applying ./patches/020-change-value-of-vht-enable-and-usb-mode.patch using plaintext: 
patching file os_dep/linux/os_intfs.c
Hunk #1 succeeded at 317 with fuzz 1 (offset 8 lines).
Hunk #2 succeeded at 415 (offset 23 lines).

Applying ./patches/020-wireless-5.8.patch using plaintext: 
patching file os_dep/linux/ioctl_cfg80211.c
Hunk #1 FAILED at 7808.
Hunk #2 FAILED at 10286.
2 out of 2 hunks FAILED -- saving rejects to file os_dep/linux/ioctl_cfg80211.c.rej
Patch failed!  Please fix ./patches/020-wireless-5.8.patch!
```

Boos4721, 2022-05-02:
> @Boos4721 build error
> 
> ```shell
> Checking out files from the git repository...
> Cloning into 'rtl88x2bu-2021-07-02-b434bafd'...
> remote: Enumerating objects: 1103, done.
> remote: Counting objects: 100% (122/122), done.
> remote: Compressing objects: 100% (46/46), done.
> remote: Total 1103 (delta 81), reused 76 (delta 76), pack-reused 981
> Receiving objects: 100% (1103/1103), 4.29 MiB | 4.38 MiB/s, done.
> Resolving deltas: 100% (496/496), done.
> Note: switching to 'b434bafd4c8a23d1a32cc4ca9ad3ed0cd5d07375'.
> 
> You are in 'detached HEAD' state. You can look around, make experimental
> changes and commit them, and you can discard any commits you make in this
> state without impacting any branches by switching back to a branch.
> 
> If you want to create a new branch to retain commits you create, you may
> do so (now or later) by using -c with the switch command. Example:
> 
>   git switch -c <new-branch-name>
> 
> Or undo this operation with:
> 
>   git switch -
> 
> Turn off this advice by setting config variable advice.detachedHead to false
> 
> HEAD is now at b434baf updated version
> Packing checkout...
> touch /home/john/Desktop/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2021-07-02-b434bafd/.prepared_4c736cedb3d68f4302df4f58b896339c_6664517399ebbbc92a37c5bb081b5c53_check
> . /home/john/Desktop/lede/include/shell.sh; xzcat /home/john/Desktop/lede/dl/rtl88x2bu-2021-07-02-b434bafd.tar.xz | tar -C /home/john/Desktop/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2021-07-02-b434bafd/.. -xf -
> [ ! -d ./src/ ] || cp -fpR ./src/. /home/john/Desktop/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2021-07-02-b434bafd
> 
> Applying ./patches/001-use-kernel-byteorder.patch using plaintext: 
> patching file include/drv_types.h
> 
> Applying ./patches/010-remove-repeat-flies.patch using plaintext: 
> patching file include/drv_conf.h
> patching file include/linux/old_unused_rtl_wireless.h (renamed from include/linux/wireless.h)
> patching file include/rtl_autoconf.h (renamed from include/autoconf.h)
> 
> Applying ./patches/020-change-value-of-vht-enable-and-usb-mode.patch using plaintext: 
> patching file os_dep/linux/os_intfs.c
> Hunk #1 succeeded at 317 with fuzz 1 (offset 8 lines).
> Hunk #2 succeeded at 415 (offset 23 lines).
> 
> Applying ./patches/020-wireless-5.8.patch using plaintext: 
> patching file os_dep/linux/ioctl_cfg80211.c
> Hunk #1 FAILED at 7808.
> Hunk #2 FAILED at 10286.
> 2 out of 2 hunks FAILED -- saving rejects to file os_dep/linux/ioctl_cfg80211.c.rej
> Patch failed!  Please fix ./patches/020-wireless-5.8.patch!
> ```

https://github.com/Boos4721/openwrt/commit/13797003dbf82d09e239424daf5c4dd58c01703f

xspeed1989, 2022-05-02:
@Boos4721 build failed
```bash
SHELL= flock /home/john/Desktop/lede/tmp/.rtl88x2bu-2022-02-21-36f5ceb8.tar.xz.flock -c '  	/home/john/Desktop/lede/scripts/download.pl "/home/john/Desktop/lede/dl" "rtl88x2bu-2022-02-21-36f5ceb8.tar.xz" "79d6b3c7679bd5366f030c300936225d8892b67a95bb32233088e200ddeae6eb" "" || (  /home/john/Desktop/lede/scripts/dl_github_archive.py --dl-dir="/home/john/Desktop/lede/dl" --url="https://github.com/morrownr/88x2bu-20210702.git" --version="36f5ceb8e8d9b8348b1aa2a92acf888b439f892a" --subdir="rtl88x2bu-2022-02-21-36f5ceb8" --source="rtl88x2bu-2022-02-21-36f5ceb8.tar.xz" --hash="79d6b3c7679bd5366f030c300936225d8892b67a95bb32233088e200ddeae6eb" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/john/Desktop/lede/tmp/dl && cd /home/john/Desktop/lede/tmp/dl && rm -rf rtl88x2bu-2022-02-21-36f5ceb8 && [ \! -d rtl88x2bu-2022-02-21-36f5ceb8 ] && git clone  https://github.com/morrownr/88x2bu-20210702.git rtl88x2bu-2022-02-21-36f5ceb8 && (cd rtl88x2bu-2022-02-21-36f5ceb8 && git checkout 36f5ceb8e8d9b8348b1aa2a92acf888b439f892a && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd rtl88x2bu-2022-02-21-36f5ceb8 && git log -1 --format='\''@%ct'\''` && rm -rf rtl88x2bu-2022-02-21-36f5ceb8/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c rtl88x2bu-2022-02-21-36f5ceb8 | 	xz -zc -7e > /home/john/Desktop/lede/tmp/dl/rtl88x2bu-2022-02-21-36f5ceb8.tar.xz && mv /home/john/Desktop/lede/tmp/dl/rtl88x2bu-2022-02-21-36f5ceb8.tar.xz /home/john/Desktop/lede/dl/ && rm -rf rtl88x2bu-2022-02-21-36f5ceb8; );  )    '
touch /home/john/Desktop/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-02-21-36f5ceb8/.prepared_3a40df9b6904002fd8303599b217262b_6664517399ebbbc92a37c5bb081b5c53_check
. /home/john/Desktop/lede/include/shell.sh; xzcat /home/john/Desktop/lede/dl/rtl88x2bu-2022-02-21-36f5ceb8.tar.xz | tar -C /home/john/Desktop/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-02-21-36f5ceb8/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/john/Desktop/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-02-21-36f5ceb8

Applying ./patches/001-use-kernel-byteorder.patch using plaintext: 
patching file include/drv_types.h

Applying ./patches/010-remove-repeat-flies.patch using plaintext: 
patching file include/drv_conf.h
patching file include/linux/old_unused_rtl_wireless.h (renamed from include/linux/wireless.h)
patching file include/rtl_autoconf.h (renamed from include/autoconf.h)

Applying ./patches/020-change-value-of-vht-enable-and-usb-mode.patch using plaintext: 
patching file os_dep/linux/os_intfs.c

Applying ./patches/020-wireless-5.8.patch using plaintext: 
patching file os_dep/linux/ioctl_cfg80211.c
Hunk #1 FAILED at 7808.
Hunk #2 FAILED at 10286.
2 out of 2 hunks FAILED -- saving rejects to file os_dep/linux/ioctl_cfg80211.c.rej
Patch failed!  Please fix ./patches/020-wireless-5.8.patch!
make[3]: *** [Makefile:72: /home/john/Desktop/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/rtl88x2bu-2022-02-21-36f5ceb8/.prepared_3a40df9b6904002fd8303599b217262b_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/john/Desktop/lede/package/kernel/rtl88x2bu'
time: package/kernel/rtl88x2bu/compile#0.21#0.35#0.47
    ERROR: package/kernel/rtl88x2bu failed to build.

```

xspeed1989, 2022-05-02:
I have given up, I will switch kernel version to 5.10

Boos4721, 2022-05-02:
> I have given up, I will switch the kernel version to 5.10 I can test 88x2bu driver if you need

u should revert mac80211 version to 5.4.x

xspeed1989, 2022-05-03:
没有能工作的驱动，最后乖乖的用回友善的r4s rom

可以没有源码，不能自己迁移到lede

WYC-2020, 2022-05-04:
> 88x2bu的USB WIFI 驱动（[https://github.com/RinCat/RTL88x2BU-Linux-Driver）](https://github.com/RinCat/RTL88x2BU-Linux-Driver%EF%BC%89) 支持比较新的内核，我在UBUNTU上面已经测试过可用，是否可以集成进来

这玩意从官方说明是支持到5.17的，具体为啥不行的分析一下

xspeed1989, 2022-05-04:
> > 88x2bu的USB WIFI 驱动（[https://github.com/RinCat/RTL88x2BU-Linux-Driver）](https://github.com/RinCat/RTL88x2BU-Linux-Driver%EF%BC%89) 支持比较新的内核，我在UBUNTU上面已经测试过可用，是否可以集成进来
> 
> 这玩意从官方说明是支持到5.17的，具体为啥不行的分析一下

@WYC-2020 额，这个主要是我不会写package的makefile

WYC-2020, 2022-05-04:
> > > 88x2bu的USB WIFI 驱动（[https://github.com/RinCat/RTL88x2BU-Linux-Driver）](https://github.com/RinCat/RTL88x2BU-Linux-Driver%EF%BC%89) 支持比较新的内核，我在UBUNTU上面已经测试过可用，是否可以集成进来
> > 
> > 
> > 这玩意从官方说明是支持到5.17的，具体为啥不行的分析一下
> 
> @WYC-2020 额，这个主要是我不会写package的makefile

有时间再看

WYC-2020, 2022-05-05:
> https://github.com/RinCat/RTL88x2BU-Linux-Driver

看了他们引用的源不同，并且那几个源没有明说支持5.15，你可以吧那个源改成这个 然后编译测试一下是否可以工作即可

WYC-2020, 2022-05-05:
> > > 88x2bu的USB WIFI 驱动（[https://github.com/RinCat/RTL88x2BU-Linux-Driver）](https://github.com/RinCat/RTL88x2BU-Linux-Driver%EF%BC%89) 支持比较新的内核，我在UBUNTU上面已经测试过可用，是否可以集成进来
> > 
> > 
> > 这玩意从官方说明是支持到5.17的，具体为啥不行的分析一下
> 
> @WYC-2020 额，这个主要是我不会写package的makefile

https://github.com/WYC-2020/lede/commit/8f3399bf0ea22fef23fbebe5d9a871d7eb876d99 用这个试下，如果可以用，记得给一个答复，有好几修了让他们测试了给答复都不答复

xspeed1989, 2022-05-05:
好的 感谢大佬，我设备在公司，周末带回家试试

xspeed1989, 2022-05-05:
@WYC-2020 编译通过了！然后暂时还没做测试

xspeed1989, 2022-05-05:
> @WYC-2020 编译通过了！然后暂时还没做测试

@WYC-2020 试了下，插着USB 会启动不起来系统， 开机后插入 运行ifconfig也会卡住，可能需要向上游反馈BUG

WYC-2020, 2022-05-06:
> > @WYC-2020 编译通过了！然后暂时还没做测试
> 
> @WYC-2020 试了下，插着USB 会启动不起来系统， 开机后插入 运行ifconfig也会卡住，可能需要向上游反馈BUG

那只能调试一波看才能知道了，有可能是和op的某些库匹配问题

xspeed1989, 2022-05-06:
> > > @WYC-2020 编译通过了！然后暂时还没做测试
> > 
> > 
> > @WYC-2020 试了下，插着USB 会启动不起来系统， 开机后插入 运行ifconfig也会卡住，可能需要向上游反馈BUG
> 
> 那只能调试一波看才能知道了，有可能是和op的某些库匹配问题

我只有周末才有时间折腾设备，到时候po一下dmesg上来

WYC-2020, 2022-05-07:
> https://github.com/RinCat/RTL88x2BU-Linux-Driver）

dev分支，我吧隔壁的patch 加入了 你再试下可以不，如果还不行 那就只能调试和像作者确认是否支持你这个了

xspeed1989, 2022-05-07:
> > [https://github.com/RinCat/RTL88x2BU-Linux-Driver）](https://github.com/RinCat/RTL88x2BU-Linux-Driver%EF%BC%89)
> 
> dev分支，我吧隔壁的patch 加入了 你再试下可以不，如果还不行 那就只能调试和像作者确认是否支持你这个了

好的 大佬，我先编译

xspeed1989, 2022-05-08:
@WYC-2020 仓库不可见了呢

WYC-2020, 2022-05-11:
> @WYC-2020 仓库不可见了呢

不开放了，看到一些吊毛就老火

xspeed1989, 2022-05-11:
啊 咋啦 😃

WYC-2020, 2022-05-14:
> […](#)

因为有的吊毛说我吃饱了闲的，所以懒得给你们搞了，爱咋玩咋玩没那个心情了，菜还不认

xspeed1989, 2022-05-16:
> > […](#)
> 
> 因为有的吊毛说我吃饱了闲的，所以懒得给你们搞了，爱咋玩咋玩没那个心情了，菜还不认

懂了 感谢

closed 2022-07-25 by aiamadeus