coolsnowwolf/lede #9344: x86版本机器经常无故重启

C-starm, 2022-05-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在x86的机器上使用commit ID为`ce2fb409e0315e2f573b13584415f18b7cb7bd63`及最新版本`30dc6d34b0f85c4dc0518ec956fab5ee5b9eddbc`均存在无故重启的情况.

### (2) 路由器型号和固件版本

A：
使用华为B515主机
软件版本：OpenWrt R22.4.1 / LuCI Master (git-22.115.51709-a2bd654)

<br class="Apple-interchange-newline" style="caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; text-decoration: none;">

### (3) 详细日志

A：机器重启，未捕捉到日志

XiaoliChan, 2022-05-03:
I highly recommend you install OpenWrt on VM platforms, not on the PC.

I tried it so many times, install on VM platforms gets more stability than on the PC. (Current I am using it on PVE)

C-starm, 2022-05-03:
hahaha, i also tried another version on esxi machine, but it will cause some strange issue, so i decide to use openwrt on phisical machine.

XiaoliChan, 2022-05-03:
> hahaha, i also tried another version on esxi machine, but it will cause some strange issue, so i decide to use openwrt on phisical machine.

Why not try PVE? ¯\_(ツ)_/¯

XiaoliChan, 2022-05-03:
Also, you can set a cronjob to reboot OpenWrt once a day at 5 AM.

C-starm, 2022-05-03:
emmm, some times, it will restart 3-4 times fo an hour, so... you know.....

XiaoliChan, 2022-05-03:
> emmm, some times, it will restart 3-4 times fo an hour, so... you know.....

Yep, I also meet the same situation as you say until I use PVE.

shu0802, 2022-05-10:
> 

升级到这个版本试试OpenWrt R22.5.5 / LuCI Master (git-22.121.65028-2a5da72)，前几天也出现同样的问题。升级后暂时没有出问题了。

coolsnowwolf, 2026-09-05:
x86 无故重启没有任何崩溃、watchdog、串口或硬件日志，无法区别电源/硬件/虚拟化问题，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf