coolsnowwolf/lede #9346: 编译报错

ydhang, 2022-05-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
X86_64
A：

### (3) 详细日志

A：
for subdir in .; do realdir=$(readlink -f /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119); grep -F /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119 /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/$subdir/Module.symvers >> /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/Module.symvers.tmp; [ "/home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119" = "$realdir" ] || grep -F $realdir /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/$subdir/Module.symvers >> /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/Module.symvers.tmp; done; sort -u /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/Module.symvers.tmp > /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/Module.symvers; mkdir -p /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers; mv /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/Module.symvers /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8152.symvers
touch /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/.built
rm -rf /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/.pkgdir/kmod-usb-net-rtl8152-vendor.installed /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/.pkgdir/kmod-usb-net-rtl8152-vendor
mkdir -p /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/.pkgdir/kmod-usb-net-rtl8152-vendor
mkdir -p /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/.pkgdir/kmod-usb-net-rtl8152-vendor/etc/modules.d; ( echo "r8152"; ) > /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/.pkgdir/kmod-usb-net-rtl8152-vendor/etc/modules.d/usb-net-rtl8152-vendor;
touch /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/.pkgdir/kmod-usb-net-rtl8152-vendor.installed
mkdir -p /home/ydhang/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/ydhang/lede/tmp/.root-copy.flock -c 'cp -fpR /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/.pkgdir/kmod-usb-net-rtl8152-vendor/. /home/ydhang/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ydhang/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-usb-net-rtl8152-vendor_installed
removed '/home/ydhang/lede/bin/targets/x86/64/packages/kmod-usb-net-rtl8152-vendor_5.15.36+2.15.20211119-x86-1_x86_64.ipk'
mkdir -p /home/ydhang/lede/bin/targets/x86/64/packages /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/ipkg-x86_64/kmod-usb-net-rtl8152-vendor/CONTROL /home/ydhang/lede/staging_dir/target-x86_64_musl/pkginfo
mkdir -p /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/ipkg-x86_64/kmod-usb-net-rtl8152-vendor/etc/modules.d; ( echo "r8152"; ) > /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/ipkg-x86_64/kmod-usb-net-rtl8152-vendor/etc/modules.d/usb-net-rtl8152-vendor;
find /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/ipkg-x86_64/kmod-usb-net-rtl8152-vendor -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/ydhang/lede/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/ydhang/lede/scripts/strip-kmod.sh" PATCHELF="/home/ydhang/lede/staging_dir/host/bin/patchelf" /home/ydhang/lede/scripts/rstrip.sh /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/ipkg-x86_64/kmod-usb-net-rtl8152-vendor
rstrip.sh: /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/ipkg-x86_64/kmod-usb-net-rtl8152-vendor/lib/modules/5.15.36/r8152.ko: relocatable
(cd /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/ipkg-x86_64/kmod-usb-net-rtl8152-vendor/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/ydhang/lede/bin/targets/x86/64/packages
/home/ydhang/lede/staging_dir/host/bin/fakeroot /home/ydhang/lede/scripts/ipkg-build -m "" /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/ipkg-x86_64/kmod-usb-net-rtl8152-vendor /home/ydhang/lede/bin/targets/x86/64/packages
Packaged contents of /home/ydhang/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/ipkg-x86_64/kmod-usb-net-rtl8152-vendor into /home/ydhang/lede/bin/targets/x86/64/packages/kmod-usb-net-rtl8152-vendor_5.15.36+2.15.20211119-x86-1_x86_64.ipk
echo "kmod-usb-net-rtl8152-vendor" >> /home/ydhang/lede/staging_dir/target-x86_64_musl/pkginfo/r8152.default.install
make[3]: Leaving directory '/home/ydhang/lede/package/lean/r8152'
time: package/lean/r8152/compile#17.87#16.53#43.76
make[2]: Leaving directory '/home/ydhang/lede'
make[1]: *** [package/Makefile:110: /home/ydhang/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ydhang/lede'
make: *** [/home/ydhang/lede/include/toplevel.mk:230: world] Error 2

EnnawYang, 2022-05-03:
`Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on`

Beginner-Go, 2022-05-03:
我看看

Beginner-Go, 2022-05-03:
刚测了 不是r8125问题，自己排查吧
[image]

EnnawYang, 2022-05-03:
> 刚测了 不是r8125问题，自己排查吧 

他这也不是8125的错误日志啊dalao（

WYC-2020, 2022-05-03:
又是一个让人猜的问题，可以直接关闭这种issue

angelandy, 2022-05-20:
```shell
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://github.com/upx/upx/releases/download/v3.95/upx-3.95-src.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:02:15 --:--:--     0
curl: (56) OpenSSL SSL_read: error:0A000126:SSL routines::unexpected eof while reading, errno 0
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/upx-3.95-src.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   169    0     0    0     0      0      0 --:--:--  0:00:05 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/upx-3.95-src.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   169    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/upx-3.95-src.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   169    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:36: /home/wu/work/lede/dl/upx-3.95-src.tar.xz] Error 2
make[3]: Leaving directory '/home/wu/work/lede/tools/upx'
time: tools/upx/compile#0.07#0.06#144.01
    ERROR: tools/upx failed to build.
make[2]: *** [tools/Makefile:162: tools/upx/compile] Error 1
make[2]: Leaving directory '/home/wu/work/lede'
make[1]: *** [tools/Makefile:158: /home/wu/work/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyynynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/wu/work/lede'
make: *** [/home/wu/work/lede/include/toplevel.mk:230：world] 错误 2

```
我这边是各种404 啊

比如这个 ： https://mirror2.openwrt.org/sources/upx-3.95-src.tar.xz
确实是404 啊   不是被墙了

cnstark, 2022-11-23:
不是被墙了，是这个文件没有了

wivire, 2023-01-01:
我的编译20221001的源码，出现这错误：
make[3]: Entering directory '/home/u/lede/package/lean/upx'
mkdir -p /home/u/lede/dl
SHELL= flock /home/u/lede/tmp/.upx-2022-01-15.tar.gz.flock -c '          /home/u/lede/scripts/dl_github_archive.py --dl-dir="/home/u/lede/dl" --url="https://github.com/upx/upx.git" --version="a46b63817a9c6ad5af7cf519332e859f11558592" --subdir="upx-2022-01-15-a46b6381" --source="upx-2022-01-15.tar.gz" --hash="x" || (  echo "Checking out files from the git repository..."; mkdir -p /home/u/lede/tmp/dl && cd /home/u/lede/tmp/dl && rm -rf upx-2022-01-15-a46b6381 && [ \! -d upx-2022-01-15-a46b6381 ] && git clone  https://github.com/upx/upx.git upx-2022-01-15-a46b6381 && (cd upx-2022-01-15-a46b6381 && git checkout a46b63817a9c6ad5af7cf519332e859f11558592 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd upx-2022-01-15-a46b6381 && git log -1 --format='\''@%ct'\''` && rm -rf upx-2022-01-15-a46b6381/.git &&    tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c upx-2022-01-15-a46b6381 |       gzip -nc > /home/u/lede/tmp/dl/upx-2022-01-15.tar.gz && mv /home/u/lede/tmp/dl/upx-2022-01-15.tar.gz /home/u/lede/dl/ && rm -rf upx-2022-01-15-a46b6381; );     '
Traceback (most recent call last):
  File "/home/u/lede/scripts/dl_github_archive.py", line 19, in <module>
    import ssl
  File "/home/u/lede/staging_dir/hostpkg/lib/python3.10/ssl.py", line 99, in <module>
    import _ssl             # if we can't import it, let the error propagate
ModuleNotFoundError: No module named '_ssl'
Checking out files from the git repository...
Cloning into 'upx-2022-01-15-a46b6381'...
remote: Enumerating objects: 35389, done.
remote: Counting objects: 100% (934/934), done.
remote: Compressing objects: 100% (347/347), done.
remote: Total 35389 (delta 642), reused 823 (delta 587), pack-reused 34455
Receiving objects: 100% (35389/35389), 60.35 MiB | 13.02 MiB/s, done.
Resolving deltas: 100% (28366/28366), done.
fatal: reference is not a tree: a46b63817a9c6ad5af7cf519332e859f11558592
make[3]: *** [Makefile:67: /home/u/lede/dl/upx-2022-01-15.tar.gz] Error 128
make[3]: Leaving directory '/home/u/lede/package/lean/upx'
time: package/lean/upx/host-compile#2.22#1.26#6.64
    ERROR: package/lean/upx [host] failed to build.
make[2]: *** [package/Makefile:114: package/lean/upx/host/compile] Error 1
make[2]: Leaving directory '/home/u/lede'
make[1]: *** [package/Makefile:110: /home/u/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/u/lede'
make: *** [/home/u/lede/include/toplevel.mk:230: world] Error 2

但是python3里import _ssl没问题呀
u@ubuntu:~$ python3
Python 3.8.10 (default, Nov 14 2022, 12:59:47)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import _ssl
>>>

为啥会报No module named '_ssl'

sunnysab, 2023-01-25:
> 我的编译20221001的源码，出现这错误： make[3]: Entering directory '/home/u/lede/package/lean/upx' mkdir -p /home/u/lede/dl SHELL= flock /home/u/lede/tmp/.upx-2022-01-15.tar.gz.flock -c ' /home/u/lede/scripts/dl_github_archive.py --dl-dir="/home/u/lede/dl" --url="https://github.com/upx/upx.git" --version="a46b63817a9c6ad5af7cf519332e859f11558592" --subdir="upx-2022-01-15-a46b6381" --source="upx-2022-01-15.tar.gz" --hash="x" || ( echo "Checking out files from the git repository..."; mkdir -p /home/u/lede/tmp/dl && cd /home/u/lede/tmp/dl && rm -rf upx-2022-01-15-a46b6381 && [ ! -d upx-2022-01-15-a46b6381 ] && git clone https://github.com/upx/upx.git upx-2022-01-15-a46b6381 && (cd upx-2022-01-15-a46b6381 && git checkout a46b63817a9c6ad5af7cf519332e859f11558592 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd upx-2022-01-15-a46b6381 && git log -1 --format='\''@%ct'\''` && rm -rf upx-2022-01-15-a46b6381/.git && tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c upx-2022-01-15-a46b6381 | gzip -nc > /home/u/lede/tmp/dl/upx-2022-01-15.tar.gz && mv /home/u/lede/tmp/dl/upx-2022-01-15.tar.gz /home/u/lede/dl/ && rm -rf upx-2022-01-15-a46b6381; ); ' Traceback (most recent call last): File "/home/u/lede/scripts/dl_github_archive.py", line 19, in import ssl File "/home/u/lede/staging_dir/hostpkg/lib/python3.10/ssl.py", line 99, in import _ssl # if we can't import it, let the error propagate ModuleNotFoundError: No module named '_ssl' Checking out files from the git repository... Cloning into 'upx-2022-01-15-a46b6381'... remote: Enumerating objects: 35389, done. remote: Counting objects: 100% (934/934), done. remote: Compressing objects: 100% (347/347), done. remote: Total 35389 (delta 642), reused 823 (delta 587), pack-reused 34455 Receiving objects: 100% (35389/35389), 60.35 MiB | 13.02 MiB/s, done. Resolving deltas: 100% (28366/28366), done. fatal: reference is not a tree: a46b63817a9c6ad5af7cf519332e859f11558592 make[3]: *** [Makefile:67: /home/u/lede/dl/upx-2022-01-15.tar.gz] Error 128 make[3]: Leaving directory '/home/u/lede/package/lean/upx' time: package/lean/upx/host-compile#2.22#1.26#6.64 ERROR: package/lean/upx [host] failed to build. make[2]: *** [package/Makefile:114: package/lean/upx/host/compile] Error 1 make[2]: Leaving directory '/home/u/lede' make[1]: *** [package/Makefile:110: /home/u/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2 make[1]: Leaving directory '/home/u/lede' make: *** [/home/u/lede/include/toplevel.mk:230: world] Error 2
> 
> 但是python3里import _ssl没问题呀 u@ubuntu:~$ python3 Python 3.8.10 (default, Nov 14 2022, 12:59:47) [GCC 9.4.0] on linux Type "help", "copyright", "credits" or "license" for more information.
> 
> > > > import _ssl
> 
> 为啥会报No module named '_ssl'

我这边出现了 `fatal: reference is not a tree: a46b63817a9c6ad5af7cf519332e859f11558592`，不知道怎么解决。 SSL 的导入没啥问题

wscloudy, 2023-01-30:
我也是，git更新后出现的

Ljzd-PRO, 2023-02-23:
我也是

WangHan0711, 2023-08-11:
> ```shell
> + curl -f --connect-timeout 20 --retry 5 --location --insecure https://github.com/upx/upx/releases/download/v3.95/upx-3.95-src.tar.xz
>   % Total    % Received % Xferd  Average Speed [...]

哥们解决了吗，我也遇到了这个问题

wscloudy, 2023-08-12:
上一次编译已经没问题了，我clean之后重新更新就好了

WangHan0711, 2023-08-17:
> 上一次编译已经没问题了，我clean之后重新更新就好了

好的，谢谢，我再试试

whao123yahoo, 2023-09-02:
> > 我的编译20221001的源码，出现这错误： make[3]: Entering directory '/home/u/lede/package/lean/upx' mkdir -p /home/u/lede/dl SHELL= flock /home/u/lede/tmp/.upx-2022-01-15.tar.gz.flock -c ' /home/u/lede/scripts/dl_github_archive.py --dl-dir="/home/u/lede/dl" --url="https://github.com/upx/upx.git" --version="a46b63817a9c6ad5af7cf519332e859f11558592" --subdir="upx-2022-01-15-a46b6381" --source="upx-2022-01-15.tar.gz" --hash="x" || ( echo "Checking out files from the git repository..."; mkdir -p /home/u/lede/tmp/dl && cd /home/u/lede/tmp/dl && rm -rf upx-2022-01-15-a46b6381 && [ ! -d upx-2022-01-15-a46b6381 ] && git clone https://github.com/upx/upx.git upx-2022-01-15-a46b6381 && (cd upx-2022-01-15-a46b6381 && git checkout a46b63817a9c6ad5af7cf519332e859f11558592 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd upx-2022-01-15-a46b6381 && git log -1 --format='\''@%ct'\''` && rm -rf upx-2022-01-15-a46b6381/.git && tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c upx-2022-01-15-a46b6381 | gzip -nc > /home/u/lede/tmp/dl/upx-2022-01-15.tar.gz && mv /home/u/lede/tmp/dl/upx-2022-01-15.tar.gz /home/u/lede/dl/ && rm -rf upx-2022-01-15-a46b6381; ); ' Traceback (most recent call last): File "/home/u/lede/scripts/dl_github_archive.py", line 19, in import ssl File "/home/u/lede/staging_dir/hostpkg/lib/python3.10/ssl.py", line 99, in import _ssl # if we can't import it, let the error propagate ModuleNotFoundError: No module named '_ssl' Checking out files from the git repository... Cloning into 'upx-2022-01-15-a46b6381'... remote: Enumerating objects: 35389, done. remote: Counting objects: 100% (934/934), done. remote: Compressing objects: 100% (347/347), done. remote: Total 35389 (delta 642), reused 823 (delta 587), pack-reused 34455 Receiving objects: 100% (35389/35389), 60.35 MiB | 13.02 MiB/s, done. Resolving deltas: 100% (28366/28366), done. fatal: reference is not a tree: a46b63817a9c6ad5af7cf519332e859f11558592 make[3]: *** [Makefile:67: /home/u/lede/dl/upx-2022-01-15.tar.gz] Error 128 make[3]: Leaving directory '/home/u/lede/package/lean/upx' time: package/lean/upx/host-compile#2.22#1.26#6.64 ERROR: package/lean/upx [host] failed to build. make[2]: *** [package/Makefile:114: package/lean/upx/host/compile] Error 1 make[2]: Leaving directory '/home/u/lede' make[1]: *** [package/Makefile:110: /home/u/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2 make[1]: Leaving directory '/home/u/lede' make: *** [/home/u/lede/include/toplevel.mk:230: world] Error 2
> > 但是python3里import _ssl没问题呀 u@ubuntu:~$ python3 Python 3.8.10 (default, Nov 14 2022, 12:59:47) [GCC 9.4.0] on linux Type "help", "copyright", "credits" or "license" for more information.
> > > > > import _ssl
> > 
> > 
> > 为啥会报No module named '_ssl'
> 
> 我这边也出现了 `fatal: reference is not a tree: a46b63817a9c6ad5af7cf519332e859f11558592`，不知道怎么解决。 SSL 的导入没啥问题

wo ye yudaole tongyang de wenti.laoge ni shi zenme jiejue de ?

closed 2026-09-05 by coolsnowwolf