coolsnowwolf/lede #9347: How to set default user interface language to english when build?

alsheirazz, 2022-05-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：how to set user interface language in luci (all parts) to english? Im from a country outside of China so this is very important.

### (2) 路由器型号和固件版本

A：Xiaomi Redmi AC2100, OpenWrt R22.4.1 (Compiled with Github Actions)

### (3) 详细日志

A：

XiaoliChan, 2022-05-03:
Select Luci → Modules → Translations under luci-base → English

alsheirazz, 2022-05-03:
> Select Luci → Modules → Translations under luci-base → English

tried it but failed when compiling (github actions)

XiaoliChan, 2022-05-03:
> tried it but failed when compiling (github actions)

Any logs?

alsheirazz, 2022-05-03:
```
make[1] world
 make[2] tools/compile
 make[2] package/cleanup
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/tar compile
 make[3] -C tools/patch compile
 make[3] -C tools/m4 compile
 make[3] -C tools/autoconf-archive compile
 make[3] -C tools/pkgconf compile
 make[3] -C tools/ninja compile
 make[3] -C tools/cpio compile
 make[3] -C tools/lzma compile
 make[3] -C tools/meson compile
 make[3] -C tools/mtools compile
 make[3] -C tools/patch-image compile
 make[3] -C tools/sstrip compile
 make[3] -C tools/ucl compile
 make[3] -C tools/xxd compile
 make[3] -C tools/zip compile
 make[3] -C tools/expat compile
 make[3] -C tools/kernel2minor compile
 make[3] -C tools/autoconf compile
 make[3] -C tools/libressl compile
 make[3] -C tools/upx compile
 make[3] -C tools/automake compile
 make[3] -C tools/missing-macros compile
 make[3] -C tools/dosfstools compile
 make[3] -C tools/libtool compile
 make[3] -C tools/cmake compile
 make[3] -C tools/mkimage compile
 make[3] -C tools/mklibs compile
 make[3] -C tools/mm-macros compile
 make[3] -C tools/patchelf compile
 make[3] -C tools/gmp compile
 make[3] -C tools/flex compile
 make[3] -C tools/e2fsprogs compile
 make[3] -C tools/fakeroot compile
 make[3] -C tools/gengetopt compile
 make[3] -C tools/mpfr compile
 make[3] -C tools/bison compile
 make[3] -C tools/mpc compile
 make[3] -C tools/bc compile
 make[3] -C tools/findutils compile
 make[3] -C tools/padjffs2 compile
 make[3] -C tools/quilt compile
 make[3] -C tools/zstd compile
 make[3] -C tools/zlib compile
 make[3] -C tools/firmware-utils compile
 make[3] -C tools/make-ext4fs compile
 make[3] -C tools/mtd-utils compile
 make[3] -C tools/squashfskit4 compile
 make[2] toolchain/compile
 make[3] -C toolchain/gdb compile
 make[3] -C toolchain/binutils compile
 make[3] -C toolchain/fortify-headers compile
 make[3] -C toolchain/gcc/initial compile
 make[3] -C toolchain/kernel-headers compile
 make[3] -C toolchain/musl compile
 make[3] -C toolchain/gcc/final compile
 make[2] target/compile
 make[3] -C target/linux compile
 make[2] buildinfo
 make[2] package/compile
 make[3] diffconfig buildversion feedsversion
 make[3] -C package/libs/libjson-c host-compile
 make[4] scripts/config/conf
 make[3] -C package/libs/toolchain compile
 make[3] -C package/libs/ncurses host-compile
 make[3] -C package/libs/zlib compile
 make[3] -C package/system/fwtool host-compile
 make[3] -C package/system/usign host-compile
 make[3] -C package/system/openwrt-keyring compile
 make[3] -C feeds/luci/modules/luci-base host-compile
 make[3] -C package/utils/lua host-compile
 make[3] -C package/libs/mbedtls compile
make: *** wait: No child processes.  Stop.
make: *** Waiting for unfinished jobs....
make: *** wait: No child processes.  Stop.
Error: Process completed with exit code 1[43](https://github.com/alsheirazz/OpenWrt-Redmi-AC2100/runs/6258057035?check_suite_focus=true#step:11:43).
```

EnnawYang, 2022-05-03:
Do not use ssh in Github Actions

alsheirazz, 2022-05-03:
> Do not use ssh in Github Actions

i actually use it for managing the menuconfig easily especially when adding english translation from luci modules. any alternative for generating the config?

XiaoliChan, 2022-05-03:
Have no idea...

EnnawYang, 2022-05-03:
> > Do not use ssh in Github Actions
> 
> i actually use it for managing the menuconfig easily especially when adding english translation from luci modules. any alternative for generating the config?

Generate .config file frist, then push .config file to your repo Using ssh will make github cancel your workflows.

alsheirazz, 2022-05-04:
Ive done it that way and still, some parts of Luci havent changed (especially the wireless tab). So, the last resort is to move to ImmortalWrt. The firmware has a default English language and the features are the same as Official OpenWrt (maybe there are some changes that I dont know).

closed 2022-05-04 by alsheirazz