coolsnowwolf/lede #9348: package/Makefile:111: recipe for target '/root/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed

MagicStarTrace, 2022-05-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

没找到解决方案

## 2. 详细叙述
OS：Ubuntu 18.04.3
commit version: 60e2759031f1535df62d7687a53b79f68490f894

[image]

### (1) 具体问题
我用Ubuntu 作为编译系统 提示如下

[image]

### (3) 详细日志

Configuring kmod-ifb.
Configuring kmod-igb.
Configuring kmod-igc.
Configuring kmod-bnx2.
Configuring qos-scripts.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * check_data_file_clashes: Package dnsmasq wants to install file /root/lede/build_dir/target-x86_64_musl/root-x86/etc/hotplug.d/ntp/25-dnsmasqsec
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /root/lede/build_dir/target-x86_64_musl/root-x86/etc/init.d/dnsmasq
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /root/lede/build_dir/target-x86_64_musl/root-x86/etc/uci-defaults/50-dnsmasq-migrate-resolv-conf-auto.sh
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /root/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/dnsmasq/dhcp-script.sh
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /root/lede/build_dir/target-x86_64_musl/root-x86/usr/sbin/dnsmasq
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /root/lede/build_dir/target-x86_64_musl/root-x86/usr/share/acl.d/dnsmasq_acl.json
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /root/lede/build_dir/target-x86_64_musl/root-x86/usr/share/dnsmasq/dhcpbogushostname.conf
	But that file is already provided by package  * dnsmasq-full
 * check_data_file_clashes: Package dnsmasq wants to install file /root/lede/build_dir/target-x86_64_musl/root-x86/usr/share/dnsmasq/rfc6761.conf
	But that file is already provided by package  * dnsmasq-full
 * opkg_install_cmd: Cannot install package dnsmasq.
package/Makefile:66: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/root/lede'
package/Makefile:111: recipe for target '/root/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/root/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/root/lede'
/root/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

我应该怎么办？ 这出现了什么问题？

谢谢~

EnnawYang, 2022-05-04:
接下来有请Google翻译

deplives, 2022-05-04:
[image]

这句话拿到Google翻译一下
[image]

然后把它大声的读出来，聪明的你应该能看懂

WYC-2020, 2022-05-04:
就一冲突而已，自己改下就可以

minlang112, 2022-05-04:
dnsmasq 和 dnsmasq-full 冲突了

closed 2022-05-05 by AmadeusGhost