coolsnowwolf/lede #9349: kern.debug kernel: test 512

triatomic, 2022-05-03:
Router: Redmi/xiaomi AX3200/Ax6s
If i disable SFE-offload and use software and hardware offload with the following config:
如果我禁用 SFE 卸载并使用以下配置的软件和硬件卸载：
```
config turboacc 'config'
        option dns_caching '0'
        option bbr_cca '1'
        option fullcone_nat '1'
        option sw_flow '1'
        option hw_flow '1'
```

```
config defaults
        option syn_flood '1'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'REJECT'
        option fullcone '1'
        option flow_offloading '1'
        option flow_offloading_hw '1'
```
Kernel will spam me with the following log:
内核将通过以下日志向我发送垃圾邮件:
```
Tue May  3 23:29:24 2022 kern.debug kernel: [  191.780821] test 512
Tue May  3 23:29:27 2022 kern.debug kernel: [  422.172098] test 512
Tue May  3 23:29:33 2022 kern.debug kernel: [  424.640840] test 512
Tue May  3 23:29:34 2022 kern.debug kernel: [  430.626773] test 512
Tue May  3 23:29:38 2022 kern.debug kernel: [  432.584737] test 512
Tue May  3 23:29:39 2022 kern.debug kernel: [  435.604967] test 512
Tue May  3 23:29:40 2022 kern.debug kernel: [  437.411654] test 512
Tue May  3 23:29:43 2022 kern.debug kernel: [  437.941660] test 512
Tue May  3 23:29:47 2022 kern.debug kernel: [  441.075181] test 512
Tue May  3 23:29:47 2022 kern.debug kernel: [  445.043499] test 512
Tue May  3 23:29:48 2022 kern.debug kernel: [  445.339555] test 512
Tue May  3 23:29:49 2022 kern.debug kernel: [  446.360772] test 512
Tue May  3 23:29:49 2022 kern.debug kernel: [  447.282016] test 512
Tue May  3 23:29:51 2022 kern.debug kernel: [  447.511708] test 512
Tue May  3 23:29:52 2022 kern.debug kernel: [  449.085660] test 512
Tue May  3 23:29:53 2022 kern.debug kernel: [  449.731337] test 512
Tue May  3 23:29:55 2022 kern.debug kernel: [  450.803589] test 512
Tue May  3 23:29:57 2022 kern.debug kernel: [  452.834375] test 512
Tue May  3 23:29:58 2022 kern.debug kernel: [  455.431837] test 512
Tue May  3 23:30:01 2022 kern.debug kernel: [  456.521158] test 512
Tue May  3 23:30:04 2022 kern.debug kernel: [  458.803438] test 512
Tue May  3 23:30:06 2022 kern.debug kernel: [  461.664985] test 512
Tue May  3 23:30:08 2022 kern.debug kernel: [  464.345804] test 512
Tue May  3 23:30:17 2022 kern.debug kernel: [  465.599658] test 512
Tue May  3 23:30:18 2022 kern.debug kernel: [  475.405121] test 512
Tue May  3 23:30:18 2022 kern.debug kernel: [  475.627834] test 512
Tue May  3 23:30:19 2022 kern.debug kernel: [  475.844433] test 512
Tue May  3 23:30:20 2022 kern.debug kernel: [  477.587721] test 512
Tue May  3 23:30:20 2022 kern.debug kernel: [  477.851574] test 512
Tue May  3 23:30:22 2022 kern.debug kernel: [  477.933862] test 512
Tue May  3 23:30:23 2022 kern.debug kernel: [  480.355429] test 512
Tue May  3 23:30:24 2022 kern.debug kernel: [  480.649134] test 512
Tue May  3 23:30:24 2022 kern.debug kernel: [  482.169404] test 512
Tue May  3 23:30:25 2022 kern.debug kernel: [  482.412597] test 512
Tue May  3 23:30:28 2022 kern.debug kernel: [  482.602584] test 512
Tue May  3 23:30:34 2022 kern.debug kernel: [  486.082955] test 512
Tue May  3 23:30:37 2022 kern.debug kernel: [  492.590123] test 512
Tue May  3 23:30:38 2022 kern.debug kernel: [  495.552340] test 512
Tue May  3 23:30:39 2022 kern.debug kernel: [  495.927559] test 512
Tue May  3 23:30:40 2022 kern.debug kernel: [  497.168878] test 512
Tue May  3 23:30:40 2022 kern.debug kernel: [  497.827117] test 512
Tue May  3 23:30:41 2022 kern.debug kernel: [  497.951015] test 512
Tue May  3 23:30:46 2022 kern.debug kernel: [  498.702858] test 512
Tue May  3 23:30:50 2022 kern.debug kernel: [  503.805383] test 512
Tue May  3 23:30:50 2022 kern.debug kernel: [  507.594384] test 512
Tue May  3 23:30:58 2022 kern.debug kernel: [  507.696067] test 512
Tue May  3 23:30:58 2022 kern.debug kernel: [  515.954697] test 512
Tue May  3 23:31:00 2022 kern.debug kernel: [  516.366323] test 512
Tue May  3 23:31:00 2022 kern.debug kernel: [  517.930759] test 512
Tue May  3 23:31:00 2022 kern.debug kernel: [  518.531003] test 512
Tue May  3 23:31:03 2022 kern.debug kernel: [  518.578057] test 512
Tue May  3 23:31:05 2022 kern.debug kernel: [  520.630081] test 512
Tue May  3 23:31:05 2022 kern.debug kernel: [  522.598567] test 512
Tue May  3 23:31:06 2022 kern.debug kernel: [  523.258477] test 512
Tue May  3 23:31:09 2022 kern.debug kernel: [  524.177738] test 512
Tue May  3 23:31:09 2022 kern.debug kernel: [  527.195179] test 512
Tue May  3 23:31:11 2022 kern.debug kernel: [  527.414795] test 512
Tue May  3 23:31:20 2022 kern.debug kernel: [  529.504595] test 512
Tue May  3 23:31:22 2022 kern.debug kernel: [  537.601968] test 512
Tue May  3 23:31:24 2022 kern.debug kernel: [  540.554561] test 512
Tue May  3 23:31:25 2022 kern.debug kernel: [  542.172335] test 512
Tue May  3 23:31:25 2022 kern.debug kernel: [  542.828276] test 512
Tue May  3 23:31:26 2022 kern.debug kernel: [  542.953634] test 512
Tue May  3 23:31:31 2022 kern.debug kernel: [  543.735044] test 512
Tue May  3 23:31:35 2022 kern.debug kernel: [  548.807384] test 512
Tue May  3 23:31:35 2022 kern.debug kernel: [  552.604184] test 512
Tue May  3 23:31:35 2022 kern.debug kernel: [  552.781682] test 512
Tue May  3 23:31:35 2022 kern.debug kernel: [  553.170294] test 512
Tue May  3 23:31:39 2022 kern.debug kernel: [  553.341300] test 512
Tue May  3 23:31:40 2022 kern.debug kernel: [  557.168237] test 512
Tue May  3 23:31:44 2022 kern.debug kernel: [  557.936095] test 512
Tue May  3 23:31:44 2022 kern.debug kernel: [  561.782959] test 512
Tue May  3 23:31:48 2022 kern.debug kernel: [  562.545483] test 512
Tue May  3 23:31:50 2022 kern.debug kernel: [  565.631696] test 512
Tue May  3 23:31:52 2022 kern.debug kernel: [  567.607313] test 512
Tue May  3 23:31:54 2022 kern.debug kernel: [  570.551447] test 512
Tue May  3 23:31:55 2022 kern.debug kernel: [  572.416797] test 512
Tue May  3 23:31:59 2022 kern.debug kernel: [  572.881839] test 512
```

masterc24, 2022-05-18:
同样的问题

triatomic, 2022-05-23:
any help ?
有什么帮助吗？

drw2005, 2022-06-16:
2022.06.13版本也有同样的问题
5.10.117内核

各位解决这个问题了吗？

iiQi, 2022-08-31:
我7月编译的版本 `Powered by LuCI Master (git-22.161.56906-efa5cfe) / OpenWrt R22.6.16` 没有这个问题，现在 master 上的最新代码又有这个问题了

drw2005, 2022-09-01:
我29号编译的这个版本，目前为止没有问题了：OpenWrt R22.8.2，R2S

Ais-Danny, 2024-01-06:
OpenWrt R23.11.11 / LuCI Master (git-23.361.43243-f3fe5b5) 版本，这个问题依旧存在（内核5.10.205）