coolsnowwolf/lede #9350: 最新源码编译报错，没搜索到相关问题

82kg, 2022-05-04:
ninja: build stopped: subcommand failed.
---------------------------------------------
Error when bootstrapping CMake:
Problem while running /home/sun/lede/staging_dir/host/bin/ninja
---------------------------------------------
Log of errors: /home/sun/lede/build_dir/host/cmake-3.22.2/Bootstrap.cmk/cmake_bootstrap.log
---------------------------------------------
make[3]: *** [Makefile:53: /home/sun/lede/build_dir/host/cmake-3.22.2/.configured] Error 9
make[3]: Leaving directory '/home/sun/lede/tools/cmake'
time: tools/cmake/compile#69.98#19.64#13.68
    ERROR: tools/cmake failed to build.
make[2]: *** [tools/Makefile:162: tools/cmake/compile] Error 1
make[2]: Leaving directory '/home/sun/lede'
make[1]: *** [tools/Makefile:158: /home/sun/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyy                                    yyynyynnyynynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/sun/lede'
make: *** [/home/sun/lede/include/toplevel.mk:230：world] 错误 2

cmake_bootstrap.log 文件如下

[cmake_bootstrap.log](https://github.com/coolsnowwolf/lede/files/8617737/cmake_bootstrap.log)

求大佬指点。

编译环境是 Debian 11

mythpan1, 2022-05-14:
ninja: build stopped: subcommand failed.
---------------------------------------------
Error when bootstrapping CMake:
Problem while running /home/mythpan/lede/staging_dir/host/bin/ninja
---------------------------------------------
Log of errors: /home/mythpan/lede/build_dir/host/cmake-3.22.2/Bootstrap.cmk/cmake_bootstrap.log
---------------------------------------------
make[3]: *** [Makefile:53: /home/mythpan/lede/build_dir/host/cmake-3.22.2/.configured] Error 9
make[3]: Leaving directory '/home/mythpan/lede/tools/cmake'
time: tools/cmake/compile#360.73#143.18#148.55
    ERROR: tools/cmake failed to build.
make[2]: *** [tools/Makefile:162: tools/cmake/compile] Error 1
make[2]: Leaving directory '/home/mythpan/lede'
make[1]: *** [tools/Makefile:158: /home/mythpan/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyynynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/mythpan/lede'
make: *** [/home/mythpan/lede/include/toplevel.mk:230: world] Error 2

cmake_bootsrap.log 
[cmake_bootstrap.log](https://github.com/coolsnowwolf/lede/files/8692456/cmake_bootstrap.log)

我应该也是一样的问题,

编译环境是 Ubuntu 22.04 LTS server版

BoveService, 2022-07-20:
I have faced the same issue by using the azure cloud, Debian 11 to make

huyi51462, 2022-08-05:
[https://github.com/coolsnowwolf/lede/issues/8951#issuecomment-1052324937](url)

huyi51462, 2022-08-05:
内存太小,增大内存后通过编译 实测 -j28的情况下 16G内存不够用 -j20 编译通过了

chenxqiyu, 2022-10-15:
http://releases.ubuntu.com/20.04/
安装虚拟机用这个系统编译
http://releases.ubuntu.com/20.04/ubuntu-20.04.5-desktop-amd64.iso

chenxqiyu, 2022-10-15:
> ## ninja: build stopped: subcommand failed.
> ## Error when bootstrapping CMake:
> Problem while running /home/sun/lede/staging_dir/host/bin/ninja
> ## Log of errors: [...]

用Ubuntu 20.04.5

chenxqiyu, 2022-10-15:
> ## ninja: build stopped: subcommand failed.
> ## Error when bootstrapping CMake:
> Problem while running /home/mythpan/lede/staging_dir/host/bin/ninja
> ## Log of errors: [...]

用Ubuntu 20.04.5

chenxqiyu, 2022-10-15:
> I have faced the same issue by using the azure cloud, Debian 11 to make

use Ubuntu 20.04.5

zerofo, 2023-03-21:
```
touch /home/zerofo/lede/build_dir/host/ninja-1.11.1/.prepared6aba017a20ab188719fe7b6ed25719b2_6664517399ebbbc92a37c5bb081b5c53
touch /home/zerofo/lede/build_dir/host/ninja-1.11.1/.configured
cd /home/zerofo/lede/build_dir/host/ninja-1.11.1 && CXX="g++" CXXFLAGS=" -I/home/zerofo/lede/staging_dir/host/include " LDFLAGS="-L/home/zerofo/lede/staging_dir/host/lib " /home/zerofo/lede/staging_dir/host/bin/python3 configure.py --bootstrap
bootstrapping ninja...
warning: A compatible version of re2c (>= 0.11.3) was not found; changes to src/*.in.cc will not affect your build.
./src/ninja.cc: In constructor '{anonymous}::DeferGuessParallelism::DeferGuessParallelism(BuildConfig*)':
./src/ninja.cc:1400:16: warning: '{anonymous}::DeferGuessParallelism::config' will be initialized after [-Wreorder]
 1400 |   BuildConfig* config;
      |                ^~~~~~
./src/ninja.cc:1399:8: warning:   'bool {anonymous}::DeferGuessParallelism::needGuess' [-Wreorder]
 1399 |   bool needGuess;
      |        ^~~~~~~~~
./src/ninja.cc:1402:3: warning:   when initialized here [-Wreorder]
 1402 |   DeferGuessParallelism(BuildConfig* config)
      |   ^~~~~~~~~~~~~~~~~~~~~
wrote build.ninja.
bootstrap complete.  rebuilding...
[17/34] CXX build/state.o
FAILED: build/state.o
g++ -MMD -MT build/state.o -MF build/state.o.d -g -Wall -Wextra -Wno-deprecated -Wno-missing-field-initializers -Wno-unused-parameter -fno-rtti -fno-exceptions -fvisibility=hidden -pipe '-DNINJA_PYTHON="python3"' -O2 -DNDEBUG -fdiagnostics-color -DUSE_PPOLL -DNINJA_HAVE_BROWSE -I.   -I/home/zerofo/lede/staging_dir/host/include  -c src/state.cc -o build/state.o
In file included from /usr/include/wchar.h:35,
                 from /usr/include/c++/11/cwchar:44,
                 from /usr/include/c++/11/bits/postypes.h:40,
                 from /usr/include/c++/11/bits/char_traits.h:40,
                 from /usr/include/c++/11/string:40,
                 from src/state.h:20,
                 from src/state.cc:15:
/usr/lib/gcc/x86_64-linux-gnu/11/include/stddef.h:209:1: internal compiler error: Segmentation fault
  209 | typedef __SIZE_TYPE__ size_t;
      | ^~~~~~~
0xe3320d internal_error(char const*, ...)
        ???:0
0xf2067c ht_lookup_with_hash(ht*, unsigned char const*, unsigned long, unsigned int, ht_lookup_option)
        ???:0
0xf45669 _cpp_lex_direct
        ???:0
0xf44ee4 _cpp_lex_token
        ???:0
0xf4a3ef _cpp_handle_directive
        ???:0
0xf44fe4 _cpp_lex_token
        ???:0
0xf6496b c_lex_with_flags(tree_node**, unsigned int*, unsigned char*, int)
        ???:0
0x149e543 c_parse_file()
        ???:0
0x148c69e c_common_parse_file()
        ???:0
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-11/README.Bugs> for instructions.
[26/34] CXX build/missing_deps.o
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/home/zerofo/lede/build_dir/host/ninja-1.11.1/configure.py", line 725, in <module>
    subprocess.check_call(rebuild_args)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['./ninja']' returned non-zero exit status 1.
make[3]: *** [Makefile:40: /home/zerofo/lede/build_dir/host/ninja-1.11.1/.built] Error 1
make[3]: Leaving directory '/home/zerofo/lede/tools/ninja'
time: tools/ninja/compile#16.60#1.83#11.38
    ERROR: tools/ninja failed to build.
make[2]: *** [tools/Makefile:219: tools/ninja/compile] Error 1
make[2]: Leaving directory '/home/zerofo/lede'
make[1]: *** [tools/Makefile:215: /home/zerofo/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyy] Error 2
make[1]: Leaving directory '/home/zerofo/lede'
make: *** [/home/zerofo/lede/include/toplevel.mk:231: world] Error 2
```

coolsnowwolf, 2026-09-05:
这是旧版 CMake 3.22 在 Debian/Ubuntu 主机上的 bootstrap 失败，评论已确认使用项目支持的 Ubuntu 20.04 环境可正常编译；没有完整 bootstrap 日志中的首个编译错误，也没有当前源码复现条件。

closed 2026-09-05 by coolsnowwolf