coolsnowwolf/lede #9351: 星际宝盒无法挂载 Extroot (/overlay) 及解决方案

KumaTea, 2022-05-04:
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 星际宝盒无法挂载 Extroot (/overlay) 及解决方案

* 路由器型号 星际宝盒 `MobiPromo CM520-79F`
* 源码版本 最新 (`OpenWrt R22.4.1 / LuCI Master (git-22.115.51709-a2bd654)`)

### 问题描述

自从使用该路由以来，我就发现任何版本的源码都无法挂载 Extroot，即通过外置 USB 扩充 `/overlay` 空间。

表现为在 `系统 - 挂载点` 设置 `/dev/sda1` 指向 `/overlay` 并重启后无效，指向的仍然是 `/dev/ubi0_2`。

`dmesg` 可发现该错误出现两次：

```log
Wed May  4 15:57:31 2022 user.info kernel: [   16.675145] block: attempting to load /etc/config/fstab
Wed May  4 15:57:31 2022 user.err kernel: [   16.687507] block: unable to load configuration (fstab: Entry not found)

Wed May  4 15:57:31 2022 user.info kernel: [   17.970782] block: attempting to load /etc/config/fstab
Wed May  4 15:57:31 2022 user.info kernel: [   17.972443] block: unable to load configuration (fstab: Entry not found)
```

~~在闪存写炸 (变只读) 之后我终于下决心修好它~~

### 解决方案

在网上搜索上述错误，发现此 issue:

**openwrt/openwrt#7352**

其中 _uutrumo_ [提到](https://github.com/openwrt/openwrt/issues/7352#issuecomment-1040314349)：

```diff
@@ -1591,7 +1591,7 @@ static int main_extroot(int argc, char *
 #endif
 
 	/* As a last resort look for /etc/config/fstab on "rootfs" partition */
-	return mount_extroot(NULL);
+	return mount_extroot("/tmp/overlay");
 }
 
 static int main_mount(int argc, char **argv)
```

但由于L大的源已有 [冲突 patch](https://github.com/coolsnowwolf/lede/blob/master/package/system/fstools/patches/0001-fstools-support-extroot-for-non-MTD-rootfs_data.patch)，故作修改。

---

1. `.config` 替换 `# CONFIG_FSTOOLS_UBIFS_EXTROOT is not set` 为 `CONFIG_FSTOOLS_UBIFS_EXTROOT=y`
2. 在 `lede/package/system/fstools/patches` 添加 `0002-mount-tmp-overlay.patch` 如下

```diff
--- a/block.c
+++ b/block.c
@@ -1591,7 +1591,7 @@ static int main_extroot(int argc, char *
        /* As a last resort look for /etc/config/fstab on "rootfs" partition */
-       return mount_extroot(NULL);
+       return mount_extroot("/tmp/overlay");
 }

 static int main_mount(int argc, char **argv)
```

3. 编译

```shell
make tools/install
make toolchain/install
make package/system/fstools/compile
```

4. 到 `bin\targets\ipq40xx\generic\packages` 提取 `block-mount` 和 `fstools` 的 ipk 文件，上传路由器后用 `opkg install --force-reinstall` 覆盖安装
5. 复制现有 `/overlay` 到U盘：`tar -C /overlay -cvf - . | tar -C /mnt/sda1 -xf -`

重启后可见已成功挂载

[image]

---

因为我对 pull requests 不是太熟悉，还望将上述第二步的 patch 合并，非常感谢！

WYC-2020, 2022-05-05:
格式改成f2fs 也不行？

KumaTea, 2022-05-05:
> 格式改成f2fs 也不行？

以前应该是不可以的，过几天我会抽空试试

WYC-2020, 2022-05-05:
> > 格式改成f2fs 也不行？
> 
> 以前应该是不可以的，过几天我会抽空试试

这你还不要说，我刚测试居然改到其他盘符后overlay没法再还原回来了，除非把那个指定的盘符删除掉重启，然后就回到loop这个文件夹了，直接在挂载哪里取消没用，重启还是会在那个盘符，搞笑哈哈

WYC-2020, 2022-05-07:
> > 格式改成f2fs 也不行？
> 
> 以前应该是不可以的，过几天我会抽空试试

你的所有格式 都必须在op上面格式化，这样才能被op识别挂载，我试过没问题了，不需要按你下面那样改，并且我说的那个 没法恢复是一个问题，所以我自己加了功能实现恢复到默认loop分区
[image]

WYC-2020, 2022-05-07:
[image]
l大的这个补丁就是修复那个扩展overlay的

cxgreat2014, 2023-01-28:
极路由 b70 (HiWifi HC5962) 实测原版代码无法正常挂载overlay，使用楼主的解决方案可以有效解决问题

@WYC-2020 不知道有空能帮忙看一下这个问题的妥善解决方案吗？

cxgreat2014, 2023-01-30:
在这里 https://github.com/openwrt/openwrt/issues/7352 发现有人提供了fstools的相关补丁： https://github.com/domo-iot/fstools/tree/extroot-squashfs ，不知道@coolsnowwolf 可以看看能否合并修改吗？

coolsnowwolf, 2023-01-30:
我这边测试下

topeqin, 2023-04-04:
请问如果要在v21.02的版本 { squash + ext4  partition for overlay)
Thu Apr  4 15:01:51 2023 user.info kernel: [   10.119972] block: attempting to load /etc/config/fstab
Thu Apr  4 15:01:51 2023 user.err kernel: [   10.120211] block: unable to load configuration (fstab: Entry not found)
Thu Apr  4 15:01:51 2023 user.err kernel: [   10.124238] block: no usable configuration

是否 0001-fstools-support-extroot-for-non-MTD-rootfs_data.patch 和 这个新的 0002-mount-tmp-overlay.patch 都需要？
thanks

cxgreat2014, 2023-04-04:
> 我这边测试下

好的，谢谢，不知道测试结果如何？需要提供测试/设备以及其他方面的帮助吗？

topeqin, 2023-04-05:
emmc flash,  squash /mmcblk0pX (/dev/root  主分区),   ext4 /mmcblk0pY （想挂载为 /overlay） 目前openwrt 不论是v19 还是v21 不打patch 一定是不行的。默认overaly会在 jffs2 ubi  等mtd 分区上。 谢谢！

coolsnowwolf, 2026-09-05:
当前 `fstools` 已使用包含 extroot 修复的上游版本，`mount_extroot("/tmp/overlay")` 逻辑已在历史同步中合入，不再需要 issue 中手工添加的 0002 补丁。旧问题对应 OpenWrt 22.04/2022 版本，按已修复关闭。

closed 2026-09-05 by coolsnowwolf