coolsnowwolf/lede #9354: kmod-ath11k_5.4.191+5.15.8-1-2_arm_xscale.ipk Error 1  ath11k.ko 缺失

TonyTeng22, 2022-05-05:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue
 ath11k.ko 缺失 

## 2. 详细叙述
2022年5月：kmod-ath11k_5.4.191+5.15.8-1-2_arm_xscale.ipk comple Error  ath11k.ko 缺失 
### (1) 具体问题

A：make 加入kmod-ath11k_5.4.191+5.15.8-1-2_arm_xscale.ipk comple Error  ath11k.ko 缺失 

### (2) 路由器型号和固件版本

A：kirwood 

### (3) 详细日志

A：
touch /media/tony/HDD/lede/build_dir/target-arm_xscale_musl_eabi/linux-kirkwood/backports-5.15.8-1/.pkgdir/kmod-ath10k.installed
mkdir -p /media/tony/HDD/lede/bin/targets/kirkwood/generic/packages /media/tony/HDD/lede/build_dir/target-arm_xscale_musl_eabi/linux-kirkwood/backports-5.15.8-1/ipkg-arm_xscale/kmod-ath11k/CONTROL /media/tony/HDD/lede/staging_dir/target-arm_xscale_musl_eabi/pkginfo
ERROR: module '/media/tony/HDD/lede/build_dir/target-arm_xscale_musl_eabi/linux-kirkwood/backports-5.15.8-1/drivers/net/wireless/ath/ath11k/ath11k.ko' is missing.
make[3]: *** [Makefile:555: /media/tony/HDD/lede/bin/targets/kirkwood/generic/packages/kmod-ath11k_5.4.191+5.15.8-1-2_arm_xscale.ipk] Error 1
make[3]: Leaving directory '/media/tony/HDD/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#308.10#84.04#249.18
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/media/tony/HDD/lede'
make[1]: *** [package/Makefile:110: /media/tony/HDD/lede/staging_dir/target-arm_xscale_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/tony/HDD/lede'
make: *** [/media/tony/HDD/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-05-05:
什么玩意，你这路由器什么芯片，确定可以选择ath11，不是看到的所有驱动都能选择的，小伙子,有的驱动屏蔽做不好，导致可以看到，但是能看到不代表可以支持这个型号的op

TonyTeng22, 2022-05-05:
> 
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.191 (tony@tony) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4465-5f81e2898)) #0 Fri Apr 29 14:56:09 2022
[    0.000000] CPU: Feroceon 88FR131 [56251311] revision 1 (ARMv5TE), cr=0005397f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] OF: fdt: Machine model: BUTONGWIFI for Marvell 88F6281
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[    0.000000] Kernel command line: console=ttyS0,115200 mtdparts=nand_mtd:0xc0000@0(uboot)ro,0x7f00000@0x100000(root) root=/dev/mtdblock2 rootfstype=squashfs
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 510608K/524288K available (5245K kernel code, 170K rwdata, 1392K rodata, 1024K init, 206K bss, 13680K reserved, 0K cma-reserved)

Marvell 88F6281 512M RAM 128M Nand Flash + MiniPCIE

TonyTeng22, 2022-05-05:
> [About ath11k](https://wireless.wiki.kernel.org/en/users/drivers/ath11k) ath11k is a Linux driver for Qualcomm [IEEE 802.11ax](https://en.wikipedia.org/wiki/IEEE_802.11ax) devices, supporting both AHB bus in SoC type devices and PCI. ath11k uses mac80211. It was first included in Linux v5.6 released March 2020.

[image]

closed 2022-05-10 by AmadeusGhost