coolsnowwolf/lede #9355: 编译R4S固件报错，actions和物理机报错一样

jumpdang, 2022-05-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
*  没有类似的issue

## 2. 详细叙述

### (1) 具体问题
编译R4S固件报错，actions和物理机报错一样
A：

### (2) 路由器型号和固件版本
R4S
A：

### (3) 详细日志
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.37/scripts/dtc/dtc -O dtb -i/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.37/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus.dtb /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus.dtb.tmp
rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-guangmiao-g4c.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.37/arch/arm64/boot/dts -I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.37/arch/arm64/boot/dts/include -I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.37/include/ -undef -D__DTS__  -o /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus-lts.dtb.tmp /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.37/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.37/scripts/dtc/dtc -O dtb -i/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.37/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus-lts.dtb /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus-lts.dtb.tmp
rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus.dtb.tmp
rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-orangepi-r1-plus-lts.dtb.tmp
mkdir -p /workdir/openwrt/bin/targets/rockchip/armv8 /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/tmp
rm -rf /workdir/openwrt/build_dir/target-aarch64_generic_musl/json_info_files
/workdir/openwrt/staging_dir/host/bin/mksquashfs4 /workdir/openwrt/build_dir/target-aarch64_generic_musl/root-rockchip /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.squashfs -nopad -noappend -root-owned -comp xz -Xpreset 9 -Xe -Xlc 0 -Xlp 2 -Xpb 2  -b 1024k -p '/dev d 755 0 0' -p '/dev/console c 600 0 0 5 1' -no-xattrs -processors 2
/workdir/openwrt/staging_dir/host/bin/make_ext4fs -L rootfs -l 167772160 -b 4096 -m 0 -J -T 1651683814 /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.ext4 /workdir/openwrt/build_dir/target-aarch64_generic_musl/root-rockchip/
error: ext4_allocate_best_fit_partial: failed to allocate 9475 blocks, out of space?
Creating filesystem with parameters:
    Size: 167772160
    Block size: 4096
    Blocks per group: 32768
make[5]: *** [/workdir/openwrt/include/image.mk:336: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.ext4] Error 1
make[5]: *** Waiting for unfinished jobs....
    Inodes per group: 5120
    Inode size: 256
    Journal blocks: 0
    Label: rootfs
    Blocks: 40960
    Block groups: 2
    Reserved blocks: 0
    Reserved block group size: 15
Pseudo file "/dev" exists in source filesystem "/workdir/openwrt/build_dir/target-aarch64_generic_musl/root-rockchip/dev".
Ignoring, exclude it (-e/-ef) to override.
Parallel mksquashfs: Using 2 processors
Creating 4.0 filesystem on /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.squashfs, block size 1048576.

[============\                                                 ] 1100/5337  20%
[================-                                             ] 1400/5337  26%
[=============================================================\] 5337/5337 100%

Exportable Squashfs 4.0 filesystem, xz compressed, data block size 1048576
	compressed data, compressed metadata, compressed fragments,
	no xattrs, compressed ids
	duplicates are removed
Filesystem size 119387.63 Kbytes (116.59 Mbytes)
	27.36% of uncompressed filesystem size (436437.22 Kbytes)
Inode table size 40582 bytes (39.63 Kbytes)
	20.50% of uncompressed inode table size (197988 bytes)
Directory table size 55426 bytes (54.13 Kbytes)
	40.01% of uncompressed directory table size (138536 bytes)
Number of duplicate files found 914
Number of inodes 5940
Number of files 5029
Number of fragments 77
Number of symbolic links  541
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 369
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
make[5]: Leaving directory '/workdir/openwrt/target/linux/rockchip/image'
make[4]: Leaving directory '/workdir/openwrt/target/linux/rockchip'
make[4]: *** [Makefile:25: install] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#169.91#3.70#93.30
make[3]: *** [Makefile:11: install] Error 2
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:19: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.
A：

EnnawYang, 2022-05-05:
`error: ext4_allocate_best_fit_partial: failed to allocate 9475 blocks, out of space?`

WYC-2020, 2022-05-05:
乱增加ipk 容量超过限制了吧

xspeed1989, 2022-05-08:
调整下容量就行了

closed 2022-05-08 by AmadeusGhost