coolsnowwolf/lede #9356: 多Wan 多DDNS问题

xjsolid, 2022-05-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
我设置了两个 wan pppoe拨号, 获取到了两个IP, 都是公网IP, 比如 1.1.1.1和1.1.1.2
我在花生壳上设置了两个域名的解析比如 www.a.cn, www.b.cn
路由器设置了两个DDNS 分别更新两个域名的IP, 有时候两个域名都能成功解析,
但是有时候某个域名解析失败,DDNS上的IP显示 0.0.0.0
此时ping这个域名显示未知的主机.
看系统log 如下

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：
固件版本 | OpenWrt R22.4.1 / LuCI Master (git-22.121.65028-2a5da72)
内核版本 | 5.15.36

### (3) 详细日志

A：
Thu May  5 08:58:16 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 08:58:32 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 09:00:00 2022 cron.err crond[8701]: USER root pid 15457 cmd /usr/share/scripts/oray.sh >>/tmp/oray.log 2>&1
Thu May  5 09:03:17 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 2/0
Thu May  5 09:03:33 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 2/0
Thu May  5 09:08:17 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 3/0
Thu May  5 09:08:33 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 3/0
Thu May  5 09:13:18 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 4/0
Thu May  5 09:18:19 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 5/0
Thu May  5 09:18:34 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 09:23:20 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 6/0
Thu May  5 09:28:21 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 7/0
Thu May  5 09:28:36 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 09:30:00 2022 cron.err crond[8701]: USER root pid 657 cmd /usr/share/scripts/oray.sh >>/tmp/oray.log 2>&1
Thu May  5 09:33:22 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 8/0
Thu May  5 09:33:37 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 2/0
Thu May  5 09:38:22 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 9/0
Thu May  5 09:38:37 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 3/0
Thu May  5 09:43:23 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 10/0
Thu May  5 09:43:38 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 4/0
Thu May  5 09:53:24 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 09:58:25 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 2/0
Thu May  5 09:58:40 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 10:00:00 2022 cron.err crond[8701]: USER root pid 17729 cmd /usr/share/scripts/oray.sh >>/tmp/oray.log 2>&1
Thu May  5 10:08:27 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 10:13:27 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 2/0
Thu May  5 10:13:41 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 10:18:28 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 3/0
Thu May  5 10:23:29 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 4/0
Thu May  5 10:30:00 2022 cron.err crond[8701]: USER root pid 32102 cmd /usr/share/scripts/oray.sh >>/tmp/oray.log 2>&1
Thu May  5 10:38:31 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 10:43:32 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 2/0
Thu May  5 10:53:33 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 11:00:00 2022 cron.err crond[8701]: USER root pid 12726 cmd /usr/share/scripts/oray.sh >>/tmp/oray.log 2>&1
Thu May  5 11:08:46 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 11:28:36 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 11:30:00 2022 cron.err crond[8701]: USER root pid 25860 cmd /usr/share/scripts/oray.sh >>/tmp/oray.log 2>&1
Thu May  5 11:38:48 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 11:43:37 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 11:58:50 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 12:00:00 2022 cron.err crond[8701]: USER root pid 6441 cmd /usr/share/scripts/oray.sh >>/tmp/oray.log 2>&1
Thu May  5 12:03:51 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 2/0
Thu May  5 12:08:39 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 12:08:52 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 3/0
Thu May  5 12:13:25 2022 daemon.err miniupnpd[4642]: recv (state0): Connection reset by peer
Thu May  5 12:13:53 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 4/0
Thu May  5 12:18:40 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 12:18:54 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 5/0
Thu May  5 12:23:41 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 2/0
Thu May  5 12:28:43 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 3/0
Thu May  5 12:30:00 2022 cron.err crond[8701]: USER root pid 22748 cmd /usr/share/scripts/oray.sh >>/tmp/oray.log 2>&1
Thu May  5 12:38:56 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 12:48:57 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 12:58:45 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 13:00:00 2022 cron.err crond[8701]: USER root pid 3595 cmd /usr/share/scripts/oray.sh >>/tmp/oray.log 2>&1
Thu May  5 13:03:46 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 2/0
Thu May  5 13:08:47 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 3/0
Thu May  5 13:23:48 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 13:30:00 2022 cron.err crond[8701]: USER root pid 17583 cmd /usr/share/scripts/oray.sh >>/tmp/oray.log 2>&1
Thu May  5 13:34:07 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 13:48:51 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 13:49:08 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 14:00:00 2022 cron.err crond[8701]: USER root pid 30638 cmd /usr/share/scripts/oray.sh >>/tmp/oray.log 2>&1
Thu May  5 14:04:10 2022 user.warn ddns-scripts[9278]: proate_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 14:13:53 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 1/0
Thu May  5 14:18:54 2022 user.warn ddns-scripts[5832]: xplushome_ipv4: Updating IP at DDNS provider failed - starting retry 2/0

WYC-2020, 2022-05-05:
咋看都是感觉失败

GreatMichaelLee, 2022-05-13:
实测没有任何问题，我3拨，三个IPv4，3个VWAN.都配了DDNS. 用的dnspod同一个token.

[image]

xjsolid, 2022-05-13:
> 实测没有任何问题，我3拨，三个IPv4，3个VWAN.都配了DDNS. 用的dnspod同一个token.
> 
> [image]

大佬能提供下版本吗?

GreatMichaelLee, 2022-05-15:
自编译版本啊，这个仓库的code啊，那就是个脚本，应该没怎么动，你那一看就是你自己的网络配置问题，跟啥版本没关系。

coolsnowwolf, 2026-09-05:
多 WAN 多 DDNS 失败日志只有 provider 更新失败，没有接口绑定、地址来源、服务配置或 provider 响应；无法确认 ddns-scripts 缺陷，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf