coolsnowwolf/lede #9358: 红米ac2100 5.15内核编译不通过

zixing131, 2022-05-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

linux5.15内核编译失败。linux5.4是可以编译的。

### (1) 具体问题

提示rt_rbus_pci_util.c中找不到pci_enable_msi定义，然后把相关的两行注释掉是可以编译成功的，但是刷入后无法启动。

A：

### (2) 路由器型号和固件版本
A：

红米ac2100，linux5.15内核，[4ddd124](https://github.com/coolsnowwolf/lede/commit/4ddd124ec853ed44b6a4c22ab02b4871863388d9)
### (3) 详细日志

A：

 /home/zixing/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/os/linux/rt_rbus_pci_util.o /home/zixing/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/os/linux/rt_rbus_pci_util.c
/home/zixing/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/os/linux/rt_rbus_pci_util.c: In function 'RtmpOsPciMsiEnable':
/home/zixing/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/os/linux/rt_rbus_pci_util.c:484:9: error: implicit declaration of function 'pci_enable_msi'; did you mean 'pci_enable_sriov'? [-Werror=implicit-function-declaration]
  return pci_enable_msi(pDev);
         ^~~~~~~~~~~~~~
         pci_enable_sriov
/home/zixing/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/os/linux/rt_rbus_pci_util.c: In function 'RtmpOsPciMsiDisable':
/home/zixing/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/os/linux/rt_rbus_pci_util.c:502:2: error: implicit declaration of function 'pci_disable_msi'; did you mean 'pci_disable_sriov'? [-Werror=implicit-function-declaration]
  pci_disable_msi(pDev);
  ^~~~~~~~~~~~~~~
  pci_disable_sriov
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:277: /home/zixing/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/os/linux/rt_rbus_pci_util.o] Error 1
make[4]: *** [Makefile:1868: /home/zixing/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap] Error 2
make[4]: Leaving directory '/home/zixing/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.37'
make[3]: *** [Makefile:254: /home/zixing/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/.built] Error 2
make[3]: Leaving directory '/home/zixing/lede/package/lean/mt/drivers/mt7615d'
time: package/lean/mt/drivers/mt7615d/compile#80.87#9.07#87.20
    ERROR: package/lean/mt/drivers/mt7615d failed to build.
make[2]: *** [package/Makefile:116: package/lean/mt/drivers/mt7615d/compile] Error 1
make[2]: Leaving directory '/home/zixing/lede'
make[1]: *** [package/Makefile:110: /home/zixing/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zixing/lede'
make: *** [/home/zixing/lede/include/toplevel.mk:230：world] 错误 2

liwenjie119, 2022-05-06:
这个问题之前就提了，但是不知道为啥大佬总说没问题，但是这个库的确能复现，其他人的仓库没有这个问题

yuanlun1983, 2022-05-07:
小米3g用5.15重来就没成功过

WYC-2020, 2022-05-07:
驱动不适配而已，修复一下就可的

yuanlun1983, 2022-05-07:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

liwenjie119, 2022-05-07:
> 驱动不适配而已，修复一下就可的

大佬有空了修一修，我个小白搞了很久没修好

WYC-2020, 2022-05-07:
> > 驱动不适配而已，修复一下就可的
> 
> 大佬有空了修一修，我个小白搞了很久没修好

无能为力我用自己仓库编译了，亲测一把过，并且我看了linux 源码这个是有定义这个函数的，你们自己万能大法试下

WYC-2020, 2022-05-07:
[image]

WYC-2020, 2022-05-07:
> 小米3g用5.15重来就没成功过

同样亲测没任何问题，默认配置编译

liwenjie119, 2022-05-07:
> > > 驱动不适配而已，修复一下就可的
> > 
> > 
> > 大佬有空了修一修，我个小白搞了很久没修好
> 
> 无能为力我用自己仓库编译了，亲测一把过，并且我看了linux 源码这个是有定义这个函数的，你们自己万能大法试下

我试了很多人的仓库，就lean的没编译通过，但是我也没对比出来是哪的问题， 拉取你的仓库target/linux/ramips以及target/linux/generic替换到lean也出现这个报错

zixing131, 2022-05-07:
> 

如果大佬有空的话，可以尝试虚拟机重新装一个ubuntu20.04然后重新拉代码编译试试，可能是现有的环境有影响

yuanlun1983, 2022-05-07:
> > 小米3g用5.15重来就没成功过
> 
> 同样亲测没任何问题，默认配置编译

好的，用你的源码试一试

WYC-2020, 2022-05-07:
> > 
> 
> 如果大佬有空的话，可以尝试虚拟机重新装一个ubuntu20.04然后重新拉代码编译试试，可能是现有的环境有影响

那不清楚 我用的ubuntu 18.04，有空在测试20.04

WYC-2020, 2022-05-07:
> > 
> 
> 如果大佬有空的话，可以尝试虚拟机重新装一个ubuntu20.04然后重新拉代码编译试试，可能是现有的环境有影响

十之八九和系统版本没关系，因为op都是全套自己编译的，工具链都是用自己编译的，和外部的环境牵扯甚少，我安装好了20.04 全程没遇到任何问题，等我编译完告诉你答案

zixing131, 2022-05-07:
> > > 
> > 
> > 
> > 如果大佬有空的话，可以尝试虚拟机重新装一个ubuntu20.04然后重新拉代码编译试试，可能是现有的环境有影响
> 
> 十之八九和系统版本没关系，因为op都是全套自己编译的，工具链都是用自己编译的，和外部的环境牵扯甚少，我安装好了20.04 全程没遇到任何问题，等我编译完告诉你答案

> > > 
> > 
> > 
> > 如果大佬有空的话，可以尝试虚拟机重新装一个ubuntu20.04然后重新拉代码编译试试，可能是现有的环境有影响
> 
> 十之八九和系统版本没关系，因为op都是全套自己编译的，工具链都是用自己编译的，和外部的环境牵扯甚少，我安装好了20.04 全程没遇到任何问题，等我编译完告诉你答案

我添加了这个东西
src-git helloworld https://github.com/fw876/helloworld
不知道有没有关系
另外，我现在正在用18.04重新编译，下午看结果吧

WYC-2020, 2022-05-07:
20.04编译完毕除了一个usb驱动在5.15上面有问题，其他没有任何问题，自己看我仓库提交记录
[image]

zixing131, 2022-05-07:
> 20.04编译完毕除了一个usb驱动在5.15上面有问题，其他没有任何问题，自己看我仓库提交记录 [image]

我也试完了，rt_rbus_pci_util.c这个文件有问题，我更新一下看看

WYC-2020, 2022-05-07:
我没遇到rt_rbus_pci_util.c 这个有问题

WYC-2020, 2022-05-07:
> > 20.04编译完毕除了一个usb驱动在5.15上面有问题，其他没有任何问题，自己看我仓库提交记录 [image]
> 
> 我也试完了，rt_rbus_pci_util.c这个文件有问题，我更新一下看看

建议你 那个编译环境 依赖，再用我仓库首页的那个 全部复制在安装一下

zixing131, 2022-05-07:
> > > 20.04编译完毕除了一个usb驱动在5.15上面有问题，其他没有任何问题，自己看我仓库提交记录 [image]
> > 
> > 
> > 我也试完了，rt_rbus_pci_util.c这个文件有问题，我更新一下看看
> 
> 建议你 那个编译环境 依赖，再用我仓库首页的那个 全部复制在安装一下

大佬，您的这个项目 https://github.com/WYC-2020/lede
和本项目（https://github.com/coolsnowwolf/lede）
代码应该是不完全一样的吧，这个报错是在这个项目里出现的，您那个项目我正在编译

liwenjie119, 2022-05-07:
> 大佬，您的这个项目 https://github.com/WYC-2020/lede 和本项目（[https://github.com/coolsnowwolf/lede）](https://github.com/coolsnowwolf/lede%EF%BC%89) 代码应该是不完全一样的吧，这个报错是在这个项目里出现的，您那个项目我正在编译

的确与环境无关，用github action在线编译18.04和20.04，lean的库均有同样问题，而https://github.com/WYC-2020/lede没有问题

WYC-2020, 2022-05-07:
> 大佬，您的这个项目 https://github.com/WYC-2020/lede 和本项目（[https://github.com/coolsnowwolf/lede）](https://github.com/coolsnowwolf/lede%EF%BC%89) 代码应该是不完全一样的吧，这个报错是在这个项目里出现的，您那个项目我正在编译

百分之九十九都是一样的，只是有我加入的个人定制以及一些优化而已，所以差距不大

zixing131, 2022-05-07:
> 百分之九十九都是一样的，只是有我加入的个人定制以及一些优化而已，所以差距不大

大佬，用您那个仓库编译通过了，只有这个仓库有问题，现在应该也算是解决了吧。

closed 2022-05-07 by zixing131

WYC-2020, 2022-05-07:
> 大佬，用您那个仓库编译通过了，只有这个仓库有问题，现在应该也算是解决了吧。

那不清楚了，我基本不会往l大仓库这里推送pr 免得被人吐槽，所以你如果要用l大的仓库 你还是以l大的为准，然后等他们修复吧，我不知道 l大这边还有这个问题没，这个只能等你们测试了

zixing131, 2022-05-07:
> 那不清楚了，我基本不会往l大仓库这里推送pr 免得被人吐槽，所以你如果要用l大的仓库 你还是以l大的为准，然后等他们修复吧，我不知道 l大这边还有这个问题没，这个只能等你们测试了

目前为止l大的仓库还是有这个问题的，先用您的仓库了，感谢。

WYC-2020, 2022-05-07:
丢人，刚我看了l大的好像是没支持5.15哦，你们肯定编译不过了，很多配置缺失，宏没定义，我这边是因为我自己加了5.15的所有自然没问题了哈哈，被你们绕进去了

liwenjie119, 2022-05-07:
> 丢人，刚我看了l大的好像是没支持5.15哦，你们肯定编译不过了，很多配置缺失，宏没定义，我这边是因为我自己加了5.15的所有自然没问题了哈哈，被你们绕进去了

https://github.com/coolsnowwolf/lede/commit/6135e15ba422fc79f64e3f6b7de8f504b87e4986
有这个，以为已经完全支持了，然后就编译不过

yuanlun1983, 2022-05-09:
> > 小米3g用5.15重来就没成功过
> 
> 同样亲测没任何问题，默认配置编译
大佬，用你的源码一样失败

liwenjie119, 2022-05-14:
补充两个文件就可以编译成功，但是暂时没条件测试，有条件的可以测试测试：
https://github.com/lede-project/source/blob/master/target/linux/ramips/mt7621/config-5.15
https://github.com/lede-project/source/blob/master/target/linux/ramips/files/drivers/mtd/nand/raw/mt7621_nand.c