coolsnowwolf/lede #9360: README.md，WSL按照编译说明文档运行命令下载源代码，更新 feeds 并选择配置，失败了。

baodp, 2022-05-06:
编译环境
WIN11  WSL2
ubuntu

WSL按照编译说明文档运行命令下载源代码，更新 feeds 并选择配置，失败了。
执行以下命令：
下载源代码，更新 feeds 并选择配置：

git clone https://github.com/coolsnowwolf/lede
cd lede
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig

运行结果如下：
bdp@DESKTOP-K24RLLE:~$ sudo apt update -y
-upgrade -y
sudo apt install -y ack antlr3 asciidoc autoconf automake autopoint binutils bison build-essential \
bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext gcc-multilib g++-multilib[sudo] password for bdp:
 \
git gperf haveged help2man intltool libc6-dev-i386 libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev \
libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libreadline-dev libssl-dev libtool lrzsz \
mkisofs msmtp nano ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 python3-pip qemu-utils \
rsync scons squashfs-tools subversion swig texinfo uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-devSorry, try again.[sudo] password for bdp:
Sorry, try again.
[sudo] password for bdp:
Hit:1 http://security.ubuntu.com/ubuntu focal-security InRelease
Hit:2 http://archive.ubuntu.com/ubuntu focal InRelease
Hit:3 http://archive.ubuntu.com/ubuntu focal-updates InRelease
Hit:4 http://archive.ubuntu.com/ubuntu focal-backports InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
54 packages can be upgraded. Run 'apt list --upgradable' to see them.
bdp@DESKTOP-K24RLLE:~$ git clone https://github.com/coolsnowwolf/lede
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfigfatal: destination path 'lede' already exists and is not an empty directory.
bdp@DESKTOP-K24RLLE:~$ cd lede
bdp@DESKTOP-K24RLLE:~/lede$ ./scripts/feeds update -a
Can't exec "make": No such file or directory at ./scripts/feeds line 22.
Use of uninitialized value in split at ./scripts/feeds line 22.
Use of uninitialized value $mkver[0] in pattern match (m//) at ./scripts/feeds line 24.
Use of uninitialized value $mkver[1] in pattern match (m//) at ./scripts/feeds line 25.
Use of uninitialized value $mkver[2] in split at ./scripts/feeds line 27.
Use of uninitialized value $mkv1 in numeric ge (>=) at ./scripts/feeds line 28.
Use of uninitialized value $mkv1 in numeric eq (==) at ./scripts/feeds line 28.
Unsupported version of make found: make
bdp@DESKTOP-K24RLLE:~/lede$ ./scripts/feeds install -a
Can't exec "make": No such file or directory at ./scripts/feeds line 22.
Use of uninitialized value in split at ./scripts/feeds line 22.
Use of uninitialized value $mkver[0] in pattern match (m//) at ./scripts/feeds line 24.
Use of uninitialized value $mkver[1] in pattern match (m//) at ./scripts/feeds line 25.
Use of uninitialized value $mkver[2] in split at ./scripts/feeds line 27.
Use of uninitialized value $mkv1 in numeric ge (>=) at ./scripts/feeds line 28.
Use of uninitialized value $mkv1 in numeric eq (==) at ./scripts/feeds line 28.
Unsupported version of make found: make
bdp@DESKTOP-K24RLLE:~/lede$ make menuconfig

Command 'make' not found, but can be installed with:

sudo apt install make        # version 4.2.1-1.2, or
sudo apt install make-guile  # version 4.2.1-1.2

bdp@DESKTOP-K24RLLE:~/lede$ make download -j8

Command 'make' not found, but can be installed with:

sudo apt install make        # version 4.2.1-1.2, or
sudo apt install make-guile  # version 4.2.1-1.2

bdp@DESKTOP-K24RLLE:~/lede$

EnnawYang, 2022-05-06:
linux基础知识先学习下

WYC-2020, 2022-05-06:
鉴定完毕真小白，哈哈，随便看个视频都能学会的

closed 2022-05-06 by AmadeusGhost