coolsnowwolf/lede #9364: 从新拉库 这错误怎么办？求助。谢谢

baodp, 2022-05-06:
bdp@DESKTOP-K24RLLE:~/lede$ make menuconfig
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... failed.
Checking 'working-gcc'... failed.
Checking 'g++'... failed.
Checking 'working-g++'... failed.
Checking 'ncurses'... failed.
Checking 'perl-data-dumper'... ok.
Checking 'perl-findbin'... ok.
Checking 'perl-file-copy'... ok.
Checking 'perl-file-compare'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'xargs'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'egrep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... failed.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'install'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'python3-distutils'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'rsync'... ok.
Checking 'which'... ok.
/home/bdp/lede/include/prereq-build.mk:192: recipe for target '/home/bdp/lede/staging_dir/host/bin/mkhash' failed
Prerequisite check failed. Use FORCE=1 to override.
/home/bdp/lede/include/toplevel.mk:179: recipe for target 'staging_dir/host/.prereq-build' failed
make: *** [staging_dir/host/.prereq-build] Error 1

baodp, 2022-05-06:
问题反馈：

按照文档直接全复制以下命令会不行。需要一行一行执行才不会出错。WSL Ubuntu环境下。

安装编译依赖

sudo apt update -y
sudo apt full-upgrade -y
sudo apt install -y ack antlr3 asciidoc autoconf automake autopoint binutils bison build-essential \
bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext gcc-multilib g++-multilib \
git gperf haveged help2man intltool libc6-dev-i386 libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev \
libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libreadline-dev libssl-dev libtool lrzsz \
mkisofs msmtp nano ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 python3-pip qemu-utils \
rsync scons squashfs-tools subversion swig texinfo uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-dev

lyc0209, 2022-05-07:
> 问题反馈：
>
> 按照文档直接全复制以下命令会不行。需要一行一行执行才不会出错。WSL Ubuntu环境下。
>
> 安装编译依赖
>
> sudo apt update -y sudo apt full-upgrade -y sudo apt install -y ack antlr3 asciidoc autoconf automake autopoint binutils bison [...]

这是三个sudo开头的命令，不可以全部复制。

WYC-2020, 2022-05-07:
明显的开发环境没有，gcc g++ 都检测失败哎

wuzhuang123, 2022-05-07:
ubuntu20下gcc和g++会有依赖问题，装18

WYC-2020, 2022-05-07:
> ubuntu20下gcc和g++会有依赖问题，装18

我刚还安装了20 编译了来，根本不是这个问题

EnnawYang, 2022-05-07:
> ubuntu20下gcc和g++会有依赖问题，装18

我都在用22了 有啥问题？

WYC-2020, 2022-05-07:
这明显就是小白编译环境没有装全嘛 没啥好讨论的 可以直接关闭

baodp, 2022-05-07:
别小白小白，你可以不看 我发出来是给有需要的人看的

closed 2022-05-07 by baodp