coolsnowwolf/lede #9365: 最近编译出现 python3 出错

zhengmz, 2022-05-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

在使用 github actions 编译时，总是出现 python3 出错，已经出现 10 多天，详细日志详见下面

而在本地，4月22日 clone 的代码编译都没问题

### (2) 路由器型号和固件版本

A: 型号配置如下

```conf
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_hiwifi_hc5962=y
```

### (3) 详细日志

```log
(cd /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/staging_dir/hostpkg/share/man/man1; ln -s python3.9.1 python3.1)
if test "xupgrade" != "xno"  ; then \
	case upgrade in \
		upgrade) ensurepip="--upgrade" ;; \
		install|*) ensurepip="" ;; \
	esac; \
	 ./python -E -m ensurepip \
		$ensurepip --root=/ ; \
fi
ERROR: Exception:
Traceback (most recent call last):
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/tmp/tmp59zq6l0p/pip-22.0.4-py3-none-any.whl/pip/_internal/cli/base_command.py", line 167, in exc_logging_wrapper
    status = run_func(*args)
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/tmp/tmp59zq6l0p/pip-22.0.4-py3-none-any.whl/pip/_internal/cli/req_command.py", line 205, in wrapper
    return func(self, options, args)
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/tmp/tmp59zq6l0p/pip-22.0.4-py3-none-any.whl/pip/_internal/commands/install.py", line 285, in run
    session = self.get_default_session(options)
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/tmp/tmp59zq6l0p/pip-22.0.4-py3-none-any.whl/pip/_internal/cli/req_command.py", line 75, in get_default_session
    self._session = self.enter_context(self._build_session(options))
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/tmp/tmp59zq6l0p/pip-22.0.4-py3-none-any.whl/pip/_internal/cli/req_command.py", line 89, in _build_session
    session = PipSession(
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/tmp/tmp59zq6l0p/pip-22.0.4-py3-none-any.whl/pip/_internal/network/session.py", line 282, in __init__
    self.headers["User-Agent"] = user_agent()
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/tmp/tmp59zq6l0p/pip-22.0.4-py3-none-any.whl/pip/_internal/network/session.py", line 97, in user_agent
    "python": platform.python_version(),
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/hostpkg/Python-3.9.12/Lib/platform.py", line 1127, in python_version
    return _sys_version()[1]
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/hostpkg/Python-3.9.12/Lib/platform.py", line 1076, in _sys_version
    raise ValueError(
ValueError: failed to parse CPython sys.version: '3.9.12 (main, May 04 2022, 2022-05-0) \n[GCC 9.4.0]'
Traceback (most recent call last):
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/hostpkg/Python-3.9.12/Lib/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/hostpkg/Python-3.9.12/Lib/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/hostpkg/Python-3.9.12/Lib/ensurepip/__main__.py", line 5, in <module>
    sys.exit(ensurepip._main())
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/hostpkg/Python-3.9.12/Lib/ensurepip/__init__.py", line 206, in _main
    return _bootstrap(
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/hostpkg/Python-3.9.12/Lib/ensurepip/__init__.py", line 125, in _bootstrap
    return _run_pip(args + [p[0] for p in _PROJECTS], additional_paths)
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/hostpkg/Python-3.9.12/Lib/ensurepip/__init__.py", line 34, in _run_pip
    return subprocess.run([sys.executable, "-c", code], check=True).returncode
  File "/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/hostpkg/Python-3.9.12/Lib/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/hostpkg/Python-3.9.12/python', '-c', '\nimport runpy\nimport sys\nsys.path = [\'/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/tmp/tmp59zq6l0p/setuptools-58.1.0-py3-none-any.whl\', \'/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/tmp/tmp59zq6l0p/pip-22.0.4-py3-none-any.whl\'] + sys.path\nsys.argv[1:] = [\'install\', \'--no-cache-dir\', \'--no-index\', \'--find-links\', \'/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/tmp/tmp59zq6l0p\', \'--root\', \'/\', \'--upgrade\', \'setuptools\', \'pip\']\nrunpy.run_module("pip", run_name="__main__", alter_sys=True)\n']' returned non-zero exit status 2.
make[4]: *** [Makefile:1276: install] Error 1
make[4]: Leaving directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/hostpkg/Python-3.9.12'
make[3]: Leaving directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/feeds/packages/lang/python/python3'
make[3]: *** [Makefile:378: /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/staging_dir/hostpkg/stamp/.python3_installed] Error 2
time: package/feeds/packages/python3/host-compile#6.75#1.39#8.28
    ERROR: package/feeds/packages/python3 [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/python3/host/compile] Error 1
make[2]: Leaving directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt'
make: *** [/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.
```

WYC-2020, 2022-05-07:
亲测没任何问题，只是我用的自己仓库测试，但是我的和l大的 基本都是一样的，所以可能还是需要你自己排查一下，最好万能大法后重新编译看下，我是在本地编译的

zhengmz, 2022-05-07:
我本地编译也是正常的

但使用 github action 编译出错了

WYC-2020, 2022-05-07:
那不清楚了，我有云编译不会去编译py 我云编译只编译x86 没任何问题，云编译懒得修麻烦

> 我本地编译也是正常的
> 
> 但使用 github action 编译出错了

zhengmz, 2022-05-07:
已找到问题，现在大家使用 github actions 脚本都是之前，修改后就可以了。

closed 2022-05-07 by zhengmz