coolsnowwolf/lede #9366: 作者沒提到的各種細節問題

PoorerImmune, 2022-05-07:
本人此次編譯兩天半依舊沒有成功，有些細節很重要，但是作者並沒有提：

1. 按照 README.md 的說明，該韌體是可以在 macOS 系統上編譯的，但是剛開始編譯，終端機上就會出現一段文字，大致意思是說 OpenWrt 只能在大小寫敏感的系統中編譯。但是我們很多人用的檔案系統是不區分大小寫的，因此編譯前必須借助「磁碟工具程式」新開闢一個 APFS（區分大小寫）的磁碟分割，然後將原始碼拉到那裡才能繼續進行。

2. 準備執行 `make menuconfig` 時，gcc 報錯，意思是說 Apple 提供的 make 不受支援（透過 Homebrew 安裝），必須使用 GNU make 才能編譯。然後我不得不到雲端服務商新開了一台 Linux 機器。

3. 作者推薦用 Debian 11 或 Ubuntu LTS 來編譯，其實我嘗試過並非如此。我一開始用的是 Ubuntu 20，編譯到中途被提示找不到 Python.h 標頭檔，我看網友解答要安裝 python-dev 這個 apt 套件才能解決，但是在 Ubuntu 中，這個套件並不存在，哪怕是安裝 python2-dev 和 python3-dev 均不管用。我折騰了半天，重灌系統為 Debian 11，才終於搞定了這個 Python 的問題。

4. 使用 `make menuconfig` 選取要編譯的項目時，不能選得太多，選得太多，中途失敗的機率就越高。我也不知道為什麼有一些 OpenWrt 元件就是編譯不成功，每次出錯我都得編輯 .config 檔案，把出問題的那一項用 # 加上註解以跳過，然後重新編譯，非常消磨時間！

5. 我明明在 .config 中要求程式給我編譯出 29.8 GB 的映像檔（因為我的路由器有 32 GB 的儲存空間，我本打算容量設大一些，免得擴容麻煩），但是不知道什麼毛病，好不容易到了最後生成 img 檔案的階段，終端機提示生成 rootfs.img 失敗，記憶體被耗盡，磁碟容量也被耗盡，然後我就給伺服器臨時分配了 30GB 的記憶體，1TB 的儲存空間（此前多次擴容，多次被提示空間不足）。結果我發現一個很怪異的現象：img 檔案能夠順利生成，但是程式 OpenWrt 並沒有按照我的要求生成指定大小的映像檔，我就盯著它的體積不停地往上漲。我分配多少磁碟空間，它就佔用多少空間，直到把磁碟容量撐滿！

整個編譯過程挫敗感還是很強的，關鍵是咱們還不知道錯在哪兒。我就搞不懂作者為什麼會覺得按照你的文檔說明一定能保證編譯成功。

WYC-2020, 2022-05-07:
macos 上面编译建议去看openwrt 官方步骤，那个上面说的很详细，l大这边只会给出基本步骤，不会说的那么细，2点我是没遇到过，我是虚拟机安装的mac系统测试的 我的步骤是这样，
1、xcode-select --install  or Command_Line_Tools_for_Xcode_11.5

2、/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"

3、brew install coreutils findutils gawk grep gnu-getopt gnu-tar wget diffutils git-extras quilt svn make ncurses pkg-config

4、
echo 'export PATH="/usr/local/opt/make/libexec/gnubin:$PATH"' >> ~/.bashrc
echo 'export PATH="/usr/local/opt/gnu-getopt/bin:$PATH"' >> ~/.bashrc
echo 'export PATH="/usr/local/opt/gettext/bin:$PATH"' >> ~/.bashrc
echo 'export PATH="/usr/local/opt/coreutils/libexec/gnubin:$PATH"' >> ~/.bashrc
echo 'export PATH="/usr/local/opt/findutils/libexec/gnubin:$PATH"' >> ~/.bashrc

5、
source ~/.bashrc
至于你说的第3点我是没遇到过，第4点是很多库有时候没更新上就会失败，或者下载失败之类的 这个需要梯子稳定，第5点我有说过，这个是有填充，自己去掉即可
另外说一下都能走到编译op了，其实很多东西不用说那么细大家都清楚，小伙子慢慢折腾吧

WYC-2020, 2022-05-07:
编译的过程就是不断的踩坑填坑，甚至有时候还会挖坑给自己，没有强大的心理不建议去编译，哈哈

hxlls, 2022-05-07:
默认的编译是能成功的，自己配置不同的东西包括容量，以及软件等，肯定需要自己要有一定的动手能力，不建议0基础小白来进行各种修改

GreatMichaelLee, 2022-05-13:
> macos 上面编译建议去看openwrt 官方步骤，那个上面说的很详细，l大这边只会给出基本步骤，不会说的那么细，2点我是没遇到过，我是虚拟机安装的mac系统测试的 我的步骤是这样， 1、xcode-select --install or Command_Line_Tools_for_Xcode_11.5
>
> 2、/usr/bin/ruby -e "$(curl -fsSL [...]

居然看到大佬回这么多.....哈哈

closed 2022-09-24 by PoorerImmune