coolsnowwolf/lede #9367: 编译ax6s 报错  ipset编译失败

LGinC, 2022-05-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在wsl2的debian11里执行`PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin make V=s -j1` 编译到ipset时出错

### (2) 路由器型号和固件版本

A：红米AX6S R22.5

### (3) 详细日志

A：
```bash
configure.ac:7: aclocal.m4 with aclocal and run automake again.
WARNING: 'automake-1.16' is probably too old.
         You should only need it if you modified 'Makefile.am' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'automake' program is part of the GNU Automake package:
         <https://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <https://www.gnu.org/software/autoconf>
         <https://www.gnu.org/software/m4/>
         <https://www.perl.org/>
make[4]: *** [Makefile:522: Makefile.in] Error 1
make[4]: Leaving directory '/home/we/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/ipset-7.15'
make[3]: *** [Makefile:76: /home/we/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/ipset-7.15/.built] Error 2
make[3]: Leaving directory '/home/we/lede/package/network/utils/ipset'
time: package/network/utils/ipset/compile#0.28#0.04#0.31
    ERROR: package/network/utils/ipset failed to build.
make[2]: *** [package/Makefile:116: package/network/utils/ipset/compile] Error 1
make[2]: Leaving directory '/home/we/lede'
make[1]: *** [package/Makefile:110: /home/we/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/we/lede'
make: *** [/home/we/lede/include/toplevel.mk:230: world] Error 2
```

WYC-2020, 2022-05-07:
升级一下automake 看下

LGinC, 2022-05-07:
```bash
$ apt search automake
Sorting... Done
Full Text Search... Done
acr/stable 1.9.4-1 all
  autoconf like tool

automake/stable,now 1:1.16.3-2 all [installed]
  Tool for generating GNU Standards-compliant Makefiles
```

WYC-2020, 2022-05-07:
那这个不清楚了没环境，因为我都是ubuntu编译，等会看下有人遇到过没把

WYC-2020, 2022-05-07:
有时间我操刀一下wsl看下

LGinC, 2022-05-07:
我的环境是win11 wsl2 debian11

eagle-lk, 2022-05-16:
我的环境是win11 wsl2 ubuntu server 2004lts，可以正常编译。

closed 2022-05-18 by AmadeusGhost