coolsnowwolf/lede #9368: 目前ramips mt7621编译测试内核5.15失败,稳定内核可以改默认5.10吗

ayancn, 2022-05-07:
## 1.关于你要提交的问题

目前ramips mt7621编译测试内核5.15都是失败已经一两个月了,稳定内核可以改成默认5.10吗
现在默认内核是5.4测试内核是5.15(编译不能成功) 之前一直用5.10现在无法直接改成5.10编译
又不想用现在的稳定版本内核5.4 希望可以把以下
https://github.com/coolsnowwolf/lede/blob/master/target/linux/ramips/Makefile
这个文件的 KERNEL_PATCHVER:=5.4 这段改成 KERNEL_PATCHVER:=5.10
让我们可以默认稳定内核就编译5.10内核的 谢谢

### (2) 路由器型号 

小米CR660X - ramips mt7621

WYC-2020, 2022-05-07:
看吧 一堆人怼你，这种自己改就是了，万一有人要用5.4的呢

EnnawYang, 2022-05-07:
放过7621吧，球球了

WYC-2020, 2022-05-07:
> 放过7621吧，球球了

哈哈

ayancn, 2022-05-07:
> 看吧 一堆人怼你，这种自己改就是了，万一有人要用5.4的呢

不是每个人都像你一样吃饱太闲可以每次编译都是新建新的存档更改一遍在编译好吗?

习惯了用自动云编译

mitsukileung, 2022-05-07:
> > 看吧 一堆人怼你，这种自己改就是了，万一有人要用5.4的呢
> 
> 不是每个人都像你一样吃饱太闲可以每次编译都是新建新的存档更改一遍在编译好吗?
> 
> 习惯了用自动云编译

用云编译不是更简单？就想别人迁就你？自己FORK个不行？

lphgor, 2022-05-07:
你都使用自动云编译了，在云编译脚本里面用sed替换一下不就好了嘛，一行代码就能搞定😓

WYC-2020, 2022-05-08:
> > 看吧 一堆人怼你，这种自己改就是了，万一有人要用5.4的呢
> 
> 不是每个人都像你一样吃饱太闲可以每次编译都是新建新的存档更改一遍在编译好吗?
> 
> 习惯了用自动云编译

笑死我，你怕是没用到云编译精髓

ayancn, 2022-05-08:
> > > 看吧 一堆人怼你，这种自己改就是了，万一有人要用5.4的呢
> > 
> > 
> > 不是每个人都像你一样吃饱太闲可以每次编译都是新建新的存档更改一遍在编译好吗?
> > 习惯了用自动云编译
> 
> 笑死我，你怕是没用到云编译精髓

嗯你这人最厉害了? 这样说你你才能有满满的优越感自信感吗?

ayancn, 2022-05-08:
> > > 看吧 一堆人怼你，这种自己改就是了，万一有人要用5.4的呢
> > 
> > 
> > 不是每个人都像你一样吃饱太闲可以每次编译都是新建新的存档更改一遍在编译好吗?
> > 习惯了用自动云编译
> 
> 用云编译不是更简单？就想别人迁就你？会云编译就不会自己FORK个？

FORK的可以自动同步上游更新吗?

EnnawYang, 2022-05-08:
> 
> FORK的可以自动同步上游更新吗?

？可以啊，难道不可以？

mitsukileung, 2022-05-08:
> > 看吧 一堆人怼你，这种自己改就是了，万一有人要用5.4的呢
> 
> 不是每个人都像你一样吃饱太闲可以每次编译都是新建新的存档更改一遍在编译好吗?
> 
> 习惯了用自动云编译

FORK和直接用LEDE都可以，FORK是最容易理解的方法！

uniessey, 2022-05-08:
是这样的，我已经编译出了5.15，我自己改了rt_liunx.c，也补齐了配置文件。编译过程很顺利，没出错。准备今天测试一下。但是我的硬件版本是NEWIFI-D2，万一硬件版本不支持，可能还是无法启动。

WYC-2020, 2022-05-08:
> > > > 看吧 一堆人怼你，这种自己改就是了，万一有人要用5.4的呢
> > > 
> > > 
> > > 不是每个人都像你一样吃饱太闲可以每次编译都是新建新的存档更改一遍在编译好吗?
> > > 习惯了用自动云编译
> > 
> > 
> > 笑死我，你怕是没用到云编译精髓
> 
> 嗯你这人最厉害了? 这样说你你才能有满满的优越感自信感吗?

说你不会玩嘛还不信，确实老子是闲的，尼玛吃饱没事天天帮你们这帮锤子解决问题，还被你们说，算了拜拜，你们慢慢玩吧，老子不配你你们折腾，爱咋玩咋玩，还是那些闷到自己定制了不开放出来的舒服

ayancn, 2022-05-09:
> > FORK的可以自动同步上游更新吗?
> 
> ？可以啊，难道不可以？

我fork编译成功5.10了,是FORK的要手动自动同步上游更新,请问如何自动更新合并上游更新,因为要用自动云编译

liwenjie119, 2022-05-10:
> > > FORK的可以自动同步上游更新吗?
> > 
> > 
> > ？可以啊，难道不可以？
> 
> 我fork编译成功5.10了,是FORK的要手动自动同步上游更新,请问如何自动更新合并上游更新,因为要用自动云编译

你这是还没搞明白云编译的目的，而且对linux不熟悉啊，给你diy-part1.sh和diy-part2.sh就是为了定制的，在diy-part1.sh最下面加一句sed不就妥了，上面也有人给你说了你也没注意吧
sed -i "s/KERNEL_PATCHVER:=5.4/KERNEL_PATCHVER:=5.10/g" target/linux/ramips/Makefile

ayancn, 2022-05-10:
> > > > FORK的可以自动同步上游更新吗?
> > > 
> > > 
> > > ？可以啊，难道不可以？
> > 
> > 
> > 我fork编译成功5.10了,是FORK的要手动自动同步上游更新,请问如何自动更新合并上游更新,因为要用自动云编译
> 
> 你这是还没搞明白云编译的目的，而且对linux不熟悉啊，给你diy-part1.sh和diy-part2.sh就是为了定制的，在diy-part1.sh最下面加一句sed不就妥了，上面也有人给你说了你也没注意吧 sed -i "s/KERNEL_PATCHVER:=5.4/KERNEL_PATCHVER:=5.10/g" target/linux/ramips/Makefile

我需要的就是这个,你完美解决了我的问题,点赞,这样是最简单的!

比楼上那个词不达意的好太多了,谢谢!

zhb7670, 2022-05-10:
虽然问的小白问题，但确实很多人不会改！知道的大神虚心一点嘛！很多人只是初学者！没必要跟小白过不去！但是小白也要虚心向他人学习

ayancn, 2022-05-11:
#
# This is free software, lisence use MIT.
# 
# Copyright (C) 2019 P3TERX <https://p3terx.com>
# Copyright (C) 2020 KFERMercer <KFER.Mercer@gmail.com>
# 
# <https://github.com/KFERMercer/OpenWrt-CI>
#

name: OpenWrt

on:
  push:
    branches: 
      - master
  schedule:
    - cron: 0 2 * * *
  watch:
    types: started

jobs:
  build:
    runs-on: ubuntu-latest
    if: github.event.repository.owner.id == github.event.sender.id

    steps:
      - name: Checkout
        uses: actions/checkout@master

      - name: Initialization environment
        env:
          DEBIAN_FRONTEND: noninteractive
        run: |
          sudo -E apt-get -yqq update
          sudo -E apt-get -yqq install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib
          sudo -E apt-get -y autoremove --purge
          sudo -E apt-get clean
      - name: Download lede
        run: |
          git clone https://github.com/coolsnowwolf/lede
          cp .config ./lede/.config
          mv ./lede/* ./
      - name: Install Helloword
        run: |
          echo "src-git helloworld https://github.com/fw876/helloworld" >> ./feeds.conf.default 
      - name: Install OpenClash
        run: |
          mkdir package/luci-app-openclash
          cd package/luci-app-openclash
          git init
          git remote add -f origin https://github.com/vernesong/OpenClash.git
          git config core.sparsecheckout true
          echo "luci-app-openclash" >> .git/info/sparse-checkout
          git pull --depth 1 origin master
          git branch --set-upstream-to=origin/master master
          pushd luci-app-openclash/tools/po2lmo
          make && sudo make install
          popd
      - name: Update feeds
        run: |
          ./scripts/feeds update -a
          ./scripts/feeds install -a
      - name: Costom configure file
        run: |
          make defconfig
      - name: Download package source code
        run: |
          make download -j8
          find dl -size -1024c -exec ls -l {} \;
          find dl -size -1024c -exec rm -f {} \;
      - name: Compile firmware
        run: |
          echo -e "$(nproc) thread build."
          make -j$(nproc) V=s
      - name : Upload artifact
        uses: actions/upload-artifact@master
        with:
          name: OpenWrt_firmware
          path: bin/targets/
          

这个如何增加 diy-part1.sh定制

在那行如何加入, 确实是小白才来问的

Wilson-coder, 2022-05-14:
> 虽然问的小白问题，但确实很多人不会改！知道的大神虚心一点嘛！很多人只是初学者！没必要跟小白过不去！但是小白也要虚心向他人学习

分明是小白，却非要老干部的待遇，那就看技术高手有没有做仆人的心情了~~~哈哈

ayancn, 2022-05-14:
> > 虽然问的小白问题，但确实很多人不会改！知道的大神虚心一点嘛！很多人只是初学者！没必要跟小白过不去！但是小白也要虚心向他人学习
> 
> 分明是小白，却非要老干部的待遇，那就看技术高手有没有做仆人的心情了~~~哈哈

不用你解释了,自己搞定再见你这所谓的技术高手

ayancn, 2022-05-14:
> > 虽然问的小白问题，但确实很多人不会改！知道的大神虚心一点嘛！很多人只是初学者！没必要跟小白过不去！但是小白也要虚心向他人学习
> 
> 分明是小白，却非要老干部的待遇，那就看技术高手有没有做仆人的心情了~~~哈哈

看你编译的.config文件垃圾那么多无用的,你还不如小白,编译的固件无用垃圾多所以大! 浪费空间

closed 2022-05-21 by AmadeusGhost