coolsnowwolf/lede #9371: I225-v B3 2.5G口老是断流重启是怎么回事？

GreatMichaelLee, 2022-05-08:
rt, 新到的N5105 4网口 软路由，配都I225-v B3 网卡，发现网卡经常随机断流重启，怎么回事啊？设备用风扇吹着，CPU才50多度，照例说不热，之前用的5.10的内核，看网上说升到5.15内核就好了，现在升到5.15版本，发现还是重启啊，一重启就要重拨号断网了，是驱动的问题吗，麻烦大佬看看。附送内核日志，那些寄存器dump的值看不懂，没手册。

```
[23089.373953] igc 0000:05:00.0 eth3: NIC Link is Down
[23089.379432] br-lan: port 2(eth3) entered disabled state
[23093.574044] igc 0000:05:00.0 eth3: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
[23093.583431] br-lan: port 2(eth3) entered blocking state
[23093.589179] br-lan: port 2(eth3) entered listening state
[23095.625498] br-lan: port 2(eth3) entered learning state
[23097.637401] br-lan: port 2(eth3) entered forwarding state
[23097.643349] br-lan: topology change detected, propagating
[28374.172849] igc 0000:02:00.0 eth0: NIC Link is Down
[28374.178413] igc 0000:02:00.0 eth0: Register Dump
[28374.183514] igc 0000:02:00.0 eth0: Register Name   Value
[28374.189367] igc 0000:02:00.0 eth0: CTRL            081c0641
[28374.195578] igc 0000:02:00.0 eth0: STATUS          40380681
[28374.201721] igc 0000:02:00.0 eth0: CTRL_EXT        10000040
[28374.207864] igc 0000:02:00.0 eth0: MDIC            18017949
[28374.213988] igc 0000:02:00.0 eth0: ICR             00000001
[28374.220124] igc 0000:02:00.0 eth0: RCTL            0440803a
[28374.226253] igc 0000:02:00.0 eth0: RDLEN[0-3]      00001000 00001000 00001000 00001000
[28374.234952] igc 0000:02:00.0 eth0: RDH[0-3]        00000049 0000005a 00000000 000000c2
[28374.243650] igc 0000:02:00.0 eth0: RDT[0-3]        00000048 00000059 000000ff 000000c1
[28374.252340] igc 0000:02:00.0 eth0: RXDCTL[0-3]     02040808 02040808 02040808 02040808
[28374.261032] igc 0000:02:00.0 eth0: RDBAL[0-3]      0d0e9000 0d0ec000 0d0ef000 0d132000
[28374.269724] igc 0000:02:00.0 eth0: RDBAH[0-3]      00000001 00000001 00000001 00000001
[28374.278418] igc 0000:02:00.0 eth0: TCTL            a503f0fa
[28374.284544] igc 0000:02:00.0 eth0: TDBAL[0-3]      0d0d7000 0d0dc000 0d0e1000 0d0e6000
[28374.293236] igc 0000:02:00.0 eth0: TDBAH[0-3]      00000001 00000001 00000001 00000001
[28374.301929] igc 0000:02:00.0 eth0: TDLEN[0-3]      00001000 00001000 00001000 00001000
[28374.310621] igc 0000:02:00.0 eth0: TDH[0-3]        0000003d 00000032 000000ca 000000b6
[28374.319314] igc 0000:02:00.0 eth0: TDT[0-3]        0000003d 00000032 000000ca 000000b6
[28374.328013] igc 0000:02:00.0 eth0: TXDCTL[0-3]     02100108 02100108 02100108 02100108
[28374.336707] igc 0000:02:00.0 eth0: Reset adapter
[28381.174368] traps: edge[23094] trap stack segment ip:41a686 sp:7ffd31b54640 error:0 in edge[403000+80000]
[28393.212179] igc 0000:02:00.0 eth0: NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
[28403.443916] pppoe-vwan2: renamed from ppp1
[28403.475957] pppoe-vwan1: renamed from ppp2
[28422.615871] pppoe-vwan1: renamed from ppp2
[28441.803773] pppoe-vwan1: renamed from ppp2
[28461.063752] pppoe-vwan1: renamed from ppp2
```

GreatMichaelLee, 2022-05-08:
下面是一次完整的断流系统日志，我也不知道到底是I225 v3的驱动有问题还是硬件问题？当时正在做iperf3 测速，一下就网口重启了。现在集成在openwrt里的I225网卡驱动是最新的或者是相对稳定的吗？

Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.373064] igc 0000:04:00.0 eth2: NIC Link is Down
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.378587] br-lan: port 1(eth2) entered disabled state
Sun May  8 20:19:52 2022 daemon.notice netifd: Network device 'eth2' link is down
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.384448] igc 0000:04:00.0 eth2: Register Dump
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.389533] igc 0000:04:00.0 eth2: Register Name   Value
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.395379] igc 0000:04:00.0 eth2: CTRL            181c0641
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.401508] igc 0000:04:00.0 eth2: STATUS          40780681
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.407636] igc 0000:04:00.0 eth2: CTRL_EXT        10000040
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.413762] igc 0000:04:00.0 eth2: MDIC            18017949
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.419888] igc 0000:04:00.0 eth2: ICR             00000001
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.426016] igc 0000:04:00.0 eth2: RCTL            0440803a
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.432147] igc 0000:04:00.0 eth2: RDLEN[0-3]      00001000 00001000 00001000 00001000
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.440844] igc 0000:04:00.0 eth2: RDH[0-3]        00000001 000000e3 000000b7 000000e4
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.449546] igc 0000:04:00.0 eth2: RDT[0-3]        00000000 000000e2 000000b6 000000e3
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.458243] igc 0000:04:00.0 eth2: RXDCTL[0-3]     02040808 02040808 02040808 02040808
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.466945] igc 0000:04:00.0 eth2: RDBAL[0-3]      0b753000 0b836000 0ba5d000 0a852000
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.475646] igc 0000:04:00.0 eth2: RDBAH[0-3]      00000001 00000001 00000001 00000001
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.484345] igc 0000:04:00.0 eth2: TCTL            a503f0fa
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.490473] igc 0000:04:00.0 eth2: TDBAL[0-3]      0ad7f000 0b668000 0a34f000 0b772000
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.499172] igc 0000:04:00.0 eth2: TDBAH[0-3]      00000001 00000001 00000001 00000001
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.507874] igc 0000:04:00.0 eth2: TDLEN[0-3]      00001000 00001000 00001000 00001000
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.516580] igc 0000:04:00.0 eth2: TDH[0-3]        00000090 0000000c 00000089 0000002d
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.525282] igc 0000:04:00.0 eth2: TDT[0-3]        00000090 0000000c 00000089 0000002d
Sun May  8 20:19:52 2022 kern.info kernel: [ 7784.533985] igc 0000:04:00.0 eth2: TXDCTL[0-3]     02100108 02100108 02100108 02100108
Sun May  8 20:19:52 2022 kern.err kernel: [ 7784.542682] igc 0000:04:00.0 eth2: Reset adapter
Sun May  8 20:19:53 2022 daemon.notice netifd: bridge 'br-lan' link is down
Sun May  8 20:19:53 2022 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun May  8 20:19:56 2022 kern.info kernel: [ 7788.517173] igc 0000:04:00.0 eth2: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Sun May  8 20:19:56 2022 kern.info kernel: [ 7788.526575] br-lan: port 1(eth2) entered blocking state
Sun May  8 20:19:56 2022 kern.info kernel: [ 7788.532326] br-lan: port 1(eth2) entered listening state
Sun May  8 20:19:56 2022 daemon.notice netifd: Network device 'eth2' link is up
Sun May  8 20:19:58 2022 kern.info kernel: [ 7790.568637] br-lan: port 1(eth2) entered learning state
Sun May  8 20:20:00 2022 kern.info kernel: [ 7792.580557] br-lan: port 1(eth2) entered forwarding state
Sun May  8 20:20:00 2022 kern.info kernel: [ 7792.586514] br-lan: topology change detected, propagating
Sun May  8 20:20:00 2022 daemon.notice netifd: bridge 'br-lan' link is up
Sun May  8 20:20:00 2022 daemon.notice netifd: Interface 'lan' has link connectivity
Sun May  8 20:20:34 2022 kern.info kernel: [ 7826.799201] igc 0000:04:00.0 eth2: NIC Link is Down
Sun May  8 20:20:34 2022 kern.info kernel: [ 7826.804666] br-lan: port 1(eth2) entered disabled state
Sun May  8 20:20:34 2022 daemon.notice netifd: Network device 'eth2' link is down
Sun May  8 20:20:35 2022 daemon.notice netifd: bridge 'br-lan' link is down
Sun May  8 20:20:35 2022 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun May  8 20:20:38 2022 kern.info kernel: [ 7830.927316] igc 0000:04:00.0 eth2: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Sun May  8 20:20:38 2022 kern.info kernel: [ 7830.936737] br-lan: port 1(eth2) entered blocking state
Sun May  8 20:20:38 2022 kern.info kernel: [ 7830.942480] br-lan: port 1(eth2) entered listening state
Sun May  8 20:20:38 2022 daemon.notice netifd: Network device 'eth2' link is up
Sun May  8 20:20:40 2022 kern.info kernel: [ 7832.966783] br-lan: port 1(eth2) entered learning state
Sun May  8 20:20:42 2022 kern.info kernel: [ 7834.982695] br-lan: port 1(eth2) entered forwarding state
Sun May  8 20:20:42 2022 kern.info kernel: [ 7834.988652] br-lan: topology change detected, propagating
Sun May  8 20:20:42 2022 daemon.notice netifd: bridge 'br-lan' link is up
Sun May  8 20:20:42 2022 daemon.notice netifd: Interface 'lan' has link connectivity

TioaChan, 2022-05-08:
同样的问题
```
Sun May  8 22:19:35 2022 kern.info kernel: [19485.217376] igc 0000:02:00.0 eth1: NIC Link is Down
Sun May  8 22:19:35 2022 kern.info kernel: [19485.222627] br-lan: port 1(eth1) entered disabled state
Sun May  8 22:19:35 2022 daemon.notice netifd: Network device 'eth1' link is down
Sun May  8 22:19:44 2022 kern.info kernel: [19494.113496] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Sun May  8 22:19:44 2022 kern.info kernel: [19494.229262] br-lan: port 1(eth1) entered blocking state
Sun May  8 22:19:44 2022 kern.info kernel: [19494.234513] br-lan: port 1(eth1) entered forwarding state
Sun May  8 22:19:44 2022 daemon.notice netifd: Network device 'eth1' link is up
Sun May  8 22:19:51 2022 kern.info kernel: [19501.377159] igc 0000:02:00.0 eth1: NIC Link is Down
Sun May  8 22:19:51 2022 kern.info kernel: [19501.382668] br-lan: port 1(eth1) entered disabled state
Sun May  8 22:19:51 2022 daemon.notice netifd: Network device 'eth1' link is down
Sun May  8 22:19:51 2022 kern.info kernel: [19501.388066] igc 0000:02:00.0 eth1: Register Dump
Sun May  8 22:19:51 2022 kern.info kernel: [19501.392677] igc 0000:02:00.0 eth1: Register Name   Value
Sun May  8 22:19:51 2022 kern.info kernel: [19501.397976] igc 0000:02:00.0 eth1: CTRL            081c0641
Sun May  8 22:19:51 2022 kern.info kernel: [19501.403552] igc 0000:02:00.0 eth1: STATUS          40380681
Sun May  8 22:19:51 2022 kern.info kernel: [19501.409130] igc 0000:02:00.0 eth1: CTRL_EXT        10000040
Sun May  8 22:19:51 2022 kern.info kernel: [19501.414687] igc 0000:02:00.0 eth1: MDIC            18017949
Sun May  8 22:19:51 2022 kern.info kernel: [19501.420244] igc 0000:02:00.0 eth1: ICR             00000001
Sun May  8 22:19:51 2022 kern.info kernel: [19501.425819] igc 0000:02:00.0 eth1: RCTL            0440803a
Sun May  8 22:19:51 2022 kern.info kernel: [19501.431378] igc 0000:02:00.0 eth1: RDLEN[0-3]      00001000 00001000 00001000 00001000
Sun May  8 22:19:51 2022 kern.info kernel: [19501.439288] igc 0000:02:00.0 eth1: RDH[0-3]        000000bc 00000033 00000056 000000be
Sun May  8 22:19:51 2022 kern.info kernel: [19501.447200] igc 0000:02:00.0 eth1: RDT[0-3]        000000bb 00000032 00000055 000000bd
Sun May  8 22:19:51 2022 kern.info kernel: [19501.455110] igc 0000:02:00.0 eth1: RXDCTL[0-3]     02040808 02040808 02040808 02040808
Sun May  8 22:19:51 2022 kern.info kernel: [19501.463022] igc 0000:02:00.0 eth1: RDBAL[0-3]      003b7000 02a19000 0385d000 03d8d000
Sun May  8 22:19:51 2022 kern.info kernel: [19501.470935] igc 0000:02:00.0 eth1: RDBAH[0-3]      00000001 00000001 00000001 00000001
Sun May  8 22:19:51 2022 kern.info kernel: [19501.478841] igc 0000:02:00.0 eth1: TCTL            a503f0fa
Sun May  8 22:19:51 2022 kern.info kernel: [19501.484421] igc 0000:02:00.0 eth1: TDBAL[0-3]      03ffc000 04627000 04587000 04643000
Sun May  8 22:19:51 2022 kern.info kernel: [19501.492336] igc 0000:02:00.0 eth1: TDBAH[0-3]      00000001 00000001 00000001 00000001
Sun May  8 22:19:51 2022 kern.info kernel: [19501.500244] igc 0000:02:00.0 eth1: TDLEN[0-3]      00001000 00001000 00001000 00001000
Sun May  8 22:19:51 2022 kern.info kernel: [19501.508150] igc 0000:02:00.0 eth1: TDH[0-3]        00000066 0000007e 00000007 000000d9
Sun May  8 22:19:51 2022 kern.info kernel: [19501.516058] igc 0000:02:00.0 eth1: TDT[0-3]        00000066 0000007e 00000011 000000d9
Sun May  8 22:19:51 2022 kern.info kernel: [19501.523970] igc 0000:02:00.0 eth1: TXDCTL[0-3]     02100108 02100108 02100108 02100108
Sun May  8 22:19:51 2022 kern.err kernel: [19501.531873] igc 0000:02:00.0 eth1: Reset adapter
Sun May  8 22:19:58 2022 kern.info kernel: [19507.749313] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Sun May  8 22:19:58 2022 kern.info kernel: [19507.865074] br-lan: port 1(eth1) entered blocking state
Sun May  8 22:19:58 2022 kern.info kernel: [19507.870306] br-lan: port 1(eth1) entered forwarding state
Sun May  8 22:19:58 2022 daemon.notice netifd: Network device 'eth1' link is up
Sun May  8 23:43:10 2022 kern.info kernel: [24499.848565] igc 0000:02:00.0 eth1: NIC Link is Down
Sun May  8 23:43:10 2022 kern.info kernel: [24499.853856] br-lan: port 1(eth1) entered disabled state
Sun May  8 23:43:10 2022 daemon.notice netifd: Network device 'eth1' link is down
```

GreatMichaelLee, 2022-05-09:
> 同样的问题
> 
> ```
> Sun May  8 22:19:35 2022 kern.info kernel: [19485.217376] igc 0000:02:00.0 eth1: NIC Link is Down
> Sun May  8 22:19:35 2022 kern.info kernel: [19485.222627] br-lan: port 1(eth1) entered disabled state
> Sun May  8 22:19:35 2022 daemon.notice netifd: Network device 'eth1' link is down
> Sun May  8 22:19:44 2022 kern.info kernel: [19494.113496] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
> Sun May  8 22:19:44 2022 kern.info kernel: [19494.229262] br-lan: port 1(eth1) entered blocking state
> Sun May  8 22:19:44 2022 kern.info kernel: [19494.234513] br-lan: port 1(eth1) entered forwarding state
> Sun May  8 22:19:44 2022 daemon.notice netifd: Network device 'eth1' link is up
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.377159] igc 0000:02:00.0 eth1: NIC Link is Down
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.382668] br-lan: port 1(eth1) entered disabled state
> Sun May  8 22:19:51 2022 daemon.notice netifd: Network device 'eth1' link is down
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.388066] igc 0000:02:00.0 eth1: Register Dump
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.392677] igc 0000:02:00.0 eth1: Register Name   Value
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.397976] igc 0000:02:00.0 eth1: CTRL            081c0641
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.403552] igc 0000:02:00.0 eth1: STATUS          40380681
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.409130] igc 0000:02:00.0 eth1: CTRL_EXT        10000040
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.414687] igc 0000:02:00.0 eth1: MDIC            18017949
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.420244] igc 0000:02:00.0 eth1: ICR             00000001
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.425819] igc 0000:02:00.0 eth1: RCTL            0440803a
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.431378] igc 0000:02:00.0 eth1: RDLEN[0-3]      00001000 00001000 00001000 00001000
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.439288] igc 0000:02:00.0 eth1: RDH[0-3]        000000bc 00000033 00000056 000000be
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.447200] igc 0000:02:00.0 eth1: RDT[0-3]        000000bb 00000032 00000055 000000bd
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.455110] igc 0000:02:00.0 eth1: RXDCTL[0-3]     02040808 02040808 02040808 02040808
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.463022] igc 0000:02:00.0 eth1: RDBAL[0-3]      003b7000 02a19000 0385d000 03d8d000
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.470935] igc 0000:02:00.0 eth1: RDBAH[0-3]      00000001 00000001 00000001 00000001
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.478841] igc 0000:02:00.0 eth1: TCTL            a503f0fa
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.484421] igc 0000:02:00.0 eth1: TDBAL[0-3]      03ffc000 04627000 04587000 04643000
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.492336] igc 0000:02:00.0 eth1: TDBAH[0-3]      00000001 00000001 00000001 00000001
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.500244] igc 0000:02:00.0 eth1: TDLEN[0-3]      00001000 00001000 00001000 00001000
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.508150] igc 0000:02:00.0 eth1: TDH[0-3]        00000066 0000007e 00000007 000000d9
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.516058] igc 0000:02:00.0 eth1: TDT[0-3]        00000066 0000007e 00000011 000000d9
> Sun May  8 22:19:51 2022 kern.info kernel: [19501.523970] igc 0000:02:00.0 eth1: TXDCTL[0-3]     02100108 02100108 02100108 02100108
> Sun May  8 22:19:51 2022 kern.err kernel: [19501.531873] igc 0000:02:00.0 eth1: Reset adapter
> Sun May  8 22:19:58 2022 kern.info kernel: [19507.749313] igc 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
> Sun May  8 22:19:58 2022 kern.info kernel: [19507.865074] br-lan: port 1(eth1) entered blocking state
> Sun May  8 22:19:58 2022 kern.info kernel: [19507.870306] br-lan: port 1(eth1) entered forwarding state
> Sun May  8 22:19:58 2022 daemon.notice netifd: Network device 'eth1' link is up
> Sun May  8 23:43:10 2022 kern.info kernel: [24499.848565] igc 0000:02:00.0 eth1: NIC Link is Down
> Sun May  8 23:43:10 2022 kern.info kernel: [24499.853856] br-lan: port 1(eth1) entered disabled state
> Sun May  8 23:43:10 2022 daemon.notice netifd: Network device 'eth1' link is down
> ```
> 
> | 内核版本 | 5.15.36 | | 本地时间 | Sun May 8 23:52:33 2022 | | 运行时间 | 6h 57m 43s | | 平均负载 | 0.04, 0.02, 0.00 | | CPU 使用率（%） | 3 % |

请大佬们看下目前原生的I225-v B3的驱动程序是不是intel最新的哈？好像是kmod-igc这个，也不知道怎么看版本，这个网卡的操作温度才70°，也是醉了。感觉这个网卡真是拉跨，从B1到B3还是这个尿性，看来英特尔的研发解决问题的能力也很堪忧。

sbwcwfhv001, 2022-05-11:
虚拟机没有这个问题pve我还是用的老内核5.4的

shu0802, 2022-05-11:
说明对虚拟设备支持得好，对物理设备反而支持得不好

iKira, 2022-05-11:
> 说明对虚拟设备支持得好，对物理设备反而支持得不好

就是驱动没写好

shu0802, 2022-05-11:
[image]
继续出现这问题

GreatMichaelLee, 2022-05-12:
> [image] 5.15.38继续出现这问题

我换7类线+加强力大风扇，目前来看貌似缓解了。这玩意估计多半就是体制问题，网卡离CPU太近，而且本身也容易发热，它操作温度只有70度，温度一高，再加上如果接的网线如果屏蔽不好干扰大的话，估计驱动就容易gg。可能网卡本身也有体质差异，可以参考一下。

596142041, 2022-05-13:
用的PVE虚拟机，同时还直通了网卡，没发现问题，N6005，运行了快半个月了

shu0802, 2022-05-14:
有没有找到官网上的网卡驱动，想手动安装试试。

GreatMichaelLee, 2022-05-14:
> 有没有找到官网上的网卡驱动，想手动安装试试。

能直接在openwrt x86下安装？

coolsnowwolf, 2022-05-16:
[image]
很明显是 B1/B2 冒充 B3 的缺陷版 I225

GreatMichaelLee, 2022-05-16:
> [image] 很明显是 B1/B2 冒充 B3 的缺陷版 I225

啊？大佬是如何看出来的？寄存器信息里有解读？如果是这样，我要找卖家理论啊

GreatMichaelLee, 2022-05-16:
[image]

被大雕老大吓坏了，赶紧lspci -vvv看了一下，如果造假能把B1/B2的片子系统里识别成B3的信息，那我真是识别不了了:(

shu0802, 2022-05-17:
> [image] 很明显是 B1/B2 冒充 B3 的缺陷版 I225

问一下大佬，你用的啥命令查看的，查看网卡就是显示如下图的结果，看不到控制器的参数。你的图中结果表示是B3的吗？谢谢
[image]

shu0802, 2022-05-17:
> 
假如已经冒充了，这个命令是看不到的吧

shu0802, 2022-05-17:
> [image]
> 
> 被大雕老大吓坏了，赶紧lspci -vvv看了一下，如果造假能把B1/B2的片子系统里识别成B3的信息，那我真是识别不了了:(

假如冒充了，这个命令可能看不出来吧。

shu0802, 2022-05-17:
> rt, 新到的N5105 4网口 软路由，配的225-v B3 网卡，发现网卡经常随机断流重启，怎么回事啊？设备用风扇吹着，CPU才50多度，照例说不热，之前用的5.10的内核，看网上说升到5.15内核就好了，现在升到5.15版本，发现还是重启啊，一重启就要重拨号断网了，是驱动的问题吗，麻烦大佬看看。附送内核日志，那些寄存器dump的值看不懂，没手册。
> 
> ```
> [23089.373953] igc 0000:05:00.0 eth3: NIC Link is Down
> [23089.379432] br-lan: port 2(eth3) entered disabled state
> [23093.574044] igc 0000:05:00.0 eth3: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
> [23093.583431] br-lan: port 2(eth3) entered blocking state
> [23093.589179] br-lan: port 2(eth3) entered listening state
> [23095.625498] br-lan: port 2(eth3) entered learning state
> [23097.637401] br-lan: port 2(eth3) entered forwarding state
> [23097.643349] br-lan: topology change detected, propagating
> [28374.172849] igc 0000:02:00.0 eth0: NIC Link is Down
> [28374.178413] igc 0000:02:00.0 eth0: Register Dump
> [28374.183514] igc 0000:02:00.0 eth0: Register Name   Value
> [28374.189367] igc 0000:02:00.0 eth0: CTRL            081c0641
> [28374.195578] igc 0000:02:00.0 eth0: STATUS          40380681
> [28374.201721] igc 0000:02:00.0 eth0: CTRL_EXT        10000040
> [28374.207864] igc 0000:02:00.0 eth0: MDIC            18017949
> [28374.213988] igc 0000:02:00.0 eth0: ICR             00000001
> [28374.220124] igc 0000:02:00.0 eth0: RCTL            0440803a
> [28374.226253] igc 0000:02:00.0 eth0: RDLEN[0-3]      00001000 00001000 00001000 00001000
> [28374.234952] igc 0000:02:00.0 eth0: RDH[0-3]        00000049 0000005a 00000000 000000c2
> [28374.243650] igc 0000:02:00.0 eth0: RDT[0-3]        00000048 00000059 000000ff 000000c1
> [28374.252340] igc 0000:02:00.0 eth0: RXDCTL[0-3]     02040808 02040808 02040808 02040808
> [28374.261032] igc 0000:02:00.0 eth0: RDBAL[0-3]      0d0e9000 0d0ec000 0d0ef000 0d132000
> [28374.269724] igc 0000:02:00.0 eth0: RDBAH[0-3]      00000001 00000001 00000001 00000001
> [28374.278418] igc 0000:02:00.0 eth0: TCTL            a503f0fa
> [28374.284544] igc 0000:02:00.0 eth0: TDBAL[0-3]      0d0d7000 0d0dc000 0d0e1000 0d0e6000
> [28374.293236] igc 0000:02:00.0 eth0: TDBAH[0-3]      00000001 00000001 00000001 00000001
> [28374.301929] igc 0000:02:00.0 eth0: TDLEN[0-3]      00001000 00001000 00001000 00001000
> [28374.310621] igc 0000:02:00.0 eth0: TDH[0-3]        0000003d 00000032 000000ca 000000b6
> [28374.319314] igc 0000:02:00.0 eth0: TDT[0-3]        0000003d 00000032 000000ca 000000b6
> [28374.328013] igc 0000:02:00.0 eth0: TXDCTL[0-3]     02100108 02100108 02100108 02100108
> [28374.336707] igc 0000:02:00.0 eth0: Reset adapter
> [28393.212179] igc 0000:02:00.0 eth0: NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
> [28403.443916] pppoe-vwan2: renamed from ppp1
> [28403.475957] pppoe-vwan1: renamed from ppp2
> [28422.615871] pppoe-vwan1: renamed from ppp2
> [28441.803773] pppoe-vwan1: renamed from ppp2
> [28461.063752] pppoe-vwan1: renamed from ppp2
> ```
[image]

看了一下大雕是用的你的日志里面的部分内容帖的图

GreatMichaelLee, 2022-05-17:
> > [image] 很明显是 B1/B2 冒充 B3 的缺陷版 I225
> 
> 问一下大佬，你用的啥命令查看的，查看网卡就是显示如下图的结果，看不到控制器的参数。你的图中结果表示是B3的吗？谢谢 [image]

肯定是这个B3啊，不过我lspci 有个“other'设备会报个错，不知道啥玩意，不管他
[image]

Jejz168, 2022-05-19:
从来就没认为225网卡好。哪来这么多b3。你看j4125的210都比225贵。还是买210吧。

pipihu, 2022-05-20:
我的j4125 i225 b3 没断流过

suizhenmin, 2022-05-21:
225被称为软网卡，也不知道是不是这样，目前看来还是螃蟹的2.5 稳定的多，英特尔在2.5网卡看来是拉夸的很

regunakyle, 2022-05-22:
> [image] 很明显是 B1/B2 冒充 B3 的缺陷版 I225

你好,能不能讲一下怎样判断网卡真伪? 我买的是康耐信的J4125, 有时候会断流, 想检查一下

mrzj110, 2022-05-24:
> > [image]很明显是 B1/B2 冒充 B3 的缺陷版 I225
> 
> 你好,能不能讲一下怎样判断网卡真伪?我买的是康耐信的J4125, 有时候会断流, 想检查一下
你是物理机安装吗

shu0802, 2022-05-24:
> 同样的问题
>
> ```
> Sun May  8 22:19:35 2022 kern.info kernel: [19485.217376] igc 0000:02:00.0 eth1: NIC Link is Down
> Sun May  8 22:19:35 2022 kern.info kernel: [19485.222627] br-lan: port 1(eth1) entered [...]

问一下，是不是只有一个网口是这样或者是全都是这样，可以换网口测试一下。今天我的也出现这样的日志，但只有一个口，另外一个口正常。是否与网线连接的设备的网口有关。
[image]
[image]

regunakyle, 2022-05-24:
> 物理机

我是在PVE上安装pfSense
每天回家后用电脑都会断几下 然后就不会断

1064298219, 2022-06-09:
我的也是  总是ikuai掉wan口 其他口都不掉 换了wan口的接口也还是一样掉  测速的时候很大概论掉

Jejz168, 2022-06-09:
发热大就会掉，这就是为什么225那么便宜的问题

3qp, 2022-06-10:
618前也是赶上了这趟车，看你们的问题，整个人都不好了。

Cregin, 2022-06-10:
前几天刚把225退了，换210

sbwcwfhv001, 2022-06-11:
我的j4125用的我三月份的固件没有你说的问题，我现在没有用pve了，直接u盘opwnwrt，运行很稳定，做旁路由。你其实不用急着退，可以换固件试试

| |
sbwcwfhv
|
|
***@***.***
|

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2022年05月11日 11:36 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.******@***.***> |
| 主题 | Re: [coolsnowwolf/lede] I225-v B3 2.5G口老是断流重启是怎么回事？ (Issue #9371) |

说明对虚拟设备支持得好，对物理设备反而支持得不好

shu0802, 2022-06-29:
> 我的j4125用的我三月份的固件没有你说的问题，我现在没有用pve了，直接u盘opwnwrt，运行很稳定，做旁路由。你其实不用急着退，可以换固件试试 | | sbwcwfhv | | ***@***.*** |
> […](#)

我就是你的想法，结果失去了退货的时间。在物理机下的op、爱快、win10三个系统测试6口的都变成5口了，现在已经找商家第二次申请售后了。
[image]
[image]
[image]

jerry5214, 2022-07-24:
真正的原因估计你的I225根本就不是B3，好多B1/B2打磨冒充B3的。要不你以为N5105+4*I225B3为什么这么便宜。

LiHang941, 2022-07-25:
> 要不你以为N5105+4*I225B3为什么这么便宜。

有办法排查吗

unliar, 2022-08-22:
on Aug 22 12:16:33 2022 daemon.notice netifd: Interface 'wan' has link connectivity
Mon Aug 22 12:16:33 2022 daemon.notice netifd: Interface 'wan' is setting up now
Mon Aug 22 12:16:33 2022 kern.info kernel: [59874.607518] igc 0000:00:10.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX 

我也遇到了 间歇性重启

unliar, 2022-08-24:
> on Aug 22 12:16:33 2022 daemon.notice netifd: Interface 'wan' has link connectivity Mon Aug 22 12:16:33 2022 daemon.notice netifd: Interface 'wan' is setting up now Mon Aug 22 12:16:33 2022 [...]

看起来是扁平网线的问题，把 光猫 -> WAN 口的网线换成圆的，现在已经快20个小时没问题了。

不要使用扁平网线，会变得不幸。

[image]

AstroProfundis, 2022-09-14:
和楼上一样问题，之前用的扁平网线，PVE 7.2, 换内核改设置都没用，端口就一两个小时 up/down 一次，换成六类圆线好了...

ikidou, 2022-12-07:
> 我的j4125 i225 b3 没断流过

请问你的是哪个型号呢

webxiaoz, 2023-02-20:
用esxi估计会好很多，我用esxi7.0u3，OpenWrt主路由一直稳定,之前用pve倒是出现过一测速立马掉的问题

coolsnowwolf, 2026-09-05:
这是 2023 年 N5105/I225-V B3 的硬件链路复位报告，当前仓库已使用更新的 igc/内核栈，但 issue 没有可在当前版本复现的配置、PCI/PHY 信息或测试结果，无法归因于 LEDE 代码，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf