coolsnowwolf/lede #9373: 编译失败，advanced报错。

sjkhsl, 2022-05-08:
编译失败。advanced插件和什么有冲突吗？

```
Collected errors:
 * check_data_file_clashes: Package luci-app-fileassistant wants to install file /home/sjkhsl/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/controller/fileassistant.lua
        But that file is already provided by package  * luci-app-advanced
 * check_data_file_clashes: Package luci-app-fileassistant wants to install file /home/sjkhsl/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/view/fileassistant.htm
        But that file is already provided by package  * luci-app-advanced
 * check_data_file_clashes: Package luci-app-fileassistant wants to install file /home/sjkhsl/lede/build_dir/target-x86_64_musl/root-x86/www/luci-static/resources/fileassistant/fb.css
        But that file is already provided by package  * luci-app-advanced
 * check_data_file_clashes: Package luci-app-fileassistant wants to install file /home/sjkhsl/lede/build_dir/target-x86_64_musl/root-x86/www/luci-static/resources/fileassistant/fb.js
        But that file is already provided by package  * luci-app-advanced
 * check_data_file_clashes: Package luci-app-fileassistant wants to install file /home/sjkhsl/lede/build_dir/target-x86_64_musl/root-x86/www/luci-static/resources/fileassistant/file-icon.png
        But that file is already provided by package  * luci-app-advanced
 * check_data_file_clashes: Package luci-app-fileassistant wants to install file /home/sjkhsl/lede/build_dir/target-x86_64_musl/root-x86/www/luci-static/resources/fileassistant/folder-icon.png
        But that file is already provided by package  * luci-app-advanced
 * check_data_file_clashes: Package luci-app-fileassistant wants to install file /home/sjkhsl/lede/build_dir/target-x86_64_musl/root-x86/www/luci-static/resources/fileassistant/link-icon.png
        But that file is already provided by package  * luci-app-advanced
 * opkg_install_cmd: Cannot install package luci-app-fileassistant.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/sjkhsl/lede'
make[1]: *** [package/Makefile:111: /home/sjkhsl/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/sjkhsl/lede'
make: *** [/home/sjkhsl/lede/include/toplevel.mk:230: world] Error 2

```

WYC-2020, 2022-05-08:
只能编译其中一个，冲突而已

sjkhsl, 2022-05-09:
> 只能编译其中一个，冲突而已

和哪个冲突？

goofysun, 2022-05-24:
luci-app-advanced 已经包含了 luci-app-fileassistant ，
去掉 luci-app-fileassistant 就可以了

closed 2022-06-15 by AmadeusGhost