coolsnowwolf/lede #9377: IPQ8072A编译成功，但是没有固件生成（但是所有插件的ipk文件成功生成）

Yofo-me, 2022-05-08:
### 路由器配置如下：
```shell 
CPU: IPQ8072A
无线芯片： QCN5024+QCN5054+QCA8081
额外接口： 1个USB3.0接口，两个2.5G以太网口，1个USB console接口，一个以太网千兆网口（支持POE）
系统内存： 1G
系统存储： 4G emmc
```

### 编译时部分配置选择截图如下：
1.
[image]

2.
[image]

3.
[image]

 ### 完整配置文件如下
- 第一次：[ipq807x.config.zip](https://github.com/coolsnowwolf/lede/files/8647062/ipq807x.config.zip)
- 第二次：[arm-ipq8072a.config.zip](https://github.com/coolsnowwolf/lede/files/8647067/arm-ipq8072a.config.zip)

### 现在的问题是，编译成功，但是固件类似于bin这样的文件根本没有，倒是固件编译时选择的插件的ipk文件全部生成了，如图
[image]

### 尝试过`make clean `,`make distclean`再重新选择编译还是同样的问题，顺带说明，无线是厂家那边直接提供的bin文件的闭源驱动。

hcym, 2022-05-08:
啥机子，多少钱的

Yofo-me, 2022-05-09:
现在编译爆出这个问题
[image]

Yofo-me, 2022-05-09:
> 啥机子，多少钱的

淘宝。深圳市康源智能有限公司。店买的，1500

closed 2023-12-22 by aiamadeus