coolsnowwolf/lede #9383: 请教大神如何让openwrt登录时自动运行sh

gztingting, 2022-05-10:
由于家里的软路由设置了frp内网穿透，最近在服务器端看到有可疑ip连接到软路由的端口，但不知道有没有成功登录进luci，我想用Telegram bot来自动发送登录信息来提高openwrt的安全性（root密码已经是很复杂的了）。
类似于在Linux服务器下通过ssh登录可以自动发送登陆信息。
[image]
详见https://github.com/mythevalentinus/ssh-login-alert-telegram
这个项目。
Ubuntu/Debian是在/etc/profile.d/下的Telegram-alert.sh,只要有人用ssh登陆就会运行这个sh，
实际是在/opt/ssh-login-alert-telegram文件夹相应文件里配置好Telegram Bot的token参数，然后运行alert.sh
不确定这个项目能不能配置到openwrt里运行，或者有没有相应的插件可以达到类似功能。
请问openwrt是在哪里放置Luci登录自动运行sh，我知道/etc/init.d是开机运行目录，这和linux是/etc/systemd/目录不太一样，而ssh登陆自动运行目录/etc/profile.d在openwrt下似乎也没有，我想知道Luci网页登录和SSH登录时即自动运行sh的目录在哪里。

hxlls, 2022-05-10:
直接写运行脚本丢/etc/init.d下就行了吧

shu0802, 2022-05-10:
计划任务里可以执行

xpader, 2022-05-11:
ssh-login-alert-telegram 这个项目可以看到它有一个 deploy.sh，原理就是把 alert.sh 脚本放在全局 /etc 下相关 shell 的初始化脚本中，当用户通过 ssh 登录时，会启动自己的 shell，比如是 bash 或 zsh，那么 /etc 下的 bash 或 zsh 初始脚本就会被执行，就是这么个原理。

你看 openwrt 下 ssh 用的是什么 shell，然后在 /etc 下对应的全局配置加上就行了。

至于 Luci 登录怎么执行就不知道了。

EnnawYang, 2022-05-11:
不如直接用serverchan？

gztingting, 2022-05-11:
> 直接写运行脚本丢/etc/init.d下就行了吧，再RC 下创建个软链接，启动顺序搞下完事

/etc/init.d/这个目录下的sh是开机即运行，我想达到的目标是开机时不允许，只有发生luci登录（即从浏览器进去web管理界面时）才运行特定的sh。

gztingting, 2022-05-11:
> 不如直接用serverchan？

用serverchan和Telegram bot都是一样的，两者都是即时消息接口（消息机器人）
都可以通过get、post、curl等method的格式化消息发到自己的消息机器人上
现在我想的是通过什么途径触发curl动作

ZeaKyX, 2022-05-12:
> > 不如直接用serverchan？
> 
> 用serverchan和Telegram bot都是一样的，两者都是即时消息接口（消息机器人） 都可以通过get、post、curl等method的格式化消息发到自己的消息机器人上 现在我想的是通过什么途径触发curl动作

有没有一种可能，他说的是插件[luci-app-serverchan](https://github.com/tty228/luci-app-serverchan)

GreatMichaelLee, 2022-05-12:
现在serverchan有没有靠谱一点的push channel啊，那些wxpush,pushpush都被WX限的很厉害，界面搞得贼不友好

gztingting, 2022-05-12:
> 

luci-app-serverchan推送的内容有ipv4变化、设备上下线、CPU负载和温度、内网连接的设备情况等，我觉得相对有用一点的可能是ipv6地址，v4都是内网接口ip和外网共享ip，没啥用。是通过cron实现的。我想要的是通过luci登录触发，不知道怎么搞。

gztingting, 2022-05-12:
> 现在serverchan里有没有靠谱一点的push channel啊，那些wxpush,pushplus都被WX限的很厉害，界面搞得贼不友好，其他的没试过，哪个体验好一点？

我用Push+还算比较正常。
用Telegram bot可能会更加友好一些。

ZeaKyX, 2022-05-12:
> > 
> 
> luci-app-serverchan推送的内容有ipv4变化、设备上下线、CPU负载和温度、内网连接的设备情况等，我觉得相对有用一点的可能是ipv6地址，v4都是内网接口ip和外网共享ip，没啥用。是通过cron实现的。我想要的是通过luci登录触发，不知道怎么搞。

https://github.com/tty228/luci-app-serverchan/blob/master/luasrc/model/cbi/serverchan/setting.lua#L277
https://github.com/tty228/luci-app-serverchan/blob/master/root/usr/bin/serverchan/serverchan#L979

gztingting, 2022-05-13:
> https://github.com/tty228/luci-app-serverchan/blob/master/luasrc/model/cbi/serverchan/setting.lua#L277 https://github.com/tty228/luci-app-serverchan/blob/master/root/usr/bin/serverchan/serverchan#L979

刚更新了最新版的serverchan，有web登陆和ssh登录信息提醒，但是web登录后一直没有触发信息发送，不知道是哪里的问题。ssh登录触发的信息是127.0.0.1透过pubkey登录了路由器。我是在外网通过内网穿透连接的ssh，似乎它获取客户端ip地址
上没有https://github.com/mythevalentinus/ssh-login-alert-telegram
这个项目那么彻底。

closed 2022-05-15 by gztingting