coolsnowwolf/lede #9384: K2P编译时选择闭源驱动与开源驱动问题

Hess488, 2022-05-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：以下问题困扰了我很久去网上查了很多帖子资料也找不到具体解释，希望大佬解答一下这几个问题，我相信对于以后编译k2p路由器的人也会有很大的帮助，万分感谢
一.
k2p路由器在编译时没改动任何驱动选项，选择插件那里默认也没勾选luci-app-mtwifi的情况下是闭源驱动还是开源驱动？
二.
K2P路由器在编译时想要选择闭源驱动应该怎么操作，需要打开哪些选项关闭哪些选项？
三.
K2P路由器在编译时想要选择开源驱动应该怎么操作，需要打开哪些选项关闭哪些选项？
四.
luci选择插件时里面的luci-app-mtwifi插件对于驱动有什么用
### (2) 路由器型号和固件版本

A：斐讯K2P路由器，最新lede仓库源码

### (3) 详细日志

A：无

EnnawYang, 2022-05-10:
为什么要重复开issue？

Hess488, 2022-05-10:
> 为什么要重复开issue？

没有重复开issue，昨天的问题是为啥编译失败，后来发现是冲突了取消一个驱动就能编译。现在发的几个问题是另外的问题，想寻求解答。大佬能花点时间解答一下吗，真心想知道那几个问题，网上找资料找了好久都没具体回复，想进lede大的群问问也进不去没密码

uniessey, 2022-05-11:
> > 为什么要重复开issue？
> 
> 没有重复开issue，昨天的问题是为啥编译失败，后来发现是冲突了取消一个驱动就能编译。现在发的几个问题是另外的问题，想寻求解答。大佬能花点时间解答一下吗，真心想知道那几个问题，网上找资料找了好久都没具体回复，想进lede大的群问问也进不去没密码

这个事情在多年前有过非常详细的讨论，您可以搜索一下，您所问的问题都有非常详细的论述。

Hess488, 2022-05-11:
> > > 为什么要重复开issue？
> 
> > 
> 
> > 没有重复开issue，昨天的问题是为啥编译失败，后来发现是冲突了取消一个驱动就能编译。现在发的几个问题是另外的问题，想寻求解答。大佬能花点时间解答一下吗，真心想知道那几个问题，网上找资料找了好久都没具体回复，想进lede大的群问问也进不去没密码
> 
> 
> 
> 这个事情在多年前有过非常详细的讨论，您可以搜索一下，您所问的问题都有非常详细的论述。

搜了关键词闭源驱动，k2p。还是没发现，搜的时候发现多年前其他路由器的驱动配置方法到是有人讨论

chobitsos, 2022-05-11:
> Author

K2P的开源驱动直到今天为止都是渣渣，大概只有闭源驱动的80%左右的效能，带机超过20个就拖后腿
就算带机量不高，信号虚标还是很严重

Hess488, 2022-05-11:
> > Author
> 
> 
> 
> K2P的开源驱动直到今天为止都是渣渣，大概只有闭源驱动的80%左右的效能，带机超过20个就拖后腿
> 
> 就算带机量不高，信号虚标还是很严重，另外闭源驱动才能开启hwnat，开源驱动只能靠sfe提升nat性能

所以我想使用闭源驱动，但是不知道不更改驱动设置，不选择luci-app-mtwifi直接编译出来的固件是不是闭源驱动，问了好几个大佬了都没人给个明确的回复

Hess488, 2022-05-11:
> > > > Author
> 
> > > 
> 
> > > 
> 
> > > K2P的开源驱动直到今天为止都是渣渣，大概只有闭源驱动的80%左右的效能，带机超过20个就拖后腿
> 
> > > 就算带机量不高，信号虚标还是很严重，另外闭源驱动才能开启hwnat，开源驱动只能靠sfe提升nat性能
> 
> > 
> 
> > 所以我想使用闭源驱动，但是不知道不更改驱动设置，不选择luci-app-mtwifi直接编译出来的固件是不是闭源驱动，问了好几> 个大佬了都没人给个明确的回复
> 
> 
> 
> 闭源驱动必须l勾选uci-app-mtwifi，要不然闭源驱动起不来，你可以不用luci-app-mtwifi的luci，不勾选的话没有启动脚本，你得自己写启动脚本，管理LUCI的话闭源就是渣渣，自己写个LUCI管理都比luci-app-mtwifi的管理要好
> 
> 

- - 我驱动配置默认啥也没动，只勾选了luci-app-mtwifi，开始编译之后提示错误mtwifi跟kmod-mt7615d_dbdc冲突，必须取消 kmod-mt7615d_dbdc才能编译成功，这是为啥。我看驱动那里默认只勾选了两个选项，kmod-mt7615d_dbdc和kmod-mt7615d，mt7615d_dbdc跟mt7615d两者有啥关系吗，为啥mt7615d不会跟mtwifi冲突，而7615d_dbdc就会跟mtwifi冲突导致编译出错

chobitsos, 2022-05-11:
> * * 我驱动配置默认啥也没动，只勾选了luci-app-mtwifi，开始编译之后提示错误mtwifi跟kmod-mt7615d_dbdc冲突，必须取消 [...]

一样得，你把mt7615d_dbd取消掉不就行了，实在不行换一个SDK反正都是万年老坑，能用就行了，不用顾及那么多

coolsnowwolf, 2022-05-11:
冷知识，目前 K2P 默认用的就是闭源驱动，只不过专门写了原生 luci 而已，不知道你在说啥。。。

Hess488, 2022-05-11:
> 冷知识，目前 K2P 默认用的就是闭源驱动，只不过专门写了原生 luci 而已，不知道你在说啥。。。

啊这样的吗，我是看到网上k2p编译openwrt的教程说的，必须选上luci-app-mtwifi这个插件才是闭源驱动，所以按您说的不需要选这个插件默认也是闭源驱动。那这个插件插件是干啥的呀，对k2p有用吗

coolsnowwolf, 2022-05-11:
luci-app-mtwifi 是给别的路由器闭源驱动控制用的。K2P 是特别处理的，做了原生的luci适配

lphgor, 2022-05-12:
> 

开源是kmod-mt7615e，闭源是kmod-mt7615d

vsvkhs, 2022-05-13:
> 

老大能不能给AC85P和dir882这类7615机器也用上闭源驱动和原生Luci呀

britneyjeanspears, 2022-05-16:
> > > 另外闭源驱动才能开启hwnat，开源驱动只能靠sfe提升nat性能
> > 
> > 
> > 所以我想使用闭源驱动，但是不知道不更改驱动设置，不选择luci-app-mtwifi直接编译出来的固件是不是闭源驱动，问了好几个大佬了都没人给个明确的回复
> 
> 闭源驱动必须l勾选luci-app-mtwifi，要不然闭源驱动起不来，你可以不用luci-app-mtwifi的管理界面，不勾选的话没有启动脚本，你得自己写启动脚本，luci管理界面的话闭源就是渣渣，自己写个LUCI管理界面都比luci-app-mtwifi的管理界面要好 闭源必勾luci-app-mtwifi，除非你会自己写启动脚本，我的做法

不懂装懂。谁给你说的k2p的闭源要luci-app-mtwifi？

chobitsos, 2022-05-16:
> 不懂装懂。谁给你说的k2p的闭源要luci-app-mtwifi？

我没说一定要啊，我还把我做法写出来了，我不说不勾选的话那就把启动脚本弄出来，但是lean的SDK做了适配，带原生luci管理，也不需要luci-app-mtwifi，但是如果不用这里的SDK，你拉官方代码，在拉大雕的仓库就必须得勾上luci-app-mtwifi，不勾上的话你就把启动脚本单独弄出来，有什么不对吗，你自己试试不就知道了！

britneyjeanspears, 2022-05-17:
> > > > > 另外闭源驱动才能开启hwnat，开源驱动只能靠sfe提升nat性能
> > > > 
> > > > 
> > > > 所以我想使用闭源驱动，但是不知道不更改驱动设置，不选择luci-app-mtwifi直接编译出来的固件是不是闭源驱动，问了好几个大佬了都没人给个明确的回复
> > > 
> > > 
> > > 如果不是用这里的SDK，那闭源驱动必须l勾选luci-app-mtwifi，要不然闭源驱动起不来，你可以不用luci-app-mtwifi的管理界面，不勾选的话没有启动脚本，你得自己写启动脚本，luci管理界面的话闭源就是渣渣，自己写个LUCI管理界面都比luci-app-mtwifi的管理界面要好
> > > 不是这里的源码闭源必勾luci-app-mtwifi，除非你会自己写启动脚本，我的做法是
> > 
> > 
> > 不懂装懂。谁给你说的k2p的闭源要luci-app-mtwifi？
> 
> 我没说一定要啊，我还把我做法写出来了，我说不勾选的话那就把启动脚本弄出来，但是lean的SDK做了适配，带原生luci管理，也不需要luci-app-mtwifi，但是如果不用这里的SDK，你拉官方代码，在拉大雕的仓库就必须得勾上luci-app-mtwifi，不勾上的话你就把启动脚本单独弄出来，要不然驱动起不来，有什么不对吗，你自己试试不就知道了！

(2) 路由器型号和固件版本

A：斐讯K2P路由器，最新lede仓库源码

别人一楼就说了用的这里的仓库，然后你回复必须用luci-app-mtwifi，是不是来误导别人？你在昨天编辑之前回复说的必须用luci-app-mtwifi，导致楼主勾上luci-app-mtwifi反倒出错，坑。

britneyjeanspears, 2022-05-17:
> * * 我驱动配置默认啥也没动，只勾选了luci-app-mtwifi，开始编译之后提示错误mtwifi跟kmod-mt7615d_dbdc冲突，必须取消 [...]

看吧，楼主就这样被误导了

chobitsos, 2022-05-19:
> 看吧，楼主就这样被误导了

确实误导了楼主，很抱歉啊，表达能力不好

closed 2022-05-21 by AmadeusGhost

ewangsoft, 2022-06-27:
我也是K2P，用的lean的MT76驱动，mt7615d和mt7615d_dbdc，这两个都是闭源驱动吧，前者和后者有何功能和性能方面的差别？