coolsnowwolf/lede #9388: ttyd到底怎样配https访问？

GreatMichaelLee, 2022-05-12:
在/etc/config/ttyd里增加配了以下三项： (cert和key路径和证书都是有效的，不用怀疑）

```
        option ssl '1'
        option ssl_cert '/mnt/sda3/cert/ttyd.crt'
        option ssl_key '/mnt/sda3/cert/ttyd.key'
```

然后service ttyd restart,总是报错啊： -S参数为什么rc.common说找不到？/etc/init.d/ttyd里的scheme有啊。

root@N5105U:/tmp/etc# service ttyd restart
/etc/rc.common: eval: line 66: -S: not found

GreatMichaelLee, 2022-05-12:
参考了#8438 没用，而且不太想改变Init脚本的方式。

GreatMichaelLee, 2022-05-13:
你们都不用ttyd的么...

galaxyskyknight, 2022-05-14:
> 这个我试过，直接在ttyd luci界面配置勾上ssl,并且把证书和密钥路径填写上。就能正常使用。那个证书用的nginx。

lede的好像没有ttyd的luci配置界面只有登录界面啊，menuconfig里没有...

GreatMichaelLee, 2022-05-16:
> 这个我试过，直接在ttyd luci界面配置勾上ssl,并且把证书和密钥路径填写上。就能正常使用。那个证书用的nginx。

LEDE的仓库里luci下面哪有你说的这个什么配置界面？

bauw2008, 2022-05-16:
配置文件修改下也可以呀。

config ttyd
	option interface '@lan'
	option command '/bin/login'
	option ssl '1'
	option ssl_cert '/etc/nginx/conf.d/_lan.crt'
	option ssl_key '/etc/nginx/conf.d/_lan.key'
	option debug '7'

GreatMichaelLee, 2022-05-16:
> 配置文件修改下也可以呀。
> 
> config ttyd option interface '@lan' option command '/bin/login' option ssl '1' option ssl_cert '/etc/nginx/conf.d/_lan.crt' option ssl_key '/etc/nginx/conf.d/_lan.key' option debug '7'

噗。。。。感情你是没看我文章第一篇的内容，就是这样配的啊，重启服务报错啊。。。rc.common不能解析配置文件里ssl '1'所对应的binary的运行参数-S，不知道咋回事

v03413, 2022-05-17:
我没具体测试过，但你可以尝试一下：`利用uhttpd反代tty，再在uhttpd配置客户端`

ZqinKing, 2022-07-26:
> 我没具体测试过，但你可以尝试一下：`利用uhttpd反代ttyd，再在uhttpd配置证书`

不行滴，需要改lede里面的配置。

coolsnowwolf, 2026-09-05:
当前 ttyd init 脚本已将 `ssl` 转换为 `-S`，并将证书和密钥分别传递为 `-C`/`-K`；报告中的 `/etc/rc.common: -S: not found` 问题已修复，关闭为“已完成”。

closed 2026-09-05 by coolsnowwolf