coolsnowwolf/lede #9390: 5.4.188固件，有线Lan口各种正常，就是无法上网，电脑ifconfig查看RX dropped 100%，救救孩子

zbluekuma, 2022-05-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述
按默认设置自编译的LEDE固件，只增加了一些自定义软件。WIFI上网一切正常，但是LAN口连接，浏览器无法上网，微信可以发送信息图片。从2月份一直折腾到现在，期间编译了5.4、5.10、5.15各版本固件都是同样问题。用固件自带软件诊断功能，默认openwrt.org网址，诊断IP4测试、TRACERoute测试、NSLOOKUP测试都显示一切正常，感觉路由器和外部链接都正常。交换机设置是固件初始默认设置，有线与路由器之间连接的电脑，登录路由器管理页面，ping局域网内其他电脑都正常，ssh可以登录。因为是小众路由器，只能自己编译固件，加上其他一切都好，只有这个lan口问题，还请大神多多指导。

### (1) 具体问题

A：最近用IFCONFIG查看网卡信息，发现LAN口连接路由器，外网数据RX dropped 100%

### (2) 路由器型号和固件版本

A：路由器型号：ARRIS SBR-AC1750（CPUMark：1921.560307 Scores）；
      架构：Qualcomm Atheros QCA9558 ver1 rev 0x1  交换芯片：QCA8337  无线芯片：QCA9880
      固件版本：OpenWrt R22.4.1/LuCi aster（git-22 .091.36880-c3be048）

### (3) 详细日志

A：
`# ifconfig情况
br-lan    Link encap:Ethernet  HWaddr D4:04:CD:FE:FD:AD  
          inet addr:192.168.50.1  Bcast:192.168.50.255  Mask:255.255.255.0
          inet6 addr: 2409:8a31:489:eb70::1/60 Scope:Global
          inet6 addr: fe80::d604:cdff:fefe:fdad/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:23262 errors:0 dropped:0 overruns:0 frame:0
          TX packets:43006 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:4618785 (4.4 MiB)  TX bytes:40538970 (38.6 MiB)

eth0      Link encap:Ethernet  HWaddr D4:04:CD:FE:FD:AD  
          inet6 addr: fe80::d604:cdff:fefe:fdad/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:21683 errors:0 dropped:0 overruns:0 frame:0
          TX packets:42108 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:4645983 (4.4 MiB)  TX bytes:40135215 (38.2 MiB)
          Interrupt:4 

eth0.1    Link encap:Ethernet  HWaddr D4:04:CD:FE:FD:AD  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:21683 errors:0 dropped:0 overruns:0 frame:0
          TX packets:42084 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:4255689 (4.0 MiB)  TX bytes:39962669 (38.1 MiB)

eth1      Link encap:Ethernet  HWaddr D4:04:CD:FE:FD:AE  
          inet6 addr: fe80::d604:cdff:fefe:fdae/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:29515 errors:0 dropped:0 overruns:0 frame:0
          TX packets:17158 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:25886485 (24.6 MiB)  TX bytes:4218341 (4.0 MiB)
          Interrupt:5 

eth1.2    Link encap:Ethernet  HWaddr D4:04:CD:FE:FD:AE  
          inet6 addr: fe80::d604:cdff:fefe:fdae/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          **RX packets:29515 errors:0 dropped:435 overruns:0 frame:0**
          TX packets:17150 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:25355215 (24.1 MiB)  TX bytes:4148845 (3.9 MiB)

lo        Link encap:Local Loopback  

`
[... 284 lines cut ...]
[  124.109143] ath: regdomain 0x8348 dynamically updated by user
[  128.646714] br-lan: port 2(wlan1) entered blocking state
[  128.652157] br-lan: port 2(wlan1) entered disabled state
[  128.657890] device wlan1 entered promiscuous mode
[  131.126135] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[  131.143466] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[  131.149874] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[  131.158045] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 24984 iram: 38672'
[  131.241696] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[  131.256563] ath10k_pci 0000:00:00.0: rts threshold -1
[  131.295472] br-lan: port 3(wlan0) entered blocking state
[  131.300903] br-lan: port 3(wlan0) entered disabled state
[  131.306688] device wlan0 entered promiscuous mode
[  133.397738] pppoe-wan: renamed from ppp1
[  133.511598] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  133.518224] br-lan: port 3(wlan0) entered blocking state
[  133.523657] br-lan: port 3(wlan0) entered forwarding state
[  138.661687] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 2 is up
[  140.673625] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  140.680569] br-lan: port 2(wlan1) entered blocking state
[  140.686005] br-lan: port 2(wlan1) entered forwarding state
[  197.585413] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[  197.593562] NFSD: Using legacy client tracking operations.
[  197.599127] NFSD: starting 10-second grace period (net f0000029)
`

GreatMichaelLee, 2022-05-14:
你是哪里看出外网drop 100%了？我还是建议你检查cpu负载，网卡软中断，以及DNS解析情况。

zbluekuma, 2022-05-15:
> 你是哪里看出外网drop 100%了？我还是建议你检查cpu负载，网卡软中断，以及DNS解析情况。
@GreatMichaelLee 您好!
路由器cpu负载很低，测试lan连接时，家里几乎无什么设备上网。
DNS解析确实没有问题的，ping 外网域名都可以正常解析。
想了很多法子，最近IFCONFIG的时候看到虚拟网卡eth1.2的RX dropped有丢包，因为对这个实在不懂，感觉是这里问题。搜索RX droppede含义，说是表示数据包已经进入了Ring Buffer，但是由于内存不够等系统原因，导致在拷贝的内存的过程中被丢弃。所以猜测是不是这里出了问题。
网卡软中断的问题，我实在排查能力有限，还请各位专家指导我。真想解决了这个问题，除了这个lan口问题，其他一切都好。

yanxin152133, 2022-05-17:
过个几分钟不就正常了

zbluekuma, 2022-05-18:
> 过个几分钟不就正常了

一直没有正常的，通过有线Lan口接入的电脑，无法浏览器上网，把人都逼疯了。因为这一点退回到原始社会又不甘心。但是有线不能用，现在家里又用旧笔记搭建了云笔记和备份服务器。

yanxin152133, 2022-05-18:
> > 过个几分钟不就正常了
> 
> 一直没有正常的，通过有线Lan口接入的电脑，无法浏览器上网，把人都逼疯了。因为这一点退回到原始社会又不甘心。但是有线不能用，现在家里又用旧笔记搭建了云笔记和备份服务器。

我用的5.4.155版本，每次重启了之后得等几分钟网络才算正常，没用过有线的

bigbugcc, 2022-05-23:
> > 过个几分钟不就正常了
> 
> 一直没有正常的，通过有线Lan口接入的电脑，无法浏览器上网，把人都逼疯了。因为这一点退回到原始社会又不甘心。但是有线不能用，现在家里又用旧笔记搭建了云笔记和备份服务器。

遇到同样的问题双LAN口，有一个正常 另一个出现此情况 能ping通就是打不开网页

yanxin152133, 2022-05-23:
> 

我感觉好像是dns的问题，我用的adguardhome，每次重启完就得等几分钟网络才能正常

bigbugcc, 2022-05-23:
> > 
> 
> 我感觉好像是dns的问题，我用的adguardhome，每次重启完就得等几分钟网络才能正常

我的没用Ad插件😂，电脑端的DNS也清过 nslookup都是正常的

yanxin152133, 2022-05-23:
自带的dhcp里面的dns 转发反正每次重启都恢复成默认的

bigbugcc, 2022-05-23:
> 自带的dhcp里面的dns 转发反正每次重启都恢复成默认的

你可以在LAN口高级设置 指定客户端使用的dns地址

yanxin152133, 2022-05-23:
> bigbug *@*.
> *>于2022年5月23日 周一下午8:16写道： 我感觉好像是dns的问题，我用的adguardhome，每次重启完就得等几分钟网络才能正常
> on GitHub <#9390 (comment)
> <https://github.com/coolsnowwolf/lede/issues/9390#issuecomment-1134600487>>,
> 自带的dhcp里面的dns 转发反正每次重启都恢复成默认的
>
> 你可以在LAN口高级设置 指定客户端使用的dns地址
我直接dhcp 下发了^_^

bigbugcc, 2022-05-23:
> 我直接dhcp 下发了^_^
> […](#)

折腾一天了 这个问题还是没有解决，两个LAN口 其中一个正常，另外一个死活不行 快哭了😭

yanxin152133, 2022-05-23:
> bigbug *@*.
>
>
>
> 我感觉好像是dns的问题，我用的adguardhome，每次重启完就得等几分钟网络才能正常 我的没用Ad插件😂，电脑端的DNS也清过
> <https://github.com/coolsnowwolf/lede/issues/9390> (comment) <#9390
> (comment)
> <https://github.com/coolsnowwolf/lede/issues/9390#issuecomment-1134600487>>>,
> 自带的dhcp里面的dns 转发反正每次重启都恢复成默认的 你可以在LAN口高级设置 指定客户端使用的dns地址 — Reply to this
> email directly, view it on GitHub <#9390 (comment)
> <https://github.com/coolsnowwolf/lede/issues/9390#issuecomment-1134618006>>,
> <#m_3552023774899928098_> Message ID: @.*>
>
> 折腾一天了 这个问题还是没有解决，两个LAN口 其中一个正常，另外一个死活不行 快哭了😭
你得发下设置

zbluekuma, 2022-05-24:
我的arris sbr-ac1750，有四个lan口，都尝试了，都不行，单凡有一个能用，我也不着急了。上面各位的建议都尝试了，我还是没有解决。准备重新下载个别人的老固件刷机去了。用了半年多的LEDE，很喜欢。依依不舍的离去

yanxin152133, 2022-05-24:
发下设置呗

zbluekuma, 2022-05-24:
> 发下设置呗 China.Z ***@***.***> 于2022年5月24日周二 20:14写道：
基本都是默认配置的。具体如下：

- [ ] #一、firewall配置

config defaults
	option syn_flood '1'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option fullcone '0'
	option flow_offloading '0'
	option flow_offloading_hw '0'

config zone
	option name 'lan'
	list network 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'

config zone
	option name 'wan'
	list network 'wan'
	list network 'wan6'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option src_ip 'fc00::/6'
	option dest_ip 'fc00::/6'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config rule
	option name 'Support-UDP-Traceroute'
	option src 'wan'
	option dest_port '33434:33689'
	option proto 'udp'
	option family 'ipv4'
	option target 'REJECT'
	option enabled 'false'

config include
	option path '/etc/firewall.user'

config include 'adbyby'
	option type 'script'
	option path '/var/etc/adbyby.include'
	option reload '1'

config rule 'adblock'
	option name 'adblock'
	option target 'DROP'
	option src 'wan'
	option proto 'tcp'
	option dest_port '8118'

config include 'mia'
	option type 'script'
	option path '/etc/mia.include'
	option reload '1'

config include 'shadowsocksr'
	option type 'script'
	option path '/var/etc/shadowsocksr.include'
	option reload '1'

config rule 'kms'
	option name 'kms'
	option target 'ACCEPT'
	option src 'wan'
	option proto 'tcp'
	option dest_port '1688'

- [ ] #二、network配置

config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd06:f71c:9fad::/48'

config interface 'lan'
	option type 'bridge'
	option ifname 'eth0.1'
	option proto 'static'
	option ipaddr '192.168.50.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

config interface 'wan'
	option ifname 'eth1.2'
	option _orig_ifname 'eth1.2'
	option _orig_bridge 'false'
	option proto 'pppoe'
	option username '*******'#隐私隐藏
	option password '*******'#隐私隐藏
	option ipv6 'auto'
	option keepalive '0'

config interface 'wan6'
	option ifname 'eth1.2'
	option _orig_ifname 'eth1.2'
	option _orig_bridge 'false'
	option proto 'pppoe'
	option username '*******'#隐私隐藏
	option password '*******'#隐私隐藏
	option ipv6 'auto'
	option keepalive '0'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option vid '1'
	option ports '0t 1 2 3 4'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option vid '2'
	option ports '5 6t'

- [ ] #三、dhcp配置

config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option nonwildcard '1'
	option localservice '1'
	option ednspacket_max '1232'
	option local '/lan/'
	option domain 'lan'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option dhcpv6 'server'
	option ra 'server'
	option ra_slaac '1'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config srvhost
	option srv '_vlmcs._tcp'
	option target 'Arris'
	option port '1688'
	option class '0'
	option weight '100'

yanxin152133, 2022-05-24:
> 基本都是默认配置的。具体如下：
>
>
>    - #一、firewall配置
>
> config defaults
> option syn_flood '1'
> option input 'ACCEPT'
> option output 'ACCEPT'
> option forward 'REJECT'
> option fullcone '0'
> option flow_offloading '0'
> option [...]
eth0.1用的桥接？
还有你的拨号是两个？
[image: image.png]

yanxin152133, 2022-05-24:
[image: image.png][image: image.png]

zbluekuma, 2022-05-24:
其实都是默认设置，我没改。wan、wan6、lan三个全部是LEDE默认生成的。我只是把wan和wan6改成了pppoe模式，所以两次拨号。
eth0.1的问题上，我今晚刷了另外一个固件，lan是好的，但是发现他的交换机设置：CPU（eth1）是对应Lan1-4口的，CPU（eth0）是对应wan口的。不知道是不是和这个有关系。

zbluekuma, 2022-05-24:
@yanxin152133
好的，谢谢。我试一下。这么晚麻烦您了。

zbluekuma, 2022-05-25:
[image]
想问一下，br-lan接口物理设置里面，为什么没有单独显示每个lan口。我的路由器应该有4个物理Lan口+1个物理wan口的

zbluekuma, 2022-05-25:
这是我的交换机界面截图，是LEDE默认的配置，但是我昨晚刷的另外一个openwrt固件，他的eth0是绑定wan口的，eth1是绑定4个lan口的。
[image]

yanxin152133, 2022-05-25:
交换机没玩过，看看你路由器

zbluekuma, 2022-05-25:
改成PPPOE连接后，发现内核日志有些变化。如下：
[   53.635915] eth0: link up (1000Mbps/Full duplex)
[   53.640620] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   53.664076] br-lan: port 1(eth0.1) entered blocking state
[   53.669561] br-lan: port 1(eth0.1) entered disabled state
[   53.675434] device eth0.1 entered promiscuous mode
[   53.680295] device eth0 entered promiscuous mode
[   53.729107] br-lan: port 1(eth0.1) entered blocking state
[   53.734625] br-lan: port 1(eth0.1) entered forwarding state
[   53.865393] eth1: link up (1000Mbps/Full duplex)
[   53.870325] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   54.661352] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   59.059811] ath: EEPROM regdomain: 0x8348
[   59.059818] ath: EEPROM indicates we should expect a country code
[   59.059825] ath: doing EEPROM country->regdmn map search
[   59.059830] ath: country maps to regdmn code: 0x3a
[   59.059835] ath: Country alpha2 being used: US
[   59.059838] ath: Regpair used: 0x3a
[   59.059844] ath: regdomain 0x8348 dynamically updated by user
[   59.059917] ath: EEPROM regdomain: 0x8348
[   59.059920] ath: EEPROM indicates we should expect a country code
[   59.059926] ath: doing EEPROM country->regdmn map search
[   59.059931] ath: country maps to regdmn code: 0x3a
[   59.059934] ath: Country alpha2 being used: US
[   59.059937] ath: Regpair used: 0x3a
[   59.059941] ath: regdomain 0x8348 dynamically updated by user
[   59.136049] pppoe-wan: renamed from ppp0
[   64.461966] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[   64.479245] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[   64.485619] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[   64.493781] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 24984 iram: 38672'
[   64.581654] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[   64.596457] ath10k_pci 0000:00:00.0: rts threshold -1
[   64.897444] br-lan: port 2(wlan0) entered blocking state
[   64.902895] br-lan: port 2(wlan0) entered disabled state
[   64.908562] device wlan0 entered promiscuous mode
[   65.391972] br-lan: port 3(wlan1) entered blocking state
[   65.397436] br-lan: port 3(wlan1) entered disabled state
[   65.403191] device wlan1 entered promiscuous mode
[   65.408177] br-lan: port 3(wlan1) entered blocking state
[   65.413619] br-lan: port 3(wlan1) entered forwarding state
[   65.621350] br-lan: port 3(wlan1) entered disabled state
[   66.671428] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   66.678195] br-lan: port 2(wlan0) entered blocking state
[   66.683649] br-lan: port 2(wlan0) entered forwarding state
[   77.610422] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   77.617104] br-lan: port 3(wlan1) entered blocking state
[   77.622535] br-lan: port 3(wlan1) entered forwarding state
[  128.760648] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[  128.768799] NFSD: Using legacy client tracking operations.
[  128.774404] NFSD: starting 10-second grace period (net f0000029)

yanxin152133, 2022-05-25:
wlan0和wlan1上网的账号都一样吗？你那ipv6可以用，我记得好像都要关ipv6

yanxin152133, 2022-05-25:
> 改成PPPOE连接后，发现内核日志有些变化。如下： [ 53.635915] eth0: link up (1000Mbps/Full duplex) [ 53.640620] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready [ 53.664076] br-lan: port 1(eth0.1) entered blocking [...]

你这是双宽带吗？还是有一个wlan是连光猫后台的？

yanxin152133, 2022-05-25:
> 这是我的交换机界面截图，是LEDE默认的配置，但是我昨晚刷的另外一个openwrt固件，他的eth0是绑定wan口的，eth1是绑定4个lan口的。 [image] 也许不是固件设置问题，可能是驱动的问题。2.4g和5g的WIFI上网是完美的，wan口上网也是好的，那么是不是交换芯片QCA8337的驱动问题呢。下面是内核日志现实，一切好像都很好的样子。 [ 0.465061] 7 fixed-partitions partitions found on MTD device ar934x-nand [ 0.471960] Creating 7 MTD partitions on "ar934x-nand": [ 0.477264] 0x000000000000-0x000000100000 : "u-boot" [ 0.483149] 0x000000100000-0x000000200000 : "boot-flag" [ 0.489236] 0x000000200000-0x000000600000 : "kernel" [ 0.495112] 0x000000600000-0x000006a00000 : "ubi" [ 0.501003] 0x000006a00000-0x000006b00000 : "ft" [ 0.506528] 0x000006b00000-0x000006d00000 : "PKI" [ 0.512152] 0x000006d00000-0x000006e00000 : "caldata" [ 0.865691] switch0: Atheros AR8337 rev. 2 switch registered on mdio.0 [ 1.520568] ag71xx 19000000.eth: connected to PHY at mdio.0:00 [uid=004dd036, driver=Atheros AR8216/AR8236/AR8316] [ 1.531654] eth0: Atheros AG71xx at 0xb9000000, irq 4, mode: rgmii [ 1.872895] ag71xx 1a000000.eth: connected to PHY at fixed-0:00 [uid=00000000, driver=Generic PHY] [ 1.882569] eth1: Atheros AG71xx at 0xba000000, irq 5, mode: sgmii [ 1.890709] NET: Registered protocol family 10 [ 1.899268] Segment Routing with IPv6 [ 1.903141] NET: Registered protocol family 17 [ 1.907703] 8021q: 802.1Q VLAN Support v1.8

交换机是不是也需要设置vlan以及开启相关设置，还是说你的op设置完交换机就会同步你的设置，这个功能我第一次见

yanxin152133, 2022-05-25:
> 这是我的交换机界面截图，是LEDE默认的配置，但是我昨晚刷的另外一个openwrt固件，他的eth0是绑定wan口的，eth1是绑定4个lan口的。 [image] 也许不是固件设置问题，可能是驱动的问题。2.4g和5g的WIFI上网是完美的，wan口上网也是好的，那么是不是交换芯片QCA8337的驱动问题呢。下面是内核日志现实，一切好像都很好的样子。 [ 0.465061] 7 [...]

要不你试试不用vlan

zbluekuma, 2022-05-26:
@WYC-2020
@coolsnowwolf 
两位大师，这两天issue重新整理编辑了一下，提供了一些新的信息，还请指导。

yanxin152133, 2022-05-26:
你这还没解决啊

zbluekuma, 2022-05-26:
@yanxin152133
没有，累死了

zbluekuma, 2022-05-26:
知乎和这个编程网站都可以秒开
[image]
[image]
下面网站可以打开，但是慢
[image]
百度网站打不开

yanxin152133, 2022-05-26:
换换dns
[image: image.png]

yanxin152133, 2022-06-12:
> 知乎和这个编程网站都可以秒开 [image] [image] 下面网站可以打开，但是慢 [image] 百度网站打不开

你试R22.5.5 版本没，我发现这个版本重启完网络好太多了

zbluekuma, 2022-06-14:
@yanxin152133
谢谢，有时间我来试试，这个路由器拿来玩了，新换了asus ax3000v2

closed 2022-07-25 by aiamadeus

yanxin152133, 2022-10-11:
>
> bigbug *@*.
>
>
> 我感觉好像是dns的问题，我用的adguardhome，每次重启完就得等几分钟网络才能正常 我的没用Ad插件😂，电脑端的DNS也清过
>
>
> #9390 <https://github.com/coolsnowwolf/lede/issues/9390> (comment) <#9390
>
> (comment)
>
> #9390 (comment)
> >>,
>
>
>
>
>
>
>
>
>
>
>
> 自带的dhcp里面的dns 转发反正每次重启都恢复成默认的 你可以在LAN口高级设置 指定客户端使用的dns地址 — Reply to this
>
> email directly, view it on GitHub <#9390
> <https://github.com/coolsnowwolf/lede/issues/9390> (comment)
>
> #9390 (comment)
> >,
>
>
>
>
>
>
>
> <#m_3552023774899928098_> Message ID: @.*>
>
> 折腾一天了 这个问题还是没有解决，两个LAN口 其中一个正常，另外一个死活不行 快哭了😭
>
>
>
> #9390 (comment)
> ,
>
>
>
>
>
>
> 你得发下设置
>
> 找到问题了，主要是因为同一个LAN有多个路由导致的 关闭其它的就正常了
啧啧啧，这就找到问题了