coolsnowwolf/lede #9397: 能否升级一下mwan3？

GreatMichaelLee, 2022-05-15:
rt, 现在package里还是2.8.8的，感觉bug非常多，毕竟这个功能还是很实用的，官方都更新到2.10.x了，能否辛苦大佬升级更新一下？直接把官网的拷贝过来编译好像有些问题，可能还需要做些改动适配。多谢。

Beginner-Go, 2022-05-15:
以前升过被大雕revert了

GreatMichaelLee, 2022-05-15:
> 以前升过被大雕revert了

原因是啥啊？2.8.8有点了，多拨时候对于conntrack的状态更新很不稳定，而且对于ip6table识别好像也有问题。

uniessey, 2022-05-16:
> > 以前升过被大雕revert了
> 
> 原因是啥啊？2.8.8有点老了，多拨时候对于conntrack的状态更新很不稳定，而且对于ip6table识别好像也有问题。

我试过几次，在“接口”那里会报错，不能正确添加虚拟设备，即便是手动指定了，也不能很好的均衡负载。

GreatMichaelLee, 2022-05-16:
> > > 以前升过被大雕revert了
> > 
> > 
> > 原因是啥啊？2.8.8有点老了，多拨时候对于conntrack的状态更新很不稳定，而且对于ip6table识别好像也有问题。
> 
> 我试过几次，在“接口”那里会报错，不能正确添加虚拟设备，即便是手动指定了，也不能很好的均衡负载。

是的，有一些莫名其妙的bug，而且我另外报了几个Performance的issue, 有点怀疑和mwan3相关。还是建议大佬们能否考虑升级一下。现在这个版本真的太老了。毕竟大家刷openwrt，几个大头无非就是科学上网，多拨负载均衡，内网穿透等几个重头（个人拙见^_^)

hxlls, 2022-05-16:
自己升级就行了啊，大众用户肯定是一稳定为主，而不是追版本更新，既然revert 了，肯定就是发现有问题了啊

GreatMichaelLee, 2022-05-16:
> 自己升级就行了啊，大众用户肯定是一稳定为主，而不是追版本更新，既然revert 了，肯定就是发现有问题了啊

说了，就是老版本不稳定啊？过去有问题revert，不代表一直有问题不稳定啊？官网也在更新，自己能升级那我肯定闷声不吭就升了（比如说n2n,我就直接改makefile和/etc/init.d/n2n_v2 加了head encrypt和federation了，现在的luci里是配不了的，包更新了，luci没更新），新版本可能luci要做些适配（个人猜测），直接把package拿过来编译是可以过，但实际用的时候会报mwan3track error 15.不能直接套用原版配置，它里面应该加了对Ip6table的一些支持。

另外我一直比较不能苟同这个观点：稳定为主，不追版本更新，稳定就用release版本啊，开发者的动力不就是追求完善功能修复bug吗？我看主干仓库每天也checkin大量代码，除了5.10升5.15稍微当时有点小插曲，其他绝大多数时间也没有说不稳定。如果因为害怕不稳定，可以找个稳定版本一劳永逸就好啦:)

GreatMichaelLee, 2022-05-16:
> > > 自己升级就行了啊，大众用户肯定是一稳定为主，而不是追版本更新，既然revert 了，肯定就是发现有问题了啊
> > 
> > 
> > 说了，就是老版本不稳定啊？过去有问题revert，不代表一直有问题不稳定啊？官网也在更新，自己能升级那我肯定闷声不吭就升了（比如说n2n,我就直接改makefile和/etc/init.d/n2n_v2 加了head encrypt和federation了，现在的luci里是配不了的，包更新了，luci没更新），新版本可能luci要做些适配（个人猜测），直接把package拿过来编译是可以过，但实际用的时候会报mwan3track error 15.不能直接套用2.8.8的配置，它里面应该加了对Ip6table的一些支持。
> > 另外我一直比较不能苟同这个观点：稳定为主，不追版本更新，稳定就用release版本啊，开发者的动力不就是追求完善功能修复bug给用户提供更多的功能和良好体验吗？这稳定和功能迭代肯定是个平衡和取舍的，我看主干仓库每天也checkin大量代码，除了5.10升5.15稍微当时有点小插曲，其他绝大多数时间也没有说不稳定。如果因为害怕不稳定，可以找个稳定版本一劳永逸就好啦:)
> 
> 首先，你用这套源码来编译，本身就是一个老的源码，如果你是追求更新的，可以考虑官方的版本，已经叠代很多，其次，这些单软件的更新，本来自己更新就简单，为什么不自己先做好测试然后提交个PR呢？

首先，我只是用户，不是源代码贡献者不是developer，不专业. 其次，单软件的更新，如果说自己更新简单，那大佬都去做kernel和底层系统更新就好啦，怎么还有这么多开发者在更新其他的单软件呢？那为何单单排斥更新mwan3? 再次，mwan3好像也是用户用的比较多的一个功能吧。如果我自己能搞定测试好，我都懒得在这里发贴提需求求助了。你说呢？

hxlls, 2022-05-16:
> > > > 自己升级就行了啊，大众用户肯定是一稳定为主，而不是追版本更新，既然revert 了，肯定就是发现有问题了啊
> > > 
> > > 
> > > 说了，就是老版本不稳定啊？过去有问题revert，不代表一直有问题不稳定啊？官网也在更新，自己能升级那我肯定闷声不吭就升了（比如说n2n,我就直接改makefile和/etc/init.d/n2n_v2 加了head encrypt和federation了，现在的luci里是配不了的，包更新了，luci没更新），新版本可能luci要做些适配（个人猜测），直接把package拿过来编译是可以过，但实际用的时候会报mwan3track error 15.不能直接套用2.8.8的配置，它里面应该加了对Ip6table的一些支持。
> > > 另外我一直比较不能苟同这个观点：稳定为主，不追版本更新，稳定就用release版本啊，开发者的动力不就是追求完善功能修复bug给用户提供更多的功能和良好体验吗？这稳定和功能迭代肯定是个平衡和取舍的，我看主干仓库每天也checkin大量代码，除了5.10升5.15稍微当时有点小插曲，其他绝大多数时间也没有说不稳定。如果因为害怕不稳定，可以找个稳定版本一劳永逸就好啦:)
> > 
> > 
> > 首先，你用这套源码来编译，本身就是一个老的源码，如果你是追求更新的，可以考虑官方的版本，已经叠代很多，其次，这些单软件的更新，本来自己更新就简单，为什么不自己先做好测试然后提交个PR呢？
> 
> 首先，我只是用户，不是源代码贡献者不是developer，不专业. 其次，单软件的更新，如果说自己更新简单门槛如此低，那大佬都去做kernel和底层系统更新就好啦，怎么还有这么多开发者在维护更新其他的单软件呢？那为何单单就要排斥更新mwan3这个单软件?：） 再次，mwan3好像也是用户用的比较多的一个功能吧。如果我自己能搞定测试好，我都懒得在这里发贴提需求求助了。你说呢？

首先，这里不是论坛，是issue，每个issue回复或者新起一个issue都会在维护者邮箱收到一封邮件，滥用issue 会造成维护人员根本不想去看邮件，太多后就会批量关闭，也就是根本不理会（PS：大雕已经很少去回复issue，很可能就是直接邮件都不看了），甚至会关闭github 的issue功能 ，到时候你连真正提交问题的地方都没有，你要发帖子讨论，可以去对应的论坛发。既然你说的感觉现有版本BUG非常多，你就要提供对应的日志或者提交去给开发者，而不是说凭感觉，感觉是一个很主观的东西。

lloyd18, 2022-05-16:
你不懂就闭嘴，事咋那么多啊 @hxlls
issue 不给提意见？那开着有什么意义？

GreatMichaelLee, 2022-05-16:
> 试过升级最新版，发现多拨无法使用了，负载均衡也出现异常

是的，我前面也说了，我也试过，直接编译会发现不能直接用，会报error 15

ljcbaby, 2022-05-16:
u1s1，mwan3 和 mwan3helper 感觉确实需要更新。我现在在用的都不知道 ip6 的规则怎么配置。

Beginner-Go, 2022-05-16:
日 我邮箱要炸了