coolsnowwolf/lede #9398: 固件是5.15的，软件源是5.10的， * pkg_hash_check_unresolved: cannot find dependency kernel

shenwrt, 2022-05-15:
* pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.113-1-bce8e44f7254c1ead36f72c6d346f19e) for kmod-libphy
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.113-1-bce8e44f7254c1ead36f72c6d346f19e) for kmod-pps
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.113-1-bce8e44f7254c1ead36f72c6d346f19e) for kmod-ptp
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.113-1-bce8e44f7254c1ead36f72c6d346f19e) for kmod-tg3
使用固件里自带的软件源安装软件，安装好多软件都报这个错误，搜了下好像是说软件源不匹配（不太懂linux，见谅）
请问怎么弄呢？download.openwrt.org  好像没有5.15的软件源，我看了下编译完固件好像同时也编译了好多软件，是不是可以自己搭建5.15的软件源呢？求大佬解答，感谢感谢

shenwrt, 2022-05-16:
自己搭建本地源搞定了..

closed 2022-05-16 by shenwrt

raofeng, 2022-05-27:
> 自己搭建本地源搞定了..

遇到同样的问题，也是自己搭建了源，就可以安装了，第三方的源就安装不了。

shenwrt, 2022-05-30:
> 

是的，我也发现了，折腾了几天，好像说要改lede编译内核成5.10就能解决，5.15不知道怎么弄，最后放弃自己编译了，直接用别家的openwrt了。。。